openwrt/packages #15215: subversion: use LTS version in stable releases of OpenWrt

BKPepe, 2021-03-22:
Maintainer: @val-kulkov 
Environment: OpenWrt 19.07

Description:

Hello @val-kulkov,

In OpenWrt 19.07 branch, there is svn 1.12.2, which is no longer supported. 
Apache currently provides two versions of the old version 1.10.7, which would be a downgrade, and it could bring unwanted issues due to new features which are not present in old versions. However, it is LTS, so it still gets security releases or rather updates it to the latest release, 1.14.1 LTS. What do you think? 

By terms of security and maintainership, it would be great if we can use LTS versions instead of regular releases because that one could be vulnerable to security issues like [CVE-2020-17525](https://lists.debian.org/debian-security-announce/2021/msg00032.html)

val-kulkov, 2021-03-22:
Hi @BKPepe,

If I understand it correctly, [CVE-2020-17525 is fixed](https://subversion.apache.org/security/CVE-2020-17525-advisory.txt) in both the 1.10.7 LTS and in the latest release, 1.14.1. So yes, it would be nice to backport either 1.10.7 or 1.14.1 to OpenWrt 19.07 branch. Personally I would think that backporting 1.14.1 would make more sense because not only 1.14.1 has the most recent security fixes but it is, well, a more modern version. However, backporting a more modern version to an older OpenWrt version may present some problems of its own. I have not explored this avenue at all and I cannot tell how feasible it is to backport 1.14.1 to OpenWrt 19.07.

@BKPepe : if you are interested into taking over the maintainership of SVN here, please do feel free to submit a PR which I will be happy to OK. Personally I am no longer using SVN, maintaining it here has been a burden and I have been doing the work to keep SVN at OpenWrt simply because no one else so far has volunteered to take care of it.

BKPepe, 2021-03-22:
@val-kulkov Hi!

I am so sorry, but I am not interested in ``svn`` here because I am not using it as I'm using ``git``daily.  I can take a look if 1.14.1 works on OpenWrt 19.07 and send you a PR.

hnyman, 2023-01-14:
19.07 is EoL.
Master, 22.03 and 21.02 do have subversion 1.14.1

closed 2023-01-14 by hnyman