openwrt/packages #15223: Luci-proto-bonding related compatibility issues

qiuzi, 2021-03-22:
Using the latest version of luci source code results in list slaves'eth2'

And proto_config_add_string "slaves" does not support list variables

he-ma, 2021-04-06:
@qiuzi
Hi,
I'm absolutely not sure what you mean.
Could you please detail which version of luci you use, what do you mean by `results in list slaves'eth2'` 
and how you got the error you describe.
Some pics could also be useful.

Regards
Helge

qiuzi, 2021-04-06:
@he-ma 
https://github.com/openwrt/luci/blob/master/protocols/luci-proto-bonding/htdocs/luci-static/resources/protocol/bonding.js#L177
https://github.com/openwrt/packages/blob/master/net/bonding/files/lib/netifd/proto/bonding.sh#L40 
not compatible with luci

qiuzi, 2021-04-06:
[image]

he-ma, 2021-04-08:
Sorry, but I couldn't reproduce your problem.
I set up a new machine and used luci based up to the commit dated on Apr 6, 2021.
Below some screenshots which show the functionality still working.
[image]
[image]
[image]
[image]
[image]
[image]

qiuzi, 2021-04-08:
@he-ma 
 `ubus call network.interface.test status`
```
{
        "up": true,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "uptime": 993,
        "l3_device": "bonding-test",
        "proto": "bonding",
        "updated": [
                "addresses"
        ],
        "metric": 0,
        "dns_metric": 0,
        "delegation": true,
        "ipv4-address": [
                {
                        "address": "111.111.111.111",
                        "mask": 24
                }
        ],
        "ipv6-address": [

        ],
        "ipv6-prefix": [

        ],
        "ipv6-prefix-assignment": [

        ],
        "route": [

        ],
        "dns-server": [

        ],
        "dns-search": [

        ],
        "neighbors": [

        ],
        "inactive": {
                "ipv4-address": [

                ],
                "ipv6-address": [

                ],
                "route": [

                ],
                "dns-server": [

                ],
                "dns-search": [

                ],
                "neighbors": [

                ]
        },
        "data": {

        }
}

```
no device information found
how is your output?

he-ma, 2021-04-08:
Well, exept the 'uptime' and 'address' values (of course) it's the same.
If you have a look at `ubus call bonding status` you'll see the slaves.

But I think that's all beyond your initial issue that LuCI won't work.

qiuzi, 2021-04-08:
@he-ma 
https://github.com/openwrt/packages/blob/master/net/bonding/files/lib/netifd/proto/bonding.sh#L160
```
	local slaves
	json_get_vars slaves
	echo "$slaves" >> /tmp/1.txt
```
Adding the output on line 160, I get an empty 1.txt file
Variables are not output correctly

/etc/config/network
```
config interface 'test'
	option proto 'bonding'
	option ipaddr '121.0.1.1'
	option netmask '255.255.255.0'
	option slaves 'eth2 eth3'
	option bonding_policy 'balance-rr'
	option packets_per_slave '1'
	option xmit_hash_policy 'layer2'
	option all_slaves_active '0'
	option link_monitoring 'off'

```
/tmp/1.txt
```

eth2 eth3
eth2
eth3
```
cat /sys/class/net/bonding-test/bonding/slaves
eth2 eth3

qiuzi, 2021-04-09:
@he-ma 
Use vmware to load the image, the problem can be reproduced
[openwrt-x86-64-generic-squashfs-combined.zip](https://github.com/openwrt/packages/files/6283155/openwrt-x86-64-generic-squashfs-combined.zip)

he-ma, 2021-04-09:
@qiuzi
Sorry, I have no vmware.
As I wrote yesterday, the LuCI and the script to set up the bonding stuff -at least for me- works.
I don't know why your config file now has `option`  for the slaves as initially you had (the correct) `list`

I fear I can't help you anymore as I can't see/can't reproduce any misbehaviour.

feckert, 2021-04-13:
Fixed with https://github.com/openwrt/packages/commit/57a77386de7dda12f60bf3021efcde7f059833c8

closed 2021-04-13 by feckert