openwrt/packages #15224: mwan3: add support for multiple gateways in a single subnet

sumpfralle, 2021-03-22:
Maintainer: @aaronjg, @feckert
Environment: OpenWrt snapshot (build time: Sun Mar 14 05:24:11 2021)

Description:

The subnet of the WAN interface of my router contains multiple gateways (different IP addresses in this subnet).
I would like to use mwan3 for switching between these gateways (as default gateways).

Since mwan3 operates on *interfaces*, I created multiple logical interface (using the same physical interface - here *eth0.2*) with different configured gateways. I configured these WAN interfaces as interfaces for mwan3.

At the moment, such a setup (multiple gateways on a single interface) does not seem to be supported by mwan3.

# Currently missing features / problems

## Routing tables
Currently mwan3 populates only the routing table for the lowest table ID properly, since `mwan3_create_iface_route` processes only the *first* matching interface (see `mwan3_route_line_dev`).

It should be easy to fix this by changing the flow of `mwan3_create_iface_route`. Additionally maybe a special handling of the `default` rules will be required.

## Firewall rules
The iptables rules for matching the traffic are based on the interface. In case of multiple gateways connected to the same interface, additional MAC-based matching would be necessary (e.g. `-m mac --mac-source ...`).

This change requires:
* an IP -> MAC lookup
* and some changes to the iptables rules flow, since `mwan3_hook` is used in `PREROUTING` and `OUTPUT` - but `--mac-source` is only allowed in `PREROUTING`, `INPUT` and `FORWARD`.

## `sockopt_wrap`
Probably no changes are required, since the `FWMARK` value can be specified explicitly (avoiding any ambiguity).

# Relevance?
I am willing to dive further into this topic, in order to come up with a clean implementation of the changes for the firewall rules and the policy routing rules. But I would only do this, if such a development is considered to be useful for *mwan3*.

Or maybe someone has an idea, how to approach this problem (multiple gateways in the same subnet) in a different way?

Alternatively, you may feel free to close this issue, if such a feature (multiple gateways available in a single subnet) is outside of the scope of mwan3's goals.

aaronjg, 2021-03-22:
There is another issue in that mwan3track currently tells sockopt_wrap to use the default routing table, so it would need to be changed to tell it to give a FWMARK for the relevant table rather than the firewall mark for the default table.

I don't quite follow how the MAC address would solve things. I believe the firewall issue that you refer to is related to tracking incoming connections? For incoming connections, you would somehow need to determine which FWMARK to give it for conntrack, and I'm not sure how MAC address matching could fix this.

In general, It would be useful to remove the tight linkage between one tracked mwan3 interface and one openwrt interface. Currently this is enforced because the mwan3 interface is chosen by the mwan3 config section name, which has to be the same as an openwrt interface.

This thread on the forum may also be relevant, though I don't believe the user ever posted his solution if he found one.
https://forum.openwrt.org/t/mwan3-in-a-multi-router-setup/82204/13

sumpfralle, 2021-03-25:
> I don't quite follow how the MAC address would solve things. I believe the firewall issue that you refer to is related to tracking incoming connections? For incoming connections, you would somehow [...]

Maybe I am on the wrong track, but if we want to distinguish whether an incoming package was received via A or B (the two gateways in the WAN interface), then I can only think of using the MAC address of the incoming packet (since no IP-related information is attached by A or B). An alternative could be something like different TOS-marks to be applied by A and B, but this requires a coordinated configuration of the three routers and is probably not really feasible.

> In general, It would be useful to remove the tight linkage between one tracked mwan3 interface and one openwrt interface.

Yes, I had a similar feeling when looking at the code. But I do not really comfortable to implement such a major change on my own.

In case you think, that *someone* would do this in the foreseeable future, then I would prefer to postpone my efforts.

> This thread on the forum may also be relevant, though I don't believe the user ever posted his solution if he found one.
> https://forum.openwrt.org/t/mwan3-in-a-multi-router-setup/82204/13

Thank you for this reference. Indeed the discussed problem matches mine quite perfectly.

ptpt52, 2021-04-21:
I prefer use ipvlan instead of multiple gateways/networks on one interface

CodeFetch, 2021-06-09:
fw3 is missing network namespace support. Implementing network namespace support is the most reasonable approach I can think of for cleaning up mwan3.

feckert, 2023-05-03:
I currently have no setup and no use for it and as @CodeFetch has already written, this would be the cleanest solution.

closed 2023-05-03 by feckert