openwrt/packages #15234: mc: segfault in SSH session

DazzyWalkman, 2021-03-23:
mc - 4.8.26-1
Environment: OpenWrt SNAPSHOT r16313-851dadc257 x86-64, Tue Mar 23 00:05:10 2021
Description:
When I invoked mc from ssh commandline, mc crashed with segfault.
dmesg output
``[   49.822486] mc[2407]: segfault at 0 ip 00007fbdfa97a246 sp 00007ffc3690f4a0 error 4 in libglib-2.0.so.0.6600.4[7fbdfa93c000+6d000]``
``[   49.827486] Code: 89 e1 f3 aa 5a 48 89 ef 5b 5d 41 5c 41 5d 41 5e 41 5f e9 5b a1 00 00 90 58 5b 5d 41 5c 41 5d 41 5e 41 5f c3 55 48 89 fd 53 51 <48> 8b 1f 85 f6 74 0b 48 89 df 31 db 67 e8 36 a1 00 00 48 89 ee bf``

Steps to reproduce:
1. Get https://downloads.openwrt.org/snapshots/targets/x86/64/openwrt-x86-64-generic-ext4-combined.img.gz
2. sysupgrade my kvm test-instance with the build obtained in step 1
3. invoke mc in a ssh session after successful sysupgrade
4. segfault then terminal text misalign (please see att.)

Note: mc does not crash in the local console. mcedit runs fine both in ssh and local console.
[image]

neheb, 2021-03-24:
More discussion on this: https://github.com/openwrt/packages/pull/14752

zyxmon, 2021-03-26:
This helped me - https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=253831
Here is a patch that may work for you - https://github.com/Entware/entware-packages/commit/56690ded2254ba8f48837811d2bdc508bb5da45c#diff-ff9798b58137188d41cb8e989b1ef10661a58a005909d68821626f3608fec7eb

zyxmon, 2021-03-26:
Patched version starts ok but can hang on directory change.

DazzyWalkman, 2021-03-29:
Thanks for the tip. @zyxmon 
`mc -u`  (no subshell) in the freebsd bug report works as a temporary workaround.
And 
https://midnight-commander.org/attachment/ticket/4213/4213_cmdline.2.patch
works for me, no segfault or freeze.
I'm glad it got reported upstream.
https://midnight-commander.org/ticket/4213
Hopefully it will be fixed in the upcoming 4.8.27 release.

DazzyWalkman, 2021-08-19:
This issue has been worked around in upstream 4.8.27 release.
Closed with c2b68710e1d0db47623d3c1d2c7f6d5be75fc3dc

closed 2021-08-19 by DazzyWalkman