openwrt/packages #15238: openconnect: dnsmasq not restarted on disconnect clean-up

audrius-u, 2021-03-24:
Environment: OpenWrt 19.07.4 on GL-Inet MT1300 (mips), openconnect 8.05-1

Description:

*/lib/netifd/vpnc-script* in function `do_disconnect()` does not restart dnsmasq upon clean-up, leaving (very likely unreachable) DNS settings still configured. Currently:
```
do_disconnect() {
        rm -f "/tmp/dnsmasq.d/openconnect.$TUNDEV"
        proto_init_update "$TUNDEV" 0
        proto_send_update "$INTERFACE"
}
```
Suggest changing to:
```
do_disconnect() {
        rm -f "/tmp/dnsmasq.d/openconnect.$TUNDEV"
        /etc/init.d/dnsmasq restart
        proto_init_update "$TUNDEV" 0
        proto_send_update "$INTERFACE"
}
```
optionally wrapping with  `if [ -f "/tmp/dnsmasq.d/openconnect.$TUNDEV" ]; then <...>`

nmav, 2021-03-28:
Would you like to send a pull request with your proposed fix?

db260179, 2022-09-09:
> Would you like to send a pull request with your proposed fix?

I've done a pull request

https://github.com/openwrt/packages/pull/19335

closed 2024-05-09 by nmav