openwrt/packages #15258: miniupnpd: multiple unhelpful messages in the log

stangri, 2021-03-25:
Maintainer: @neheb 
Environment: x86-64, Sophos SG-105, 21.02-SNAPSHOT r15926

Description:

I keep seeing the messages like below in the log. Is it missing a `d` at the end (as in checking is ignored)? There's no indication on why is it being reported or what caused it. Is it possible to improve this message to make it more clear?

```
00:11:33 miniupnpd[5981]: private/reserved address checking is ignore
00:11:33 miniupnpd[5981]: private/reserved address checking is ignore
00:11:33 miniupnpd[5981]: private/reserved address checking is ignore
00:17:38 miniupnpd[5981]: private/reserved address checking is ignore
00:17:39 miniupnpd[5981]: private/reserved address checking is ignore
00:17:39 miniupnpd[5981]: private/reserved address checking is ignore
00:35:23 miniupnpd[5981]: private/reserved address checking is ignore
00:35:24 miniupnpd[5981]: private/reserved address checking is ignore
00:35:24 miniupnpd[5981]: private/reserved address checking is ignore
```

neheb, 2021-03-25:
I don't really maintain this. It looks like some upstream bug.

pali, 2021-04-27:
@neheb: this change is not part of upstream and it was accepted into openwrt by [you](https://github.com/openwrt/packages/pull/14031).

Moreover this patch is incorrect and and is breaking upnp protocol as desribed in [comments](https://github.com/miniupnp/miniupnp/pull/511).

So please do not instruct users to forward openwrt bugs and issues to upstream projects like it already [happened](https://github.com/miniupnp/miniupnp/issues/542) because upstream developers really do not have a time to investigate issues caused by downstream openwrt patches.

stangri, 2021-04-28:
@ptpt52 -- can you please advise me when have I started seeing these messages in the log? What do they mean?

ptpt52, 2021-07-02:
I am trying to drop this option, Please help test https://github.com/openwrt/packages/pull/16011

Neustradamus, 2021-09-16:
There is a new patch from @ptpt52:
- https://github.com/openwrt/packages/pull/16011#issuecomment-920538698

It replaces the old 301-ext_ip_reserved_ignore.patch.

Tested by @openips!

tiagogaspar8, 2022-08-25:
Hi @stangri, is this still vaild?

stangri, 2022-08-25:
I no longer use miniupnpd, not sure if this issue persists.

closed 2022-08-25 by stangri