openwrt/packages #15267: workflows: seeing multi-arch-test-build.yml compute wrong changed packages

pprindeville, 2021-03-26:
Maintainer: @aparcar 
Environment: Test Builds in GitHub

Description:

On PR #14711 two files are changed from `master`:

- `net/strongswan/Makefile`
- `net/strongswan/files/ipsec.init`

Yet looking at (this test build)[https://pipelines.actions.githubusercontent.com/5yKFRfuu84LeYF9JWXeroTFD7H8ygjTZsUcmjLvv85kJt4ah0c/_apis/pipelines/1/runs/4397/signedlogcontent/11?urlExpires=2021-03-25T06%3A09%3A01.4076076Z&urlSigningMethod=HMACV1&urlSignature=cPtMpRoQUKoDyR2XAD%2B%2BAq84DKz4as%2F%2BYtDPSv%2FXAec%3D] we see the following:

```
...
2021-03-25T01:10:09.5633256Z ##[group]Run # only detect packages with changes
2021-03-25T01:10:09.5634146Z # only detect packages with changes
2021-03-25T01:10:09.5634988Z PKG_ROOTS=$(find . -name Makefile | grep -v ".*/src/Makefile" | sed -e 's@./\(.*\)/Makefile@\1/@')
2021-03-25T01:10:09.5635886Z CHANGES=$(git diff --diff-filter=d --name-only origin/master)
2021-03-25T01:10:09.5636699Z 
2021-03-25T01:10:09.5637130Z for ROOT in $PKG_ROOTS; do
2021-03-25T01:10:09.5637714Z   for CHANGE in $CHANGES; do
2021-03-25T01:10:09.5638227Z     if [[ "$CHANGE" == "$ROOT"* ]]; then
2021-03-25T01:10:09.5638864Z       PACKAGES+=$(echo "$ROOT" | sed -e 's@.*/\(.*\)/@\1 @')
2021-03-25T01:10:09.5639394Z       break
2021-03-25T01:10:09.5639781Z     fi
2021-03-25T01:10:09.5640223Z   done
2021-03-25T01:10:09.5640657Z done
2021-03-25T01:10:09.5641409Z 
2021-03-25T01:10:09.5642154Z # fallback to test packages if nothing explicitly changes this is
2021-03-25T01:10:09.5642983Z # should run if other mechanics in packages.git changed
2021-03-25T01:10:09.5643763Z PACKAGES="${PACKAGES:-vim tmux bmon}"
2021-03-25T01:10:09.5644250Z 
2021-03-25T01:10:09.5644706Z echo "Building $PACKAGES"
2021-03-25T01:10:09.5645368Z echo "PACKAGES=$PACKAGES" >> $GITHUB_ENV
2021-03-25T01:10:09.5692714Z shell: /usr/bin/bash -e {0}
2021-03-25T01:10:09.5693197Z ##[endgroup]
2021-03-25T01:10:11.6396291Z Building ninja gitlab-runner pkgconf gcc asu meson emailrelay pigeonhole netdata syslog-ng backuppc sudo ksmbd mtd-rw php7 php7-pecl-imagick lua-libmodbus lua-ev python-aiohttp pyodbc python3-libselinux python-docker openpyxl python-websocket-client python-zipp pillow python3-libsemanage php8 forked-daapd madplay pulseaudio upmpdcli espeak mpd joe haserl nnn yq watchcat vim sysstat docker ssdeep tang dockerd setools attr findutils auc hwdata prometheus whois gddrescue fish yara mc tar qemu zstd augeas smartmontools nano hashdeep powertop reptyr bluez readsb dosfstools acl stoken btrfs-progs openocd taskwarrior selinux-python collectd runc syncthing semodule-utils mariadb unrar spi-tools stress-ng conmon nginx iperf openvpn static-neighbor-reports uacme adblock ibrdtnd subversion clamav miniupnpc ibrdtn-tools knot-resolver aircrack-ng openssh linknx vncrepeater banip keepalived xray-core aria2 wavemon safe-search libreswan cshark lftp nmap strongswan nextdns vpn-policy-routing miniupnpd tor git family-dns mosquitto tor-hs knxd modemmanager wg-installer adguardhome ooniprobe znc haproxy ksmbd-tools kismet frr bind simple-adblock udpxy isc-dhcp sysrepo i2pd net-snmp dcwapd umurmur hs20 https-dns-proxy mwan3 libwebsockets uci2 freetype libqmi libuv pugixml libpng libupnpp libmatroska taglib minizip xmlrpc-c libnetconf2 libplist libstrophe libredblack libv4l libre2 gpgme libndpi redis protobuf libdrm libuhttpd tiff liburing ibrdtn libmraa check libx264 libpqxx libgd log4cplus ibrcommon uw-imap nss db47 libebml libnpupnp vips mjpg-streamer imagemagick v4l2rtspserver gerbera ffmpeg 
2021-03-25T01:10:11.6600648Z ##[group]Run BRANCH="${GITHUB_BASE_REF#refs/heads/}"
2021-03-25T01:10:11.6601557Z BRANCH="${GITHUB_BASE_REF#refs/heads/}"
2021-03-25T01:10:11.6602113Z echo "Building for $BRANCH"
2021-03-25T01:10:11.6602650Z echo "BRANCH=$BRANCH" >> $GITHUB_ENV
2021-03-25T01:10:11.6647862Z shell: /usr/bin/bash -e {0}
2021-03-25T01:10:11.6648258Z env:
2021-03-25T01:10:11.6655541Z   PACKAGES: ninja gitlab-runner pkgconf gcc asu meson emailrelay pigeonhole netdata syslog-ng backuppc sudo ksmbd mtd-rw php7 php7-pecl-imagick lua-libmodbus lua-ev python-aiohttp pyodbc python3-libselinux python-docker openpyxl python-websocket-client python-zipp pillow python3-libsemanage php8 forked-daapd madplay pulseaudio upmpdcli espeak mpd joe haserl nnn yq watchcat vim sysstat docker ssdeep tang dockerd setools attr findutils auc hwdata prometheus whois gddrescue fish yara mc tar qemu zstd augeas smartmontools nano hashdeep powertop reptyr bluez readsb dosfstools acl stoken btrfs-progs openocd taskwarrior selinux-python collectd runc syncthing semodule-utils mariadb unrar spi-tools stress-ng conmon nginx iperf openvpn static-neighbor-reports uacme adblock ibrdtnd subversion clamav miniupnpc ibrdtn-tools knot-resolver aircrack-ng openssh linknx vncrepeater banip keepalived xray-core aria2 wavemon safe-search libreswan cshark lftp nmap strongswan nextdns vpn-policy-routing miniupnpd tor git family-dns mosquitto tor-hs knxd modemmanager wg-installer adguardhome ooniprobe znc haproxy ksmbd-tools kismet frr bind simple-adblock udpxy isc-dhcp sysrepo i2pd net-snmp dcwapd umurmur hs20 https-dns-proxy mwan3 libwebsockets uci2 freetype libqmi libuv pugixml libpng libupnpp libmatroska taglib minizip xmlrpc-c libnetconf2 libplist libstrophe libredblack libv4l libre2 gpgme libndpi redis protobuf libdrm libuhttpd tiff liburing ibrdtn libmraa check libx264 libpqxx libgd log4cplus ibrcommon uw-imap nss db47 libebml libnpupnp vips mjpg-streamer imagemagick v4l2rtspserver gerbera ffmpeg 
2021-03-25T01:10:11.6662891Z ##[endgroup]
2021-03-25T01:10:11.6743510Z Building for master
2021-03-25T01:10:11.6934295Z ##[group]Run openwrt/gh-action-sdk@v1
2021-03-25T01:10:11.6934792Z env:
2021-03-25T01:10:11.6942009Z   PACKAGES: ninja gitlab-runner pkgconf gcc asu meson emailrelay pigeonhole netdata syslog-ng backuppc sudo ksmbd mtd-rw php7 php7-pecl-imagick lua-libmodbus lua-ev python-aiohttp pyodbc python3-libselinux python-docker openpyxl python-websocket-client python-zipp pillow python3-libsemanage php8 forked-daapd madplay pulseaudio upmpdcli espeak mpd joe haserl nnn yq watchcat vim sysstat docker ssdeep tang dockerd setools attr findutils auc hwdata prometheus whois gddrescue fish yara mc tar qemu zstd augeas smartmontools nano hashdeep powertop reptyr bluez readsb dosfstools acl stoken btrfs-progs openocd taskwarrior selinux-python collectd runc syncthing semodule-utils mariadb unrar spi-tools stress-ng conmon nginx iperf openvpn static-neighbor-reports uacme adblock ibrdtnd subversion clamav miniupnpc ibrdtn-tools knot-resolver aircrack-ng openssh linknx vncrepeater banip keepalived xray-core aria2 wavemon safe-search libreswan cshark lftp nmap strongswan nextdns vpn-policy-routing miniupnpd tor git family-dns mosquitto tor-hs knxd modemmanager wg-installer adguardhome ooniprobe znc haproxy ksmbd-tools kismet frr bind simple-adblock udpxy isc-dhcp sysrepo i2pd net-snmp dcwapd umurmur hs20 https-dns-proxy mwan3 libwebsockets uci2 freetype libqmi libuv pugixml libpng libupnpp libmatroska taglib minizip xmlrpc-c libnetconf2 libplist libstrophe libredblack libv4l libre2 gpgme libndpi redis protobuf libdrm libuhttpd tiff liburing ibrdtn libmraa check libx264 libpqxx libgd log4cplus ibrcommon uw-imap nss db47 libebml libnpupnp vips mjpg-streamer imagemagick v4l2rtspserver gerbera ffmpeg 
2021-03-25T01:10:11.6949234Z   BRANCH: master
2021-03-25T01:10:11.6949648Z   ARCH: x86_64-master
2021-03-25T01:10:11.6950109Z   FEEDNAME: packages_ci
2021-03-25T01:10:11.6950546Z ##[endgroup]
...
```

so clearly `CHANGES` is not being calculated correctly.

This results in too many packages being rebuilt, and sometimes blocking on failures when these unrelated packages themselves are failing to build do to being unmaintained.

In my case, my builds aren't even getting to `strongswan` because `pigeonhole` consistently fails.

pprindeville, 2021-03-26:
@ynezz Following up on what we talked about on IRC earlier today.

aparcar, 2021-03-26:
Hi, the linked buildlog does no longer work. Looking at the change detection step, it looks rather fine:
[image]

What's currently faulty and I [already proposed to revert](https://github.com/openwrt/packages/pull/15273) is the detection of depending packages:
[image]

Before the recent contribution of @lipnitsk or 13c1f2bcda33ab8fc17ede1f43f60e0aac8b7cab the detection might have been faulty anyway.

The current CI error seems to be due to the above mentioned faulty detection of depending packages:

```
 make[1]: Entering directory '/home/build/openwrt'
make[1]: *** No rule to make target 'package/strongswan-mod-eap-identity/download'.  Stop.
make[1]: Leaving directory '/home/build/openwrt'
make: *** [/home/build/openwrt/include/toplevel.mk:223: package/strongswan-mod-eap-identity/download] Error 2```

dangowrt, 2021-03-26:
This is definitely a regression because CI is quite useless now, failing because random unrelated packages cannot be found. This is new and must have deteriorated within the past couple of weeks, I've used the CI on PRs before and never seen this problem -- now it occurs every time on each PR I open.

lipnitsk, 2021-03-26:
Do we need to enforce that folder name == PKG_NAME?

champtar, 2021-03-26:
We can't as many Makefile define multiple packages

pprindeville, 2021-03-26:
> This is definitely a regression because CI is quite useless now, failing because random unrelated packages cannot be found. This is new and must have deteriorated within the past couple of weeks, [...]

I've been seeing it for 2+ months.

pprindeville, 2021-03-26:
> Do we need to enforce that folder name == PKG_NAME?

Conversely we could just add scripting to extract the PKG_NAME from the folder/Makefile?

closed 2022-05-15 by neheb