openwrt/packages #15279: <forked-daapd>: is potentially malware

jordanbs, 2021-03-27:
Maintainer: `ejurgensen` : DO NOT ALERT, INVESTIGATE ASAP
Environment: Raspbian

Description:
WARNING: POST CONTAINS DANGEROUS LINKS, CLICK WITH CAUTION.

The installation instructions for forked-daapd on openwrt and raspbian asks users to add an opkg repository to a URL which blatantly distributes malware.

The site at the opkg domain, `http://www.gyfgafguf.dk`, after clicking two divs, attempts to download a suspicious file named `RemoteControl.avi`. I have not download nor inspected this file to verify it is malware.

[image]

Given that the install instructions asks users to trust opkg and debian package repositories hosted at this URL, this is concerning.

From https://openwrt.org/docs/guide-user/services/media_server/forked-daapd:
```
Release | Arch | opkg line
Attitude Adjustment 12.09 | brcm47xx | src/gz forked-daapd http://www.gyfgafguf.dk/openwrt/attitude_adjustment/12.09/brcm47xx/packages
Attitude Adjustment 12.09 | ar71xx | src/gz forked-daapd http://www.gyfgafguf.dk/openwrt/attitude_adjustment/12.09/ar71xx/packages
Barrier Breaker 14.07 | brcm47xx | src/gz forked-daapd http://www.gyfgafguf.dk/openwrt/barrier_breaker/14.07/brcm47xx/packages
Barrier Breaker 14.07 | ar71xx | src/gz forked-daapd http://www.gyfgafguf.dk/openwrt/barrier_breaker/14.07/ar71xx/packages
```

Similarly, on Raspbian, the official instructions request users to add this apt repository and a corresponding apt-key. From https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=49928:
```
How to install
Add repository key
wget -q -O - http://www.gyfgafguf.dk/raspbian/forked-daapd.gpg | sudo apt-key add -
Add either of these lines to /etc/apt/sources.list:
(for stretch) deb http://www.gyfgafguf.dk/raspbian/forked-daapd/ stretch contrib
(for buster) deb http://www.gyfgafguf.dk/raspbian/forked-daapd/ buster contrib
Run sudo apt update and then sudo apt install forked-daapd
```
Perhaps ejurgensen is referring to this at the end of their raspbian post, saying `Use at own risk  ;)`.

BKPepe, 2021-03-27:
Thank you for letting us know about this issue in community OpenWrt documentation. You should not run OpenWrt 14.07 or 12.09, because all of these versions are deprecated for ages. For now, there is only supported version is OpenWrt 19.07.

What I could do from my side, I removed from the article any mentions about older versions of OpenWrt. Because forked-daapd is present in this repository, which means that you can install it by `opkg update && opkg install forked-daapd`, so you don't need to add any other changes, but this was said that before my changes. I just make it more clear.

Regarding the content of the website, you need to contact domain owner and let him know about it. For example, you can use PM on RPI's forum.

neheb, 2021-03-27:
Those instructions are not only wrong, they're from 2013. How is this relevant to OpenWrt?

jordanbs, 2021-03-28:
The [potential malware-installing] instructions are in openwrt.org’s official documentation. There is no strong indication that they are wrong or out of date. Perhaps you should remove the page if it is entirely defunct?

brada4, 2021-04-18:
virustotal does not back your malware claim. It is just unmaintained site that lost avi mime type over time

brada4, 2021-04-19:
[image]
Here you go

ejurgensen, 2021-04-19:
Lol I love that this thread is bringing attention to that old claymation video. I can add that it even made it to national tv, which is where my media career peaked. 

As it is hopefully already clear I do of course not host anything malicious on my web site. Scanning, Googling, internet archive or whatever should proove it. The site also isn’t even used for OpenWrt’s forked-daapd any more, so the content should be irrelevant.

jordanbs, 2021-04-20:
Utmost apologies @ejurgensen. I got spooked by an auto-downloaded file named RemoteControl.avi.

My house is planning a screening of your film. Subscribe for the review.

closed 2021-04-20 by jordanbs