openwrt/packages #15309: cryptsetup-2.1.0: Command failed with code -3 (out of memory).

mmokrejs, 2021-03-30:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment:

```
Netgear Nighthawk X4S R7800
ARMv7 Processor rev 0 (v7l)
OpenWrt 19.07.4 r11208-ce6496d796 / LuCI openwrt-19.07 branch git-21.079.58580-41ab871
4.14.195
```

Description: cryptsetup does not out any output and silently does not create /dev/mapper/ device. An error can be seen when `--debug` is included. maybe newer version of the package would behave better. There is `2.3.5` out already.

```
# cryptsetup --debug -d mykey luksOpen /dev/sda1 luks-1f5cedfe-9262-4c75-b19f-b2f03bbec3fc
# cryptsetup 2.1.0 processing "cryptsetup --debug -d mykey luksOpen /dev/sda1 luks-1f5cedfe-9262-4c75-b19f-b2f03bbec3fc"
# Running command open.
# Locking memory.
# Installing SIGINT/SIGTERM handler.
# Unblocking interruption on signal.
# Allocating context for crypt device /dev/sda1.
# Trying to open and read device /dev/sda1 with direct-io.
# Initialising device-mapper backend library.
# Trying to load any crypt type from device /dev/sda1.
# Crypto backend (Linux 4.14.195 kernel cryptoAPI) initialized in cryptsetup library version 2.1.0.
# Detected kernel Linux 4.14.195 armv7l.
# Loading LUKS2 header (repair disabled).
# Opening lock resource file /run/cryptsetup/L_8:1
# Acquiring read lock for device /dev/sda1.
# Verifying read lock handle for device /dev/sda1.
# Device /dev/sda1 READ lock taken.
# Trying to read primary LUKS2 header at offset 0x0.
# Opening locked device /dev/sda1
# Veryfing locked device handle (bdev)
# LUKS2 header version 2 of size 16384 bytes, checksum sha256.
# Checksum:c2c37213d637669882283a76ac2cfa011ca4b99e1527a9871d55775be3b8c9cf (on-disk)
# Checksum:c2c37213d637669882283a76ac2cfa011ca4b99e1527a9871d55775be3b8c9cf (in-memory)
# Trying to read secondary LUKS2 header at offset 0x4000.
# Opening locked device /dev/sda1
# Veryfing locked device handle (bdev)
# LUKS2 header version 2 of size 16384 bytes, checksum sha256.
# Checksum:0f1bb9b142e9e75002ab8e912940c94fa88ad2d4a3efe9ca4b280c725c16c600 (on-disk)
# Checksum:0f1bb9b142e9e75002ab8e912940c94fa88ad2d4a3efe9ca4b280c725c16c600 (in-memory)
# Device size 256059448832, offset 16777216.
# Device /dev/sda1 READ lock released.
# Only 2 active CPUs detected, PBKDF threads decreased from 4 to 2.
# Not enough physical memory detected, PBKDF max memory decreased from 1048576kB to 239198kB.
# PBKDF argon2i, hash sha256, time_ms 2000 (iterations 0), max_memory_kb 239198, parallel_threads 2.
# Activating volume luks-1f5cedfe-9262-4c75-b19f-b2f03bbec3fc using token -1.
# File descriptor passphrase entry requested.
# Activating volume luks-1f5cedfe-9262-4c75-b19f-b2f03bbec3fc [keyslot -1] using passphrase.
# dm version   [ opencount flush ]   [16384] (*1)
# dm versions   [ opencount flush ]   [16384] (*1)
# Detected dm-ioctl version 4.37.0.
# Detected dm-crypt version 1.18.1.
# Device-mapper backend running with UDEV support disabled.
# dm status luks-1f5cedfe-9262-4c75-b19f-b2f03bbec3fc  [ opencount noflush ]   [16384] (*1)
# Keyslot 0 priority 1 != 2 (required), skipped.
# Keyslot 1 priority 1 != 2 (required), skipped.
# Trying to open LUKS2 keyslot 0.
# Keyslot 0 (luks2) open failed with -12.
# Releasing crypt device /dev/sda1 context.
# Releasing device-mapper backend.
# Unlocking memory.
Command failed with code -3 (out of memory).
#
#  
# free
              total        used        free      shared  buff/cache   available
Mem:         478396      137564      289864        3552       50968      304460
Swap:             0           0           0
#
```

ja-pa, 2021-03-30:
Version 2.3.5 is in master see https://github.com/openwrt/packages/commit/c644917ca7e822234b27ade566534f6c0a14ed35
Maybe this upstream issue is related to your problem https://gitlab.com/cryptsetup/cryptsetup/-/issues/488

mmokrejs, 2021-04-01:
Thank you, will test the package once it is appears in available updates. sounds like you found the right cause. Here is my case:

```
# cryptsetup luksDump /dev/sda1 
LUKS header information
Version:       	2
Epoch:         	4
Metadata area: 	16384 [bytes]
Keyslots area: 	16744448 [bytes]
UUID:          	1f5cedfe-9262-4c75-b19f-b2f03bbec3fc
Label:         	(no label)
Subsystem:     	(no subsystem)
Flags:       	(no flags)

Data segments:
  0: crypt
	offset: 16777216 [bytes]
	length: (whole device)
	cipher: aes-xts-plain64
	sector: 512 [bytes]

Keyslots:
  0: luks2
	Key:        512 bits
	Priority:   normal
	Cipher:     aes-xts-plain64
	Cipher key: 512 bits
	PBKDF:      argon2i
	Time cost:  7
	Memory:     1048576
	Threads:    4
	Salt:       
	            
	AF stripes: 4000
	AF hash:    sha256
	Area offset:32768 [bytes]
	Area length:258048 [bytes]
	Digest ID:  0
  1: luks2
	Key:        512 bits
	Priority:   normal
	Cipher:     aes-xts-plain64
	Cipher key: 512 bits
	PBKDF:      argon2i
	Time cost:  7
	Memory:     1048576
	Threads:    4
	Salt:       
	            
	AF stripes: 4000
	AF hash:    sha256
	Area offset:290816 [bytes]
	Area length:258048 [bytes]
	Digest ID:  0
Tokens:
Digests:
  0: pbkdf2
	Hash:       sha256
	Iterations: 107084
	Salt:       
	            
	Digest:
```

Likewise, I cannot convert the key on the ARM router"

```
# cryptsetup --debug -d key luksConvertKey -S1 --pbkdf-memory 131072 --pbkdf pbkdf2 /dev/sda1`
# Running command luksConvertKey.
# Locking memory.
[... 189 lines cut ...]
	Key:        512 bits
	Priority:   normal
	Cipher:     aes-xts-plain64
	Cipher key: 512 bits
	PBKDF:      pbkdf2
	Hash:       sha256
	Iterations: 2226276
	Salt:       
	            
	AF stripes: 4000
	AF hash:    sha256
	Area offset:290816 [bytes]
	Area length:258048 [bytes]
	Digest ID:  0
Tokens:
Digests:
  0: pbkdf2
	Hash:       sha256
	Iterations: 107084
	Salt:       
	            
	Digest:     
```

After that, I could execute the `cryptsetup -d key luksOpen /dev/sda1 luks-1f5cedfe-9262-4c75-b19f-b2f03bbec3fc` command and indeed, after a while of crunching the `/dev/mapper/luks-1f5cedfe-9262-4c75-b19f-b2f03bbec3fc` device showed up.

neheb, 2021-04-13:
I assume this is fixed?

mmokrejs, 2021-04-24:
I don't think so. I think openwrt should either adjust the defaults or at least document how to config the tools for low-mem device. See `pbkdf2` and `pbkdf2i` algorithms, if memory serves me right. The latter might be less hungry. Please check that yourself.

And regarding the package version bump ... i did not get a binary package at that time offered .... is it rolled out already?