openwrt/packages #15313: Ocserv configuration item under master branch is not working!!

zxlhhyccc, 2021-03-30:
[image]

ptpt52, 2021-03-30:
same issue report here.

ptpt52, 2021-03-30:
fixup by:
```
diff --git a/libs/libseccomp/Makefile b/libs/libseccomp/Makefile
index 355841df4..e82ec751b 100644
--- a/libs/libseccomp/Makefile
+++ b/libs/libseccomp/Makefile
@@ -56,7 +56,6 @@ endef
 define Package/libseccomp
 $(call Package/libseccomp/Default)
   TITLE+= (library)
-  DEPENDS+= @!arc
 endef
 
 define Package/scmp_sys_resolver

```

ptpt52, 2021-03-30:
@dangowrt  this changes https://github.com/openwrt/packages/commit/b29e609701987072fbd991a9ffc203103f99b943

dangowrt, 2021-03-30:
https://github.com/openwrt/packages/commit/b29e609701987072fbd991a9ffc203103f99b943 reflects the reality that libseccomp doesn't build on arc and is hence not available. Complicating this more with additionally depending on `@!arm||BROKEN` would probably not make things better.

Please try this:
```diff
diff --git a/net/ocserv/Config.in b/net/ocserv/Config.in
index b6460cd8e..cf86bef17 100644
--- a/net/ocserv/Config.in
+++ b/net/ocserv/Config.in
@@ -9,6 +9,7 @@ config OCSERV_PAM
 
 config OCSERV_SECCOMP
 	bool "enable seccomp"
+	depends on !arc
 	default n
 
 config OCSERV_RADIUS
```

zxlhhyccc, 2021-03-31:
> [b29e609](https://github.com/openwrt/packages/commit/b29e609701987072fbd991a9ffc203103f99b943) reflects the reality that libseccomp doesn't build on arc and is hence not available. Complicating this more with additionally depending on `@!arm||BROKEN` would probably not make things better.
> 
> Please try this:
> 
> ```diff
> diff --git a/net/ocserv/Config.in b/net/ocserv/Config.in
> index b6460cd8e..cf86bef17 100644
> --- a/net/ocserv/Config.in
> +++ b/net/ocserv/Config.in
> @@ -9,6 +9,7 @@ config OCSERV_PAM
>  
>  config OCSERV_SECCOMP
>  	bool "enable seccomp"
> +	depends on !arc
>  	default n
>  
>  config OCSERV_RADIUS
> ```

Trying this method is still the same problem

PR #15422 (2021-04-15): ocserv: don't build with libseccomp on ARC
Maintainer: @nmav 
Compile tested: tested with `make menuconfig`, with targets mvebu and ARC
Run tested: none

Description:
libseccomp can't be built on ARC, so we must disable the option here as
well.  A different fix was first proposed by @zxlhhyccc in #15377.

Fixes: #15313
commit ee1a3654 (2021-04-15): ocserv: don't build with libseccomp on ARC

libseccomp can't be built on ARC, so we must disable the option here as
well.  A different fix was first proposed by @zxlhhyccc in #15377.

Fixes: #15313
closed 2021-04-15 by cotequeiroz