openwrt/packages #15322: https-dns-proxy: force_dns requires service restart to function

rhester72, 2021-04-01:
Maintainer: @\stangri
Environment: aarch64, r16357-f8669c174e (Raspberry Pi 4 snapshot as of 2021-03-29)

Description:

After reboot, the force_dns function of https-dns-proxy does not function properly.  Instead of redirecting port 53 DNS requests to the local dnsmasq instance, requests instead time out.  Issuing "service https-dns-proxy restart" restores expected behavior (and answered public requests to f.e. Google DNS properly redirected to dnsmasq) until the next reboot.  This is likely due to a race condition at boot, but I cannot determine specifically what is happening out of sequence.

Relevant lines from /etc/config/https-dns-proxy:

```
config main 'config'
        option update_dnsmasq_config '*'
        list force_dns_port '53'
        list force_dns_port '853'
        option force_dns '1'
```

stangri, 2021-04-08:
Can you capture the relevant iptables rules in the failed state?

rhester72, 2021-04-10:
Interesting.  I either did the test wrong on the dev host (and I don't think I did), or you have to have real, live networks up and running to see it.

https-dns-proxy is enabled on boot.  After boot, the process is running, but redirects aren't working.  This will persist indefinitely. 
 At this point, the iptables ruleset looks like this:

`-A zone_lan_forward -p tcp -m tcp --dport 53 -m comment --comment "!fw3: ubus:https-dns-proxy[main] rule 0" -j reject`
`-A zone_lan_forward -p udp -m udp --dport 53 -m comment --comment "!fw3: ubus:https-dns-proxy[main] rule 0" -j reject`
`-A zone_lan_forward -p tcp -m tcp --dport 853 -m comment --comment "!fw3: ubus:https-dns-proxy[main] rule 1" -j reject`
`-A zone_lan_forward -p udp -m udp --dport 853 -m comment --comment "!fw3: ubus:https-dns-proxy[main] rule 1" -j reject`

When I manually restart the https-dns-proxy service post-boot, the ruleset looks like this:

`-A zone_lan_prerouting -p tcp -m tcp --dport 53 -m comment --comment "!fw3: ubus:https-dns-proxy[main] redirect 0" -j REDIRECT --to-ports 53`
`-A zone_lan_prerouting -p udp -m udp --dport 53 -m comment --comment "!fw3: ubus:https-dns-proxy[main] redirect 0" -j REDIRECT --to-ports 53`
`-A zone_lan_forward -p tcp -m tcp --dport 853 -m comment --comment "!fw3: ubus:https-dns-proxy[main] rule 1" -j reject`
`-A zone_lan_forward -p udp -m udp --dport 853 -m comment --comment "!fw3: ubus:https-dns-proxy[main] rule 1" -j reject`

which exactly matches observed behavior.  Test case is as follows:

- Enable service to start on boot
- Reboot
- Request a *local* record from 1.1.1.1 via any LAN client (e.g. nslookup myhost.lan 1.1.1.1)
- At this point, the request will appear to time out from the client's point of view - not NXDOMAIN, just a timeout
- Restart the service and wait until the rules reload
- Request a *local* record from 1.1.1.1 via any LAN client (e.g. nslookup myhost.lan 1.1.1.1)
- Now, the request will be answered by dnsmasq with a legitimate A record that _appears_ to come from Cloudflare (but of course cannot by definition)

I'm not sure why the rule is defined as a reject on boot and changes to a redirect on restart - the latter is the expected behavior in all cases.

stangri, 2021-04-10:
Thank you for additional testing, it's clear what the problem is now, I'll fix it this weekend.

rhester72, 2021-04-10:
No problem and thank you!

stangri, 2021-04-10:
Hey, if you don't mind testing something for me, it would be greatly appreciated!

The reason it works the way it is, is that https-dns-proxy starts before dnsmasq and if it can't detect dnsmasq's port 53 open, it blocks requests sent to that port rather than hijack then. I can fix that by always hijacking (rather than blocking) requests to port 53, but I'd also like to address the issue of the service starting too soon.

If you can open the `/etc/init.d/https-dns-proxy` file on your router, there's a line number 7 which is: `START=80`, can you experiment with pushing the number towards 90 (90 is the highest allowed value, anything above 90 may lead to the service starting very early as far as I understand) and rebooting and seeing if the https-dns-proxy would detect dnsmasq then?

Thanks!

rhester72, 2021-04-11:
This may be a slight challenge, only because I can't seem to reproduce this specific sequence reliably on my (disconnected) dev host, only my live-production-my-family-will-kill-me-if-there's-a-long-outage one.  =)  I'll attempt to do so in the wee hours of the morning tomorrow if I can manage.  ;)

rhester72, 2021-04-11:
OK - I iterated from 81 to 91 (just to see what would happen), and got the same result on every boot - always the reject rule.

stangri, 2021-04-11:
Thanks for testing, for an immediate fix, change the line 113 from:
```sh
			if netstat -tuln | grep 'LISTEN' | grep ":${c}" >/dev/null 2>&1; then
```
to
```sh
			if netstat -tuln | grep 'LISTEN' | grep ":${c}" >/dev/null 2>&1 || [ "$c" = "53" ]; then
```

I'll send PRs for 19.07, 21.02 and master, but it may take a few days for buildbots to update the IPKs.

stangri, 2021-04-11:
Oh, please let me know when you test this, so I can submit PRs.

rhester72, 2021-04-11:
Confirmed it fixes the boot issue, but I'm not sure if it created an unintended side effect.

If I do this:

service dnsmasq stop
service https-dns-proxy restart

would you expect the rules to be REDIRECT or reject at that point?

stangri, 2021-04-11:
redirect.

rhester72, 2021-04-11:
All good then :)  Thanks again!

stangri, 2021-04-20:
@aparcar @jow- as I don't have merge rights for packages, I can't close this.

closed 2021-04-20 by aparcar