openwrt/packages #15356: mpd-mini + alsa-lib: failed to open ALSA device

Vladdrako, 2021-04-06:
Device: ASUS RT-N14U (mt7620)
Maintainer: @thess @tripolar 
<pre>
root@RT-N14U:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux 5.4.109 ehci_hcd EHCI Host Controller
Bus 002 Device 002: ID 1b3f:2008 GeneralPlus USB Audio Device
Bus 002 Device 001: ID 1d6b:0001 Linux 5.4.109 ohci_hcd Generic Platform OHCI controller
root@RT-N14U:~# lsusb -v -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/1p, 12M
    ID 1d6b:0001
    |__ Port 1: Dev 2, If 0, Class=, Driver=snd-usb-audio, 12M
        ID 1b3f:2008
    |__ Port 1: Dev 2, If 1, Class=, Driver=snd-usb-audio, 12M
        ID 1b3f:2008
    |__ Port 1: Dev 2, If 2, Class=, Driver=snd-usb-audio, 12M
        ID 1b3f:2008
    |__ Port 1: Dev 2, If 3, Class=, Driver=usbhid, 12M
        ID 1b3f:2008
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    ID 1d6b:0002
root@RT-N14U:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
root@RT-N14U:~# alsactl init
alsa-lib main.c:1014:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
Found hardware: "USB-Audio" "USB Mixer" "USB1b3f:2008" "" ""
Hardware is initialized using a generic method
root@RT-N14U:~# mpc play
RetroFM: Retro FM
[paused]  #1/1   0:00/0:00 (0%)
volume: n/a   repeat: off   random: off   single: off   consume: off
ERROR: Failed to open "My ALSA Device" (alsa); Failed to open ALSA device "hw:0,0": No such file or directory
</pre>
MPD config:
<pre>
audio_output {
	type		"alsa"
	name		"My ALSA Device"
	auto_resample		"no"
#	format          "44100:16:2"
	device		"hw:0,0"	# optional
	mixer_type      "hardware"	# optional
#	mixer_device	"default"	# optional
	mixer_control	"Speaker"		# optional
#	mixer_index	"0"		# optional
}
</pre>
Previously I was on the build from 28.01.2021 and it works as expected. Same bug on 5.10 kernel.

neheb, 2021-04-08:
run alsactl init with strace

Vladdrako, 2021-04-08:
@neheb [here (broken and working)](https://gist.github.com/Vladdrako/8ea4402f6c747ee6e2121b2856abad11)

neheb, 2021-04-08:
Looked at it closer. colordiff just shows allocations to different memory locations. There are also no changes to alsa-utils or lib within your timeframe.

This is probably a kernel issue.

closed 2021-04-09 by Vladdrako