openwrt/packages #15360: ModemManager: Naming Devices (user-provided names) not working in OpenWRT

wicadmin, 2021-04-06:
Maintainer: @aleksander0m
Environment: x86 19.07

Description:

OpenWRT wouldn't connect when this is configured. `mmcli 1.12.8`

https://sigquit.wordpress.com/2016/10/06/naming-devices-in-modemmanager/

```
root@OpenWRT:~# cat 78-mm-naming.rules 
ACTION!="add|change|move", GOTO="mm_naming_rules_end"
DEVPATH=="/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4",ENV{ID_MM_PHYSDEV_UID}="em7455"
LABEL="mm_naming_rules_end"
```

```
config interface 'wan1'
	option proto 'modemmanager'
	option device 'em7455'
	option apn 'r.ispsn'
	option delegate '0'
	option iptype 'ipv4'
```

`mmcli -m em7455` works which is all I want to use this for but OpenWRT wouldn't connect is the side effect. Also tried (keeping it the default way) but didn't work:

```
config interface 'wan1'
	option proto 'modemmanager'
	option device '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4'
	option apn 'r.ispsn'
	option delegate '0'
	option iptype 'ipv4'
```

wicadmin, 2021-04-10:
@dedeckeh - I'm not sure who to tag for netifd for this. Would you happen to know the correct maintainer/dev for this? Thanks.

aleksander0m, 2021-04-10:
@wicadmin I think we should simplify this by asking the user to provide the sysfs path in the config file as well; so that:
 * If custom naming tags are not used, the single `device` config would contain the sysfs path of the modem.
 * If custom naming tags are used, the `device` config would contain the sysfs path of the modem, **and** a new `name` config would include the name of the modem.

Maybe a bit error prone? But I think that would be better than having the protocol handler read and parse the MM udev tags...

@nickberry17 what do you think?