openwrt/packages #15370: python-greenlet not compiling on 32-bit MIPS

neheb, 2021-04-08:
Maintainer: @ja-pa 
Environment: 32-bit mips

https://downloads.openwrt.org/snapshots/faillogs/mipsel_24kc_24kf/packages/python-greenlet/compile.txt

@commodo @jefferyto I tried adding PKG_USE_MIPS16:=0 but I still see -mips16 being applied to CFLAGS. Any idea what's going on?

commodo, 2021-04-08:
hmm;
will take a look

it's likely that some python setup magic is getting in the way;

commodo, 2021-04-09:
so, the problem is that Python3 is not being built with `PKG_USE_MIPS16:=0`
the CFLAGS for Python modules is taken from `./staging_dir/target-mips_24kc_musl/usr/lib/python3.9/_sysconfigdata.py`

kind of like a `pkg-config` mechanism;

if i rebuild Python with `PKG_USE_MIPS16:=0` greenlet builds fine;

this is the first package that exposes this issue like this with MIPS;
or, maybe @jefferyto knows of other cases, and i wasn't paying attention;

when @ja-pa packaged it, he disabled it for `mips` and `arc` but not for `mipsel`

[1] so, we could add `mipsel`  to https://github.com/openwrt/packages/blob/master/lang/python/python-greenlet/Makefile#L34 

[2] or, we could just add  `PKG_USE_MIPS16:=0` to Python3 altogether;
i see that other interpreters do it already (node, php, golang)

Python is also in the class of these too-big-interpreters-for-small-OpenWrt-devices anyway

between [1] & [2] I am leaning towards [2];
but I don't have a really strong opinion; we can also have [1]

neheb, 2021-04-09:
The annoying thing is that this compiles fine under GCC10.

commodo, 2021-04-10:
> The annoying thing is that this compiles fine under GCC10.

Well, annoying or not, it happens with compilers. They get better over time, and we need to wait until old ones get phased out.

If no objections, I will convert Python3 to  `PKG_USE_MIPS16:=0` next week

BKPepe, 2021-04-10:
That's up to you as this is the first package, which exposes this issue, I'd be in favor to disable ``mipsel`` in `python-greenlet` package.

commodo, 2021-04-12:
Well, there were some users of greenlet that used to package it outside of OpenWrt feeds for MIPS.
See https://github.com/wlanslovenija/firmware-packages-opkg/blob/330bc94dccd16a3e92ac2fdde08c81a598e12f94/lang/python-greenlet/Makefile

This was removed, but they were manually doing:
```
# This is required in addition to PKG_USE_MIPS16:=0 because otherwise MIPS16
# flags are inherited from the Python base package (via sysconfig module)
ifdef CONFIG_USE_MIPS16
TARGET_CFLAGS += -mno-mips16 -mno-interlink-mips16
endif
```

I guess it's the best of both words if we do it the same, with the same comment.

commodo, 2021-04-12:
so, i created PR https://github.com/openwrt/packages/pull/15404 for this

closed 2021-04-14 by neheb