openwrt/packages #15387: Exception Haserl version 0.9.36 scripts not run

DoroteoArango, 2021-04-11:
Haserl version 0.9.36 in snapshot r16464-e887049fbb

When I try run a script (I try many scripts and all run ok in others OpenWrt versions) a "haserl CGI Error Exception." is launched. 

Maintainer: @\<mstorchak > (find it by checking history of the package Makefile)
Environment: (Tp-Link Archer C20 v5, OpenWrt snapshot r16464-e887049fbb)

Description:

```
When I try to run a script (I try many scripts with this problem) a "haserl CGI Error Exception." is launched,
all scripts dont run an produce the CGI error, the same files run ok in 6 moths old snapshot OpenWRT versions.```
```

mstorchak, 2021-04-11:
Ok, I'll double-check it. In my last test it worked fine both for shell and lua scripts.

Meanwhile, could you please share more details, like a sample script that fails in your case, build options, anything else relevant?
Is this a stock build or did you build the packages yourself?

DoroteoArango, 2021-04-12:
Hello mstorchak,

Thanks a lot for your work, I share with you that in SNAPSHOT, r14850-f5c5a8abd2 all the following scripts work well.

I cant test with other routers, all my test was with TpLink Archer C20 v5.

I test with several scripts, one very simple that fail is pru.sh:
`#!/bin/ash
       content-type: text/plain

       <%# This is a sample "env" script %>
       <% env %>`

Other one that fail too is pru1.sh:
`#!/bin/ash

echo 'Hello `World'`

I install the follow packages:
DEFAULT_PACKAGES:=base-files libc libgcc busybox dropbear mtd uci netifd fstools kmod-tun libcurl librt haserl uhttpd php7 php7-cgi

DEFAULT_PACKAGES.nas:=block-mount fdisk lsblk mdadm

DEFAULT_PACKAGES.router:=dnsmasq iptables kmod-nf-nathelper odhcpd iptables-mod-nat-extra

I tryed too download the firmware from:
https://downloads.openwrt.org/snapshots/targets/ramips/mt76x8/openwrt-ramips-mt76x8-tplink_archer-c20-v5-squashfs-sysupgrade.bin

Install in my router the .bin file, so install uhttpd and haserl by opkg update, opkg install haserl, opkg install uhttpd.

I connect by wifi,and in my web browser 192.168.1.1/index.sh and throw "haserl CGI Error Exception." 

I tried too with command line.

haserl pru.sh

and result is Exception.

Best regards.

mstorchak, 2021-04-13:
@DoroteoArango , this error is somehow related to the way you run your scripts. I was able to reproduce the issue:
```
root@vortex-box:/# opkg list_installed haserl
haserl - 0.9.36-1
root@vortex-box:/# haserl --version
This is haserl version 0.9.36 (http://haserl.sourceforge.net)

root@vortex-box:/# ls -la test.cgi
-rwxr-xr-x    1 root     root            42 Apr 13 13:12 test.cgi

root@vortex-box:/# cat ./test.cgi
#!/usr/bin/haserl
<% env | grep -v SSH %>

root@vortex-box:/# haserl ./test.cgi
Exception.
```
However, if I either run the script directly or pass the full path of haserl binary, it works fine:
```
root@vortex-box:/# ./test.cgi
USER=root
HASERLVER=0.9.36
HASERL_SHELL=/bin/sh
SHLVL=4
HOME=/tmp
OLDPWD=/tmp
HASERL_UPLOAD_DIR=/tmp
PS1=\[\e]0;\u@\h: \w\a\]\u@\h:\w\$
ENV=/etc/shinit
HASERL_UPLOAD_LIMIT=0
LOGNAME=root
TERM=xterm-color
HASERL_ACCEPT_ALL=0
PATH=/usr/sbin:/usr/bin:/sbin:/bin
SHELL=/bin/ash
SESSIONID=37256075998b
PWD=/

root@vortex-box:/# /usr/bin/haserl  ./test.cgi
USER=root
HASERLVER=0.9.36
...
```
Please try to configure your webserver to either run the scripts directly, or set the handler for the scripts to the full path to haserl binary, `/usr/bin/haserl`

mstorchak, 2021-04-13:
@DoroteoArango , here's why it freaks out:
haserl-0.9.36/src/h_script.c:
```
        /* check to make sure the string following the #! matches argc[0] */
        if (memcmp (scriptbuf->buf+2, global.exec_name, strlen(global.exec_name)) == 0) {
                scriptbuf->bang_script=1;
        }
```
haserl-0.9.36/src/haserl.c:
```
  if ( scriptbuf->uid == 0 ) {
        if ( scriptbuf->bang_script > 0 ) {
                /* Then we change permissions */
                setgid (scriptbuf->gid);
                setgid (getgid ());

                setuid (scriptbuf->uid);
                setuid (getuid ());

                /* Following is from Timo Teras */
                setgroups (1, &(scriptbuf->gid));
        } else {
                die_with_message ( NULL, NULL, "Exception.");
        }

  }
```
So it looks like your `#!/bin/ash` header and root ownership don't give it a chance to pass this check. Start the scripts with `#!/usr/bin/haserl`, and they should work fine.

DoroteoArango, 2021-04-13:
@mstorchak

Thaks for your answer, you are right, I put #!/usr/bin/haserl in my scripts and all is ok.

I run my scripts with #!/bin/ash without problem for some years and now not run anymore with the same configuration, if that is correct in the OpenWrt new version I will close the issue.

Thanks a lot for your work and your answer.

closed 2021-04-13 by DoroteoArango