openwrt/packages #15391: mwan3: interface with no tracking and initial_state 'online' down since 21.02

champtar, 2021-04-11:
Maintainer: @feckert @aaronjg 
Environment: OpenWrt 21.02-SNAPSHOT r16003-226d22dcd3

Description:

I just upgraded from 19.07 to latest 21.02, so running mwan3 2.10.8-1
As I'm used to mwan3 for true multiwan (and it has been wonderfull),
I also use mwan3 on one of my router for 'static' routing, ie send lan_us via wg_us (wireguard tunnel to a us vps) ans so on ...
After boot the mwan3 interfaces are down

```
# opkg list | grep -E 'mwan3|ip-full'
ip-full - 5.11.0-3
luci-app-mwan3 - git-21.099.45423-3fff72a
mwan3 - 2.10.8-1
```

```
...
config globals 'globals'
	option mmx_mask '0x3F00'

config interface 'wg_ca'
	option enabled '0'
	option initial_state 'online'
	option family 'ipv4'

config interface 'wg_fr'
        option enabled '1'
        option initial_state 'online'
        option family 'ipv4'

config interface 'wg_us'
        option initial_state 'online'
        option family 'ipv4'
        option enabled '1'
...
```

```
# mwan3 interfaces
Interface status:
 interface wg_ca is offline and tracking is not enabled
 interface wg_fr is offline 00h:00m:00s, uptime 00h:06m:10s and tracking is not enabled
 interface wg_us is offline 00h:00m:00s, uptime 00h:06m:10s and tracking is not enabled
```

```
# logread | grep -E 'mwan3|wg_'
Sun Apr 11 02:02:56 2021 user.notice mwan3-hotplug[2554]: mwan3 hotplug on lan not called because interface disabled
Sun Apr 11 02:02:58 2021 daemon.notice netifd: Interface 'wg_fr' is setting up now
Sun Apr 11 02:02:58 2021 daemon.notice netifd: Interface 'wg_us' is setting up now
Sun Apr 11 02:03:01 2021 user.notice mwan3-hotplug[3504]: mwan3 hotplug on loopback not called because interface disabled
Sun Apr 11 02:03:01 2021 user.notice mwan3-hotplug[3721]: mwan3 hotplug on lan_iot not called because interface disabled
Sun Apr 11 02:03:02 2021 user.notice mwan3-hotplug[3829]: mwan3 hotplug on lan_fr not called because interface disabled
Sun Apr 11 02:03:03 2021 user.notice mwan3-hotplug[4074]: mwan3 hotplug on lan_us not called because interface disabled
Sun Apr 11 02:03:04 2021 daemon.notice netifd: wg_fr (2855): Try again: `REDACTED'. Trying again in 1.00 seconds...
Sun Apr 11 02:03:04 2021 daemon.notice netifd: wg_us (2856): Try again: `REDACTED'. Trying again in 1.00 seconds...
Sun Apr 11 02:03:04 2021 user.notice mwan3-hotplug[4393]: mwan3 hotplug on wan not called because interface disabled
Sun Apr 11 02:03:05 2021 daemon.debug dnsmasq[1717]: listening on wg_us(#22): 10.255.2.8 port 53
Sun Apr 11 02:03:05 2021 daemon.notice netifd: Interface 'wg_us' is now up
Sun Apr 11 02:03:05 2021 daemon.debug dnsmasq[1717]: listening on wg_fr(#23): 10.255.2.7 port 53
Sun Apr 11 02:03:05 2021 daemon.notice netifd: Interface 'wg_fr' is now up
Sun Apr 11 02:03:05 2021 daemon.notice netifd: Network device 'wg_us' link is up
Sun Apr 11 02:03:05 2021 daemon.notice netifd: Network device 'wg_fr' link is up
Sun Apr 11 02:03:05 2021 user.notice mwan3-hotplug[4653]: Execute ifup event on interface wg_us (wg_us)
Sun Apr 11 02:03:05 2021 user.warn mwan3-hotplug[4653]: failed to add 10.255.2.8 dev wg_us proto static scope link metric 201 to table 3
Sun Apr 11 02:03:05 2021 user.warn mwan3-hotplug[4653]: failed to add 10.255.8.0/24 dev wg_us proto static scope link metric 201 to table 3
Sun Apr 11 02:03:06 2021 user.info mwan3track[2043]: Detect ifup event on interface wg_us ()
Sun Apr 11 02:03:06 2021 user.notice mwan3track[2043]: Interface wg_us () is disconnecting
[... 59 lines cut ...]
192.168.3.0/24 dev br-lan_iot proto kernel scope link src 192.168.3.1 
192.168.5.0/24 dev br-lan_fr proto kernel scope link src 192.168.5.1 
192.168.6.0/24 dev br-lan_us proto kernel scope link src 192.168.6.1
Table 3: default dev wg_us proto static scope link metric 201 
10.255.2.8 dev wg_us proto static scope link metric 201 
10.255.8.0/24 dev wg_us proto static scope link metric 201 
REDACTED via REDACTED dev eth2 proto static metric 100 
REDACTED/23 dev eth2 proto static scope link metric 100 
REDACTED via REDACTED dev eth2 proto static metric 100 
192.168.2.0/24 dev br-lan proto kernel scope link src 192.168.2.1 
192.168.3.0/24 dev br-lan_iot proto kernel scope link src 192.168.3.1 
192.168.5.0/24 dev br-lan_fr proto kernel scope link src 192.168.5.1 
192.168.6.0/24 dev br-lan_us proto kernel scope link src 192.168.6.1
```

```
22: wg_us: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1420 qdisc noqueue state UNKNOWN group default qlen 1000
    link/none 
    inet 10.255.2.8/32 brd 255.255.255.255 scope global wg_us
       valid_lft forever preferred_lft forever
23: wg_fr: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1420 qdisc noqueue state UNKNOWN group default qlen 1000
    link/none 
    inet 10.255.2.7/32 brd 255.255.255.255 scope global wg_fr
       valid_lft forever preferred_lft forever
```

feckert, 2021-04-19:
I am sorry to say that I have not considered that there can also be interfaces that do not have a tracking set.
I can now remember this was possible in the last version! My usecase is always tracking on the interface!
This is not meant to be an excuse that I/we have forgotten this.
That was not the intention! I have to fix this. @champtar Could you test my changes?

I would also like to bring @aaronjg  on board, who implemented the improvements and the changes for procd.

Since the conversion to procd, tracking is started for each interface section in the mwan3 uci configuration.
https://github.com/openwrt/packages/blob/master/net/mwan3/files/etc/init.d/mwan3#L38

For this to work again, the config option **inital_state** of the interface must be set to **online** and the actions that mwan3track does with a USR2 must be executed.
https://github.com/openwrt/packages/blob/master/net/mwan3/files/etc/hotplug.d/iface/15-mwan3#L72
https://github.com/openwrt/packages/blob/master/net/mwan3/files/usr/sbin/mwan3track#L155
https://github.com/openwrt/packages/blob/master/net/mwan3/files/usr/sbin/mwan3track#L118

I still have to think about how I/we will implement this now!

feckert, 2021-04-19:
@champtar The quickest thing I can think of now is that we send a `procd_send_signal mwan3 "track_$INTERACE"` USR2 after starting the tracking on the next line https://github.com/openwrt/packages/blob/master/net/mwan3/files/etc/init.d/mwan3#L24

Since this is a shell script for the tracker, we have to insert a sleep, otherwise I think it will receive the signal even though it has not yet installed a signal handler. This is not nice, but to get the old behavior back we have to start somewhere.

You also must check if there interface does have a tracking IP set and if not do not send the signal. 
Then leave the work to the tracker and the hotplug scripts. So this workaround only works for interfaces that does not have a tracking IP set.

champtar, 2021-04-19:
@feckert I'll be happy to test any changes and thanks for the detailed explanation.
There is no rush, if I have time I'll test your idea soonish, but I can live with having to set the interfaces manually up for now, or I could set tracking to the other end of the vpn tunnel

Without looking at the code, a way to implement "no tracking" is just a tracker that set the initial_state and either sleep forever or exit without error ?

aaronjg, 2021-04-20:
> I am sorry to say that I have not considered that there can also be interfaces that do not have a tracking set.
> I can now remember this was possible in the last version! 

Yes, you are absolutely right. If no track_ips were set, it would not start a tracking process:
https://github.com/openwrt/packages/blob/openwrt-19.07/net/mwan3/files/lib/mwan3/mwan3.sh#L677

I missed this case when I ported things to use procd. I will work on an update to replicate the old behavior, which would be to just leave the status as whatever the initial_state was.

PR #15537 (2021-05-06): mwan3: allow interfaces with no tracking IPs
Maintainer: me, @feckert 
Compile tested: Not needed
Run tested: 19.07.7

In the procd refactor, support for interfaces with no tracking IPs was
inadvertentiy removed. This commit restores the previous behavior

fixes #15391
closed 2021-05-06 by feckert