openwrt/packages #15419: Add LIBCurl support to zabbix-agentd

diasdmhub, 2021-04-13:
Zabbix Agent was not compiled with LIBCurl support.

```
# zabbix_agentd -t net.tcp.service[https]
net.tcp.service[https]                        [m|ZBX_NOTSUPPORTED] [Support for HTTPS check was not compiled in.]
```

According to [Zabbix documentation "net.tcp.service" checks](https://www.zabbix.com/documentation/current/manual/appendix/items/service_check_details), this key "uses (and only works with) libcurl".

Please, add LIBCurl support to zabbix-agentd to allow it to perform "net.tcp.service" and "net.tcp.service.perf" monitoring of host HTTPS service.

fidelix, 2024-06-06:
Honestly without this the proxy becomes basically useless.
The proxy as it is has no snmp support either, so it can't do much.

diasdmhub, 2024-06-07:
The agent is supposed to be installed on endpoint devices, such as routers. They can have Web interfaces that allow localhost monitoring. If the interface uses SSL, LIBCurl support is required for the agent to monitor it.
A proxy wouldn't be necessary in such cases, and it's not recommended for devices with limited resources.

fidelix, 2024-06-07:
@diasdmhub the proxy is barely useful without libcurl.

Using a proxy is the best way to have a centralized zabbix instance monitor devices in different networks. Currently the openwrt package for the proxy has no support for snmp nor ipmi nor curl.

The proxy in passive mode is very lightweight, especially now in zabbix 7 because it can run in-memory (to avoid writes to disk) or with sqlite3.

diasdmhub, 2024-06-07:
I see. Let's hope Zabbix packages are updated soon.