openwrt/packages #15434: mwan3: Ping -I <interface>: ping: sendto: Permission denied

brianjmurrell, 2021-04-15:
Maintainer: @feckert 
Environment: 19.07.07 x86_64
Description: When using interface with `ping -I` ping gets an EPERM, but works with an IP address.

```
# ping -I pppoe-wan1 2001:4860:4860::8844
PING 2001:4860:4860::8844 (2001:4860:4860::8844): 56 data bytes
ping: sendto: Permission denied

# ping -I 2607:abcd:8006::2fe7 2001:4860:4860::8844
PING 2001:4860:4860::8844 (2001:4860:4860::8844) from 2607:abcd:8006::2fe7: 56 data bytes
64 bytes from 2001:4860:4860::8844: seq=0 ttl=118 time=12.070 ms
64 bytes from 2001:4860:4860::8844: seq=1 ttl=118 time=12.113 ms
64 bytes from 2001:4860:4860::8844: seq=2 ttl=118 time=11.783 ms
64 bytes from 2001:4860:4860::8844: seq=3 ttl=118 time=11.662 ms
64 bytes from 2001:4860:4860::8844: seq=4 ttl=118 time=12.152 ms
^C
--- 2001:4860:4860::8844 ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 11.662/11.956/12.152 ms
```

And I verified it's using the correct interface in the latter case.

luizluca, 2021-04-15:
Please share your IPv6 routing tables (ip -6 route show table all) and ip rules (ip -6 rules) and interface addresses (ip -6 add). If you mask your addresses (for privacy and security reasons), keep the change consistent amongst command outputs.

brianjmurrell, 2021-04-15:
```
# ip -6 route show table all
2001:123:aa:123::/64 dev 6in4-henet table 3 proto static metric 10 pref medium
2001:123:ab:123::/64 dev br-lan table 3 proto static metric 1024 pref medium
2607:abcd:931c:7200::/64 dev br-lan table 3 proto static metric 1024 pref medium
fd31:aeb1:48df::/64 dev br-lan table 3 proto static metric 1024 pref medium
fd8c:5604:c29c:c562::/64 dev tun0 table 3 proto kernel metric 256 pref medium
fe80::/10 dev pppoe-wan1 table 3 metric 1 pref medium
fe80::/10 dev pppoe-wan1 table 3 proto kernel metric 256 pref medium
default dev 6in4-henet table 3 metric 10 pref medium
2001:123:aa:123::/64 dev 6in4-henet table 5 proto static metric 10 pref medium
2001:123:ab:123::/64 dev br-lan table 5 proto static metric 1024 pref medium
2607:abcd:931c:7200::/64 dev br-lan table 5 proto static metric 1024 pref medium
fd31:aeb1:48df::/64 dev br-lan table 5 proto static metric 1024 pref medium
fd8c:5604:c29c:c562::/64 dev tun0 table 5 proto kernel metric 256 pref medium
fe80::/10 dev pppoe-wan1 table 5 metric 1 pref medium
fe80::/10 dev pppoe-wan1 table 5 proto kernel metric 256 pref medium
default via fe80::200:ff:fe00:0 dev pppoe-wan1 table 5 metric 20 pref medium
default from 2001:123:aa:123::/64 dev 6in4-henet proto static metric 10 pref medium
default from 2001:123:ab:123::/64 dev 6in4-henet proto static metric 10 pref medium
default from 2607:abcd:8006::50d3 via fe80::200:ff:fe00:0 dev pppoe-wan1 proto static metric 512 pref medium
default from 2607:abcd:931c:7200::/56 via fe80::200:ff:fe00:0 dev pppoe-wan1 proto static metric 512 pref medium
2001:123:aa:123::/64 dev 6in4-henet proto static metric 10 pref medium
2001:123:ab:123::/64 dev br-lan proto static metric 1024 pref medium
unreachable 2001:123:ab:123::/64 dev lo proto static metric 2147483647 error 4294967183 pref medium
2607:abcd:931c:7200::/64 dev br-lan proto static metric 1024 pref medium
unreachable 2607:abcd:931c:7200::/56 dev lo proto static metric 2147483647 error 4294967183 pref medium
fd31:aeb1:48df::/64 dev br-lan proto static metric 1024 pref medium
unreachable fd31:aeb1:48df::/48 dev lo proto static metric 2147483647 error 4294967183 pref medium
fd8c:5604:c29c:c562::/64 dev tun0 proto kernel metric 256 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
fe80::/64 dev eth0.3 proto kernel metric 256 pref medium
fe80::/64 dev br-lan proto kernel metric 256 pref medium
fe80::/64 dev br-guest proto kernel metric 256 pref medium
fe80::/64 dev eth0.2 proto kernel metric 256 pref medium
fe80::/64 dev 6in4-henet proto kernel metric 256 pref medium
fe80::/64 dev tun0 proto kernel metric 256 pref medium
fe80::/64 dev ifb4eth0.2 proto kernel metric 256 pref medium
fe80::/10 dev pppoe-wan1 metric 1 pref medium
fe80::/10 dev pppoe-wan1 proto kernel metric 256 pref medium
local ::1 dev lo table local proto kernel metric 0 pref medium
anycast 2001:123:aa:123:: dev 6in4-henet table local proto kernel metric 0 pref medium
local 2001:123:aa:123::2 dev 6in4-henet table local proto kernel metric 0 pref medium
anycast 2001:123:ab:123:: dev br-lan table local proto kernel metric 0 pref medium
local 2001:123:ab:123::1 dev br-lan table local proto kernel metric 0 pref medium
local 2607:abcd:8006::50d3 dev pppoe-wan1 table local proto kernel metric 0 pref medium
anycast 2607:abcd:931c:6f00:: dev br-lan table local proto kernel metric 0 pref medium
local 2607:abcd:931c:6f00::1 dev br-lan table local proto kernel metric 0 pref medium
anycast 2607:abcd:931c:7200:: dev br-lan table local proto kernel metric 0 pref medium
local 2607:abcd:931c:7200::1 dev br-lan table local proto kernel metric 0 pref medium
anycast 2607:abcd:9876:5432:: dev br-lan table local proto kernel metric 0 pref medium
local 2607:abcd:9876:5432::1 dev br-lan table local proto kernel metric 0 pref medium
anycast fd31:aeb1:48df:: dev br-lan table local proto kernel metric 0 pref medium
local fd31:aeb1:48df::1 dev br-lan table local proto kernel metric 0 pref medium
anycast fd8c:5604:c29c:c562:: dev tun0 table local proto kernel metric 0 pref medium
local fd8c:5604:c29c:c562::1 dev tun0 table local proto kernel metric 0 pref medium
anycast fe80:: dev br-lan table local proto kernel metric 0 pref medium
anycast fe80:: dev eth0.3 table local proto kernel metric 0 pref medium
anycast fe80:: dev eth0 table local proto kernel metric 0 pref medium
anycast fe80:: dev br-guest table local proto kernel metric 0 pref medium
anycast fe80:: dev eth0.2 table local proto kernel metric 0 pref medium
anycast fe80:: dev 6in4-henet table local proto kernel metric 0 pref medium
anycast fe80:: dev tun0 table local proto kernel metric 0 pref medium
anycast fe80:: dev pppoe-wan1 table local proto kernel metric 0 pref medium
anycast fe80:: dev ifb4eth0.2 table local proto kernel metric 0 pref medium
local fe80::8717:33bb dev 6in4-henet table local proto kernel metric 0 pref medium
local fe80::1890:7e95:a31f:fff1 dev tun0 table local proto kernel metric 0 pref medium
local fe80::6eb0:ceff:fef5:1e4b dev eth0.2 table local proto kernel metric 0 pref medium
local fe80::92fb:a6ff:fe4c:25c dev br-lan table local proto kernel metric 0 pref medium
local fe80::92fb:a6ff:fe4c:25c dev eth0.3 table local proto kernel metric 0 pref medium
[... 60 lines cut ...]
10: eth0.3@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 fe80::92fb:a6ff:fe4c:25c/64 scope link 
       valid_lft forever preferred_lft forever
[... cut ...]
       valid_lft forever preferred_lft forever
84: pppoe-wan1: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1492 state UNKNOWN qlen 3
    inet6 2607:abcd:8006::50d3/128 scope global dynamic noprefixroute 
       valid_lft 80335sec preferred_lft 80335sec
    inet6 fe80::fc02:e00f:7e3e:7d47/10 scope link 
       valid_lft forever preferred_lft forever
107: ifb4eth0.2: <BROADCAST,NOARP,UP,LOWER_UP> mtu 1500 state UNKNOWN qlen 32
    inet6 fe80::bc09:42ff:fecb:963e/64 scope link 
       valid_lft forever preferred_lft forever
```

luizluca, 2021-04-15:
These are default routes:

```
default from 2001:123:aa:123::/64 dev 6in4-henet proto static metric 10 pref medium
default from 2001:123:ab:123::/64 dev 6in4-henet proto static metric 10 pref medium
default from 2607:abcd:8006::50d3 via fe80::200:ff:fe00:0 dev pppoe-wan1 proto static metric 512 pref medium
default from 2607:abcd:931c:7200::/56 via fe80::200:ff:fe00:0 dev pppoe-wan1 proto static metric 512 pref medium

default dev 6in4-henet table 3 metric 10 pref medium

default via fe80::200:ff:fe00:0 dev pppoe-wan1 table 5 metric 20 pref medium

```

Table 3 and 5 were created by mwan3. Notice that it removes the "from filter" as the original prefix might not match the outgoing interface selected. Those extra tables are selected by:

```
1003:	from all iif 6in4-henet lookup 3 
1005:	from all iif pppoe-wan1 lookup 5 
2003:	from all fwmark 0x300/0x3f00 lookup 3 
2005:	from all fwmark 0x500/0x3f00 lookup 5 
32766:	from all lookup main 
```

As iif does not matter for traffic originated by the router, only fwmark will be in use or it will look up main table (router without mentioning "table") . fwmark is set by "ip6table -t mangle" rules created by mwan3. I'm not sure how fwmark and "ping6 -I" will interact but fwmark might not matter here. I'm not sure wman3 is playing a part here. Does the problem happen if mwan3 is disabled?

An IPv6 /128 interface (like your pppoe-wan1) is a tricky one. Even when you specify Linux to use it as outgoing interface, Linux prefer a different source address, normally one from br-lan. It will only use that /128 address if you use it in -I arg.
My guess is that "ping6 -I pppoe-wan1" is selecting an address from br-lan that uses the prefix from 6in4-henet, like 2001:123:ab:123::1. There is no route to dns.google (default) for a packet that uses 2001:123:ab:123::1 as source address and outgoing interface as pppoe-wan1. It might not even generate a packet that tcpdump could catch. You can check if that is true.

You can try to manually hint Linux about the wanted outgoing address for "default from .. dev pppoe-wan1" router adding a "src <br-lan ip using a pppoe-wan1 prefix>" option using "ip route". Alternatively, you can temporary remove ipv6 address from br-lan that uses the prefix from 6in4-henet using "ip address remove". Although both are volatile configurations and they will be lost when interface is reloaded, it will diagnose the problem.

If that is the case, I'm not sure what would be the best solution for hinting the kernel about the correct outgoing address. You can add the hint in a ifup script as your network settings are not static.

brianjmurrell, 2021-04-15:
> It might not even generate a packet that tcpdump could catch. You can check if that is true.

Does not seem to.

> Alternatively, you can temporary remove ipv6 address from br-lan that uses the prefix from 6in4-henet using "ip address remove".

That did not resolve the issue.

feckert, 2021-04-17:
Mwan3 is not so well equipped for IPv6. Almost everything has been tested and designed for IPv4 only. In the next stable, a lot was done for IPv6.

HikariWS, 2021-04-20:
I'm having the same issue. When I execute `ip -6 route add default dev pppoe-want` the permission denied error is fixed and ping works, but when network is restarted or OS is rebooted it comes back.

I'm currently using openwrt-21.02-snapshot-r16046 on AMD64 and will upgrade to RC1 when its build is available.

luizluca, 2021-04-20:
> I'm having the same issue. When I execute `ip -6 route add default dev pppoe-want` the permission denied error is fixed and ping works, but when network is restarted or OS is rebooted it comes back.

It would fix pppoe-want but it would break the other wan you might have. Without from field, it would be used unconditionally and, without metric, it will have precedence.

A "ping -I otherwan" might work if that interface have a non /128 address but it might break "ping -I otherwanIpv6".

I'm still betting the issue is with source address selection. Adding a source to each default route might help.

luizluca, 2021-04-20:
It would be nice if we had a gns3 VM (or alike) with all this variant cases to let mwan3 maintainers test independently from what isp they have.

I could list these IPv6 wan variants:
* Static
* Slaac
* dhcpv6 with /64 (also without prefix delegation), /112 and /128
* Pppoe
* 6in4

And addresses translations with MASQ and NETMAP.

jamesmacwhite, 2021-04-25:
Have you tried setting a static IPv6 route for each IPv6 interface with a target of ::/0?

This helped my IPv6 usage with mwan3.

https://forum.openwrt.org/t/ping-and-traceroute-failing-for-eth0-3-on-ipv6/44680/11

HikariWS, 2021-05-01:
> It would fix pppoe-want but it would break the other wan you might have. Without from field, it would be used unconditionally and, without metric, it will have precedence.
>
> A "ping -I otherwan" [...]

I didn't know that, that's sad :/

Notice this isn't a mwan3 specific issue, it's a routing issue. It happens even without mwan3 installed.

All cases I had seen happen with pppoe, so I guess the cause might be related to pppoe, maybe how a IPv6 DHCPv6 interface is used to provide IPv6 connectivity for connections made on pppoe.

Another hint, in my previous router I had ISP1 misconfigured, I had both wan6 from wan6 manual config and wan_6 from wan's config. And the permission denied happened. Now on my new router I fixed it and I only have want and want6 interfaces, and it still happens.

So, it's not related to a automatic interface being created to provide IPv6 under pppoe, but it may be related to a virtual/alias (automatic or manual) interface for pppoe.

If this is indeed related to pppoe, replicating it on VM won't work, as I believe no VM manager supports pppoe connection. BUT, a few months ago I stopped using VMware's vnat and started using a VM with OpenWRT to route VMs into LAN. That provides IPv6 connectivity, which VMware's vnat seems to not support, and allows to test and learn.

If OpenWRT or any other route OS that runs on AMD64 is able to serve pppoe on a simple ethernet port, we might be able to build a system with 2 VMs and replicate it.

feckert, 2021-09-03:
So this is not a mwan3 issue I close it

brianjmurrell, 2021-09-03:
> So this is not a mwan3 issue I close it

To be clear, if this is not a mwan3 issue, what issue is it and where should the ticket for that be opened so that it can be fixed.

brianjmurrell, 2021-09-08:
> @brianjmurrell I can open the ticket again if you like.

Well, that's not the actual goal if the problem isn't actually in mwan3.

There seem to be a number (or at least one) of people in this thread that are positing that this is not an mwan3 issue and that this is a problem even without mwan3.

There have been suggestions to routing table changes that are supposed to make the problem go away also, but frankly I'm not sure I ever got any of those to work.  But I could be just mis-recalling.

What I am suggesting is that if somebody here feels like this is a plain old routing issue on OpenWRT and knows what the solution is, opening a generic OpenWRT ticket to get that fixed would be the best way to go.

I most certainly would do it but I don't think I am understanding what the underlying routing issue is well enough to formulate a useful ticket.