openwrt/packages #15437: node: Compile error from WRT32x on 21.02 branch.

zdz2019, 2021-04-16:
/usr/bin/ld: /home/test/wrt/openwrt-life/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.16.1/out/Release/obj.host/v8_base_without_compiler/deps/v8/src/regexp/regexp-compiler-tonode.o: in function `v8::internal::CharacterRange::AddCaseEquivalents(v8::internal::Isolate*, v8::internal::Zone*, v8::internal::ZoneList<v8::internal::CharacterRange>*, bool)':
regexp-compiler-tonode.cc:(.text._ZN2v88internal14CharacterRange18AddCaseEquivalentsEPNS0_7IsolateEPNS0_4ZoneEPNS0_8ZoneListIS1_EEb+0x58): undefined reference to `icu_68::UnicodeSet::UnicodeSet()'
/usr/bin/ld: regexp-compiler-tonode.cc:(.text._ZN2v88internal14CharacterRange18AddCaseEquivalentsEPNS0_7IsolateEPNS0_4ZoneEPNS0_8ZoneListIS1_EEb+0x12a): undefined reference to `icu_68::UnicodeSet::add(int, int)'
/usr/bin/ld: regexp-compiler-tonode.cc:(.text._ZN2v88internal14CharacterRange18AddCaseEquivalentsEPNS0_7IsolateEPNS0_4ZoneEPNS0_8ZoneListIS1_EEb+0x15e): undefined reference to `icu_68::UnicodeSet::UnicodeSet(icu_68::UnicodeSet const&)'
/usr/bin/ld: regexp-compiler-tonode.cc:(.text._ZN2v88internal14CharacterRange18AddCaseEquivalentsEPNS0_7IsolateEPNS0_4ZoneEPNS0_8ZoneListIS1_EEb+0x16c): undefined reference to `icu_68::UnicodeSet::removeAll(icu_68::UnicodeSet const&)'
/usr/bin/ld: regexp-compiler-tonode.cc:(.text._ZN2v88internal14CharacterRange18AddCaseEquivalentsEPNS0_7IsolateEPNS0_4ZoneEPNS0_8ZoneListIS1_EEb+0x176): undefined reference to `icu_68::UnicodeSet::closeOver(int)'
/usr/bin/ld: regexp-compiler-tonode.cc:(.text._ZN2v88internal14CharacterRange18AddCaseEquivalentsEPNS0_7IsolateEPNS0_4ZoneEPNS0_8ZoneListIS1_EEb+0x184): undefined reference to `icu_68::UnicodeSet::removeAll(icu_68::UnicodeSet const&)'
/usr/bin/ld: regexp-compiler-tonode.cc:(.text._ZN2v88internal14CharacterRange18AddCaseEquivalentsEPNS0_7IsolateEPNS0_4ZoneEPNS0_8ZoneListIS1_EEb+0x18f): undefined reference to `icu_68::UnicodeSet::removeAll(icu_68::UnicodeSet const&)'
/usr/bin/ld: regexp-compiler-tonode.cc:(.text._ZN2v88internal14CharacterRange18AddCaseEquivalentsEPNS0_7IsolateEPNS0_4ZoneEPNS0_8ZoneListIS1_EEb+0x1d3): undefined reference to `icu_68::UnicodeSet::getRangeCount() const'
/usr/bin/ld: regexp-compiler-tonode.cc:(.text._ZN2v88internal14CharacterRange18AddCaseEquivalentsEPNS0_7IsolateEPNS0_4ZoneEPNS0_8ZoneListIS1_EEb+0x1e4): undefined reference to `icu_68::UnicodeSet::getRangeStart(int) const'
/usr/bin/ld: regexp-compiler-tonode.cc:(.text._ZN2v88internal14CharacterRange18AddCaseEquivalentsEPNS0_7IsolateEPNS0_4ZoneEPNS0_8ZoneListIS1_EEb+0x1f3): undefined reference to `icu_68::UnicodeSet::getRangeEnd(int) const'
/usr/bin/ld: regexp-compiler-tonode.cc:(.text._ZN2v88internal14CharacterRange18AddCaseEquivalentsEPNS0_7IsolateEPNS0_4ZoneEPNS0_8ZoneListIS1_EEb+0x222): undefined reference to `icu_68::UnicodeSet::~UnicodeSet()'
/usr/bin/ld: regexp-compiler-tonode.cc:(.text._ZN2v88internal14CharacterRange18AddCaseEquivalentsEPNS0_7IsolateEPNS0_4ZoneEPNS0_8ZoneListIS1_EEb+0x22a): undefined reference to `icu_68::UnicodeSet::~UnicodeSet()'
/usr/bin/ld: regexp-compiler-tonode.cc:(.text._ZN2v88internal14CharacterRange18AddCaseEquivalentsEPNS0_7IsolateEPNS0_4ZoneEPNS0_8ZoneListIS1_EEb+0x29a): undefined reference to `icu_68::UnicodeSet::add(int, int)'
/usr/bin/ld: /home/test/wrt/openwrt-life/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.16.1/out/Release/obj.host/v8_base_without_compiler/deps/v8/src/strings/char-predicates.o: in function `v8::internal::IsIdentifierStartSlow(int)':
char-predicates.cc:(.text._ZN2v88internal21IsIdentifierStartSlowEi+0x1e): undefined reference to `u_hasBinaryProperty_68'
/usr/bin/ld: /home/test/wrt/openwrt-life/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.16.1/out/Release/obj.host/v8_base_without_compiler/deps/v8/src/strings/char-predicates.o: in function `v8::internal::IsIdentifierPartSlow(int)':
char-predicates.cc:(.text._ZN2v88internal20IsIdentifierPartSlowEi+0x1e): undefined reference to `u_hasBinaryProperty_68'
/usr/bin/ld: /home/test/wrt/openwrt-life/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.16.1/out/Release/obj.host/v8_base_without_compiler/deps/v8/src/strings/char-predicates.o: in function `v8::internal::IsWhiteSpaceSlow(int)':
char-predicates.cc:(.text._ZN2v88internal16IsWhiteSpaceSlowEi+0x1c): undefined reference to `u_charType_68'
/usr/bin/ld: /home/test/wrt/openwrt-life/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.16.1/out/Release/obj.host/v8_base_without_compiler/gen/src/regexp/special-case.o: in function `v8::internal::BuildSpecialAddSet()':
special-case.cc:(.text._ZN2v88internal18BuildSpecialAddSetEv+0x1c): undefined reference to `icu_68::UnicodeSet::UnicodeSet()'
/usr/bin/ld: special-case.cc:(.text._ZN2v88internal18BuildSpecialAddSetEv+0x26): undefined reference to `icu_68::UnicodeSet::add(int)'
/usr/bin/ld: special-case.cc:(.text._ZN2v88internal18BuildSpecialAddSetEv+0x30): undefined reference to `icu_68::UnicodeSet::add(int)'
/usr/bin/ld: special-case.cc:(.text._ZN2v88internal18BuildSpecialAddSetEv+0x3a): undefined reference to `icu_68::UnicodeSet::add(int)'
/usr/bin/ld: special-case.cc:(.text._ZN2v88internal18BuildSpecialAddSetEv+0x44): undefined reference to `icu_68::UnicodeSet::add(int)'
/usr/bin/ld: special-case.cc:(.text._ZN2v88internal18BuildSpecialAddSetEv+0x51): undefined reference to `icu_68::UnicodeSet::add(int)'
/usr/bin/ld: /home/test/wrt/openwrt-life/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.16.1/out/Release/obj.host/v8_base_without_compiler/gen/src/regexp/special-case.o:special-case.cc:(.text._ZN2v88internal18BuildSpecialAddSetEv+0x5e): more undefined references to `icu_68::UnicodeSet::add(int)' follow
/usr/bin/ld: /home/test/wrt/openwrt-life/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.16.1/out/Release/obj.host/v8_base_without_compiler/gen/src/regexp/special-case.o: in function `v8::internal::BuildSpecialAddSet()':
special-case.cc:(.text._ZN2v88internal18BuildSpecialAddSetEv+0xa7): undefined reference to `icu_68::UnicodeSet::freeze()'
/usr/bin/ld: /home/test/wrt/openwrt-life/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.16.1/out/Release/obj.host/v8_base_without_compiler/gen/src/regexp/special-case.o: in function `v8::internal::BuildIgnoreSet()':
special-case.cc:(.text._ZN2v88internal14BuildIgnoreSetEv+0x1c): undefined reference to `icu_68::UnicodeSet::UnicodeSet()'
/usr/bin/ld: special-case.cc:(.text._ZN2v88internal14BuildIgnoreSetEv+0x29): undefined reference to `icu_68::UnicodeSet::add(int)'
/usr/bin/ld: special-case.cc:(.text._ZN2v88internal14BuildIgnoreSetEv+0x36): undefined reference to `icu_68::UnicodeSet::add(int)'
/usr/bin/ld: special-case.cc:(.text._ZN2v88internal14BuildIgnoreSetEv+0x43): undefined reference to `icu_68::UnicodeSet::add(int)'
/usr/bin/ld: special-case.cc:(.text._ZN2v88internal14BuildIgnoreSetEv+0x50): undefined reference to `icu_68::UnicodeSet::add(int)'
/usr/bin/ld: special-case.cc:(.text._ZN2v88internal14BuildIgnoreSetEv+0x5d): undefined reference to `icu_68::UnicodeSet::add(int)'
/usr/bin/ld: /home/test/wrt/openwrt-life/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.16.1/out/Release/obj.host/v8_base_without_compiler/gen/src/regexp/special-case.o:special-case.cc:(.text._ZN2v88internal14BuildIgnoreSetEv+0x6a): more undefined references to `icu_68::UnicodeSet::add(int)' follow
/usr/bin/ld: /home/test/wrt/openwrt-life/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.16.1/out/Release/obj.host/v8_base_without_compiler/gen/src/regexp/special-case.o: in function `v8::internal::BuildIgnoreSet()':
special-case.cc:(.text._ZN2v88internal14BuildIgnoreSetEv+0x7d): undefined reference to `icu_68::UnicodeSet::add(int, int)'
/usr/bin/ld: special-case.cc:(.text._ZN2v88internal14BuildIgnoreSetEv+0x8a): undefined reference to `icu_68::UnicodeSet::add(int)'
/usr/bin/ld: special-case.cc:(.text._ZN2v88internal14BuildIgnoreSetEv+0x97): undefined reference to `icu_68::UnicodeSet::add(int)'
/usr/bin/ld: special-case.cc:(.text._ZN2v88internal14BuildIgnoreSetEv+0xa4): undefined reference to `icu_68::UnicodeSet::add(int)'
/usr/bin/ld: special-case.cc:(.text._ZN2v88internal14BuildIgnoreSetEv+0xb1): undefined reference to `icu_68::UnicodeSet::add(int)'
/usr/bin/ld: special-case.cc:(.text._ZN2v88internal14BuildIgnoreSetEv+0xbe): undefined reference to `icu_68::UnicodeSet::add(int)'
[... cut ...]
/usr/bin/ld: /home/test/wrt/openwrt-life/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.16.1/out/Release/obj.host/v8_base_without_compiler/gen/src/regexp/special-case.o: in function `v8::internal::BuildIgnoreSet()':
special-case.cc:(.text._ZN2v88internal14BuildIgnoreSetEv+0x105): undefined reference to `icu_68::UnicodeSet::add(int, int)'
/usr/bin/ld: special-case.cc:(.text._ZN2v88internal14BuildIgnoreSetEv+0x118): undefined reference to `icu_68::UnicodeSet::add(int, int)'
/usr/bin/ld: special-case.cc:(.text._ZN2v88internal14BuildIgnoreSetEv+0x120): undefined reference to `icu_68::UnicodeSet::freeze()'
collect2: error: ld returned 1 exit status

neheb, 2021-04-17:
ping @nxhack

nxhack, 2021-04-18:
Hi @zdz2019 

Are you also building ICU packages of OpenWrt packages? 

The workaround is to either remove the ICU package altogether or to select system-icu on the node.js package

zdz2019, 2021-04-19:
Thank you! I tried to set it to CONFIG_NODEJS_ICU_NONE, the compilation is normal.
But why should it be set to NODEJS_ICU_SMALL is default?
@nxhack

nxhack, 2021-04-19:
This change is a result of the arguments set out below.

https://github.com/openwrt/packages/pull/15005
https://github.com/openwrt/packages/pull/15392

The reason behind this is that the default configuration of node.js is 'full-icu', i.e. the full set of icu4c, and this has led to an increase in code that assumes the existence of 'Intl' objects.

If there are too many issues, we'll look at reverting to the previous version, but I will try to find out if there's a way around the errors you've described.

nxhack, 2021-04-20:
A workaround has been found. I will update the package as soon as I finish testing.  The test will take some time.

zdz2019, 2021-05-04:
@neheb 
that's good!

closed 2021-05-04 by zdz2019