openwrt/packages #15446: strongswan: swanctl init script doesn't load connections

stintel, 2021-04-18:
Maintainer: @pprindeville @Thermi
Environment: OpenWrt master r16565-37958f0d115 on octeon (Ubiquiti EdgeRouter Lite)

Description:
The swanctl init script doesn't load connections defined in /etc/swanctl/conf.d/. The file /etc/swanctl/conf.d/ contains `include conf.d/*.conf`, and running `swanctl --load-all` manually works fine. The deprecated ipsec script is disabled.

```
# /etc/init.d/swanctl restart
# swanctl -L
#
# swanctl --load-all
loaded certificate from '/etc/swanctl/x509/ar1.crt'
loaded ECDSA key from '/etc/swanctl/private/ar1.key'
no authorities found, 0 unloaded
no pools found, 0 unloaded
loaded connection 'esk-be'
successfully loaded 1 connections, 0 unloaded
root@ar1.sof.bg.adlevio.net:~# swanctl -L
ar0: IKEv2, no reauthentication, rekeying every 14400s
  local:  %any
  remote: [HIDDEN]
  local public key authentication:
    id: ar1
    certs:
  remote public key authentication:
    id: vr0
  ar0-1: TUNNEL, rekeying every 3600s
    local:  0.0.0.0/0 ::/0
    remote: 0.0.0.0/0 ::/0
```

stintel, 2021-04-18:
A while ago when I was trying to implement swanctl support myself, I also had this issue. The solution I came up with back then was https://github.com/stintel/openwrt-packages/commit/b489d87f645023194670cad5957e10f8d5c5a0c7, but it's quite hackish.

stintel, 2021-04-19:
> Did you have a preexisting `/etc/strongswan.conf` file?
```
# cat /etc/strongswan.conf
# strongswan.conf - strongSwan configuration file
#
# Refer to the strongswan.conf(5) manpage for details
#
# Configuration changes should be made in the included files

charon {
        load_modular = yes
        plugins {
                include strongswan.d/charon/*.conf
        }

        install_routes = no
}

include strongswan.d/*.conf
```

pprindeville, 2021-04-19:
Hmm...

```
root@OpenWrt2:~# grep '^include' /etc/swanctl/swanctl.conf 
include conf.d/*.conf
include /var/swanctl/swanctl.conf
root@OpenWrt2:~# 
```

I don't see anything in `/etc/init.d/swanctl` that requires UCI for it to launch.

Thermi, 2021-04-19:
> Hmm...
> 
> ```
> root@OpenWrt2:~# grep '^include' /etc/swanctl/swanctl.conf 
> include conf.d/*.conf
> include /var/swanctl/swanctl.conf
> root@OpenWrt2:~# 
> ```
> 
> I don't see anything in `/etc/init.d/swanctl` that requires UCI for it to launch.

The issue here is that /var/swanctl/strongswan.conf isn't included by default for some reason.

Thermi, 2021-04-19:
Because the generated config files are ephemeral and storing them in /var guarantees that they do not conflict with user defined files, packaged files, or are included in a tarball of the configuration (which would be redundant because they are a product of a script and the UCI config that is in in /etc).

The swanctl.init script generates a strongswan.conf file to set the options necessary for the correct functioning of the generated configs. It also generates a swanctl.conf file that contains the generated tunnel configurations.

On the test machine the inclusion worked fine. We now have to figure out why it doesn't work for you.

I propose just packaging a very short config file in swanctl/conf.d and strongswan.d that include the corresponding file in /var.

stintel, 2021-04-19:
`/var/swanctl/strongswan.conf` doesn't exist on my system.
`/var/swanctl/swanctl.conf` does, but it's empty.

Thermi, 2021-04-20:
@stintel Please check for opkg-new files.

stintel, 2021-04-20:
> @stintel Please check for opkg-new files.

There aren't any related to strongSwan
```
/etc/config/vallumd-opkg
/etc/keepalived/keepalived.conf-opkg
/overlay/upper/etc/config/vallumd-opkg
/overlay/upper/etc/keepalived/keepalived.conf-opkg
/rom/usr/lib/opkg/info/luci-app-opkg.control
/rom/usr/lib/opkg/info/luci-app-opkg.files-sha256sum
/rom/usr/lib/opkg/info/luci-app-opkg.list
/rom/usr/lib/opkg/info/luci-app-opkg.prerm
/rom/usr/share/luci/menu.d/luci-app-opkg.json
/rom/usr/share/rpcd/acl.d/luci-app-opkg.json
/usr/lib/opkg/info/luci-app-opkg.control
/usr/lib/opkg/info/luci-app-opkg.files-sha256sum
/usr/lib/opkg/info/luci-app-opkg.list
/usr/lib/opkg/info/luci-app-opkg.prerm
/usr/share/luci/menu.d/luci-app-opkg.json
/usr/share/rpcd/acl.d/luci-app-opkg.json
```

Thermi, 2021-04-20:
Weird.

With -5, the includes were just appended to the config files. See https://github.com/openwrt/packages/commit/643df01275798dc4a44d33ac1a8b630c91a80c8c.

pprindeville, 2021-04-20:
What does the tail of your `/etc/swanctl/swanctl.conf` file look like?  It should have:

```
...
# Include config snippets
include conf.d/*.conf

include /var/swanctl/swanctl.conf
```

stintel, 2021-04-20:
```
# tail /etc/swanctl/swanctl.conf
        # cert_uri_base =

    # }

# }

# Include config snippets
include conf.d/*.conf

include /var/swanctl/swanctl.conf
```

Either way, not relevant as `/var/swanctl/swanctl.conf` is empty anyway.

Thermi, 2021-04-20:
It *is* relevant because that file will contain the config generated by the swanctl init script.

Thermi, 2021-04-20:
It is relevant as far as we can see that at the very least this file was installed correctly.

stintel, 2021-05-04:
Any ideas? As you're deprecating the old ipsec script you're actively suggesting people to migrate from something that works to something that is broken...

Thermi, 2021-05-05:
The package contains the correct strongswan config files to include the ones generated by the script, so your setup is the odd one. On our test machine the issue does not occur. Please try to reproduce it.

Also, the reason the old script is getting deprecated is because upstream is removing the ipsec/stroke config backend in version 6.0.0, which is relatively close. With version 6.0.0, the old script won't work as intended anymore because the generated ipsec.conf file won't be read anymore.
Son it's not our decision. And I have no leverage on the issue.

stintel, 2021-05-05:
Forcing people to migrate to UCI is not acceptable. If the swanctl script does not load connections configured in /etc/swanctl/swanctl.conf (or files included from there), that is a bug. This combined with the fact that the ipsec init script now actively tells people it is deprecated *will* result in people losing access to devices in the field. This is never acceptable, and especially not in the current times with travel restrictions. I've stressed this multiple times when I was still maintaining the package.

The fact that strongSwan 6.0.0 will deprecate ipsec entirely doesn't change anything to this. While I agree that something needs to be done before strongSwan 6.0.0 arrives, the risk of people losing access to devices should be avoided at all cost.

Thermi, 2021-05-05:
It does load these configs. That is why there is an include there.

stintel, 2021-05-05:
> It does load these configs. That is why there is an include there.

No, it does not.
```
# tail -n4 /etc/swanctl/swanctl.conf
# Include config snippets
include conf.d/*.conf

include /var/swanctl/swanctl.conf
```
None of the connections in the files matched by the first include are loaded by the init script. I have to manually run `swanctl --load-all` and they appear. This confirms the configs on their own are fine, but the init script for some reason does not call `swanctl --load-all`.

Thermi, 2021-05-05:
That is because for some reason, despite the packaged file contains the include for the generated strongswan.conf in /var, the file that you have on your system does not have the include. It appears to be the old version. And you stated you have no opkg-new file for it either.
That is why I asked you to reproduce the issue.

By default strongswan does not load any swanctl configs. It has to made to so either by using the init daemon (charon-systemd and system work fine 
like that because charon-systemd tells systemd when it's ready and then it can execute the command to tell it to do so.
That does not work reliably with any other init daemon because none of them implements anything that could be used to notify it of system daemon readiness (e.g. charon/charon-systemd of strongswan). In these cases charon.start-scripts needs to be used to tell it to load the configs. The file 
generated by the new swanctl init script does that. It contains the relevant lines. So if it is set up, all configs are loaded.

Please try to reproduce the issue.

>     It does load these configs. That is why there is an include there.
> 
> No, it does not.
> 
> |# tail -n4 /etc/swanctl/swanctl.conf # Include config snippets include 
conf.d/*.conf include /var/swanctl/swanctl.conf |
> 
> None of the connections in the files matched by the first include are loaded by the init script. I have to manually run |swanctl --load-all| and 
they appear. This confirms the configs on their own are fine, but the init script for some reason does not call |swanctl --load-all|.

stintel, 2021-05-05:
Just remove /etc/config/ipsec.

Thermi, 2021-05-05:
What do you mean with that?

stintel, 2021-05-05:
I'm assuming loading the connections is handled by https://github.com/openwrt/packages/blob/master/net/strongswan/files/swanctl.init#L555-L557

This is part of [the config_ipsec() function](https://github.com/openwrt/packages/blob/master/net/strongswan/files) which is called [here](https://github.com/openwrt/packages/blob/master/net/strongswan/files/swanctl.init#L570). When /etc/config/ipsec does not exist, the init script will not run config_ipsec at all so it will not append that start-script snippet to the included config file and `swanctl --load-all` will not be executed.

stintel, 2021-05-05:
Additionally I was lacking `include /var/ipsec/strongswan.conf` in `/etc/strongswan.conf`. Why does it need to be appended to the file if the default already has `include strongswan.d/*.conf`.

Imo it would be better to just ship a file  /etc/strongswan.d/start-scripts.conf or so in strongswan-swanctl that contains:
```
charon {
  start-scripts {
    load-all = /usr/sbin/swanctl --load-all --noprompt
  }
```
This would avoid appending in the Makefile, and reduce the appending in the init script, for a much cleaner solution imo.

stintel, 2021-05-05:
I gave up maintainership of this package because I no longer have the energy to devote much time on this. I'm glad you took over that role, but please, do a better job. The swanctl init script was rushed, imo, and it shows.

Thermi, 2021-05-05:
> Additionally I was lacking `include /var/ipsec/strongswan.conf` in `/etc/strongswan.conf`. Why does it need to be appended to the file if the default already has `include strongswan.d/*.conf`.
>
> Imo [...]

No, that'd change the behaviour of the daemon even if the service was not used and it would cause unnecessary opkg-new files to be created when the package is updated and the user edited them.
Also /etc is not for runtime generated files. That's what /var is for. That is why this is done the way it is done.
And it was not rushed. We took much time to deliberate on the details.

pprindeville, 2021-05-05:
> I'm assuming loading the connections is handled by https://github.com/openwrt/packages/blob/master/net/strongswan/files/swanctl.init#L555-L557
>
> This is part of [the config_ipsec() [...]

Please try out PR #15575.

stintel, 2021-05-05:
> No, that'd change the behaviour of the daemon even if the service was not used and it would cause unnecessary opkg-new files to be created when the package is updated and the user edited them.
> Also [...]

If you ship the file with the package it wouldn't be runtime generated. And you could either exclude the file from conffiles, or add a header that says the file is not to be edited. That would solve the -opkg file creation.

As for the changing behaviour of the daemon, I think an unconditional load-all is exactly what happens on systemd-based systems. I don't see why this would be a problem. If a user doesn't want the service he can disable it or not install it at all.

pprindeville, 2021-05-07:
Resolved with PR #15575.

closed 2021-05-07 by pprindeville