openwrt/packages #15447: BIND 9.17.x: This is an unstable development release

protogen, 2021-04-18:
Maintainer: @nmeyerhans

Description:
The entire BIND 9.17 branch is an **unstable** development release.  The latest stable version of BIND is 9.16.13 (see references below).  Please use the latest **stable** version.

https://bind9.readthedocs.io/en/latest/notes.html#introduction
https://www.isc.org/download/ (click on BIND 9 half way down)
https://en.wikipedia.org/wiki/BIND

Note - BIND 9.17.10 and 9.17.11 do not work properly. Upon start-up, they immediately log numerous errors and some functions fail.

nmeyerhans, 2021-04-18:
>  The entire BIND 9.17 branch is an unstable development release. The latest stable version of BIND is 9.16.13 (see references below). Please use the latest stable version.

The "*unstable* development release" is packaged in the development branch of OpenWRT packages. Stable packages branches that contain an unstable bind release will be updated to the the stable bind branch once it's made. Stable packages branches that already contain a stable bind branch will remain on that bind branch, but will be updated to the latest release from that branch as it comes available (and as the packaging work is done.) In other words, the `openwrt-19.07` branch will remain on ISC's 9.16.x bind branch, and the `openwrt-21.02` branch will contain 9.17.x until a stable 9.18 release is made upstream, at which point it will switch to following that release stream.

> Note - BIND 9.17.10 and 9.17.11 do not work properly. Upon start-up, they immediately log numerous errors and some functions fail.

Please report issues either here or with ISC. Otherwise they never get fixed. Note that 9.17.11 works find in daily use here.

closed 2021-04-18 by nmeyerhans

protogen, 2021-04-18:
@nmeyerhans

Thank you for the information / clarification.

I'm actually trying to fix this in [Entware](https://github.com/Entware/Entware/issues/639).  It appears their stable branch may be pulling from your development branch (therefore I'm getting an unstable version of BIND).

protogen, 2021-04-19:
Just to close the loop on this one - the problem here was my own ignorance!  Entware **always** sync with an OpenWrt **dev** feed.  I wasn't aware of this.  My fault.

Sorry for the confusion.   I hope I didn't waste too much of your time. :-)

nmeyerhans, 2021-04-19:
Thanks for closing the loop. If you *do* encounter issues that may be related to the software or the package, please do report them. Like I said, they won't get fixed if they're not reported. (Obviously the issues could be specific to Entware, in which case there may not be much I can do to help, but they could also be with bind.)

protogen, 2021-04-20:
I will report the issues with BIND 9.17.10/11 once I've had a chance to investigate them properly.  I'm just not sure when that will be.  I immediately rolled back to BIND 9.16.8 last weekend because I ran out of time.