openwrt/packages #15450: luajit: host build fails

milani, 2021-04-19:
Maintainer: @milani (myself)
Environment: Openwrt 19.07.7, MIPS MT 7620, HOST: ubuntu 20.04

Description:

I have a private package that depends on luajit host. It used to work on LEDE 17.01. I tried to compile it using the latest openwrt and I get this error:

```bash
$ make package/milani/luvi/{clean,prepare,compile} V=sc

SHELL= flock ./openwrt/tmp/.LuaJIT-2.1.0-beta3.tar.gz.flock -c '      ./openwrt/scripts/download.pl "./openwrt/dl" "LuaJIT-2.1.0-beta3.tar.gz" "1ad2e34b111c802f9d0cdf019e986909123237a28c746b21295b63c9e785d9c3" "" "https://luajit.org/download"    '
. ./openwrt/include/shell.sh; gzip -dc ./openwrt/dl/LuaJIT-2.1.0-beta3.tar.gz | tar -C ./openwrt/build_dir/hostpkg/luajit-2.1.0-beta3/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* ./openwrt/build_dir/hostpkg/luajit-2.1.0-beta3

Applying ./patches/010-lua-path.patch using plaintext: 
can't find file to patch at input line 3
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- a/src/luaconf.h
|+++ b/src/luaconf.h
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
Patch failed!  Please fix ./patches/010-lua-path.patch!
make[2]: *** [Makefile:94: ./openwrt/build_dir/hostpkg/luajit-2.1.0-beta3/.preparede05e848a17dbc478f763c32492e526f0_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[2]: Leaving directory './openwrt/feeds/packages/lang/luajit'
time: package/feeds/packages/luajit/host-compile#0.25#0.06#0.26
    ERROR: package/feeds/packages/luajit [host] failed to build.

```

Here is the relevant part of my private package:
```
include $(TOPDIR)/rules.mk
  
PKG_NAME:=luvi
PKG_RELEASE:=1

PKG_SOURCE_PROTO:=git
PKG_SOURCE_URL:=git://github.com/luvit/luvi.git
PKG_SOURCE_VERSION:=v2.7.6
PKG_SOURCE_DATE:=2016-12-16

PKG_LICENSE:=Apache License v2.0
PKG_LICENSE_FILES:=LICENSE

PKG_BUILD_DEPENDS:=luajit/host

include $(INCLUDE_DIR)/package.mk
include $(INCLUDE_DIR)/cmake.mk

define Package/luvi
 SECTION:=libs
 CATEGORY:=Morteza
 TITLE:=Luvi
 URL:=https://github.com/luvit/luvi
 MAINTAINER:=Morteza Milani <mrtz.milani@gmail.com>
 DEPENDS:=+luajit +luv +libpcre +libopenssl +zlib +libuv
endef

define Package/luvi/description
 Luvi is a project to work with bundled luvit applications
endef

$(eval $(call BuildPackage,luvi))
```

Is there anything changed in the way OpenWRT builds host packages? Is something missing here?

milani, 2021-04-19:
@neheb would love to have your feedback here.

neheb, 2021-04-19:
I assume setting PKG_BUILD_DIR would work.

neheb, 2021-04-19:
There's also HOST_BUILD_DIR.

closed 2025-11-14 by GeorgeSapkin