openwrt/packages #15504: dnscrypt-proxy2: default config is blocking legitimate sites

ipha, 2021-04-23:
Maintainer: @BKPepe 
Environment: x86, qemu, 19.07.7

Description:
After banging my head against an API that refused to connect at home, I discovered that openwrt is changing the default behavior of dnscrypt-proxy and filtering DNS queries.

A DNS server shouldn't be filtering queries unless configured to do so or advertised as as ad-blocking solution. Additionally, the example blocklist provided by upstream is very crude and I don't believe is intended to be used as-is.

jamesmacwhite, 2021-04-24:
This is interesting, as I recent tried out `dnscrypt-proxy2`. Some domain queries were just failing outright, I first thought it was adblock but confirming the domain had no matching rules it then led me to DNSCrypt. In the end I decided to go with Stubby and do DNS-over-TLS, but it is interesting it wasn't just me experiencing this. I didn't really look into it too much in the end.

BKPepe, 2021-05-07:
On the same day, when I received the report from you, I cherry-picked the update of **dnscrypt-proxy2** from `openwrt-21-02` and `master` branch to `openwrt-19.07 together with changes in blocking websites. There are fewer sites right now, so it should be better.

Reference:
- https://github.com/openwrt/packages/commit/139217f8a62bb2f6c35746650d38d445d9e00f42
- https://github.com/openwrt/packages/commit/3412c7a4564dcc9e819340bb2d610dc6dcde3138

In any case, if you are willing to be a maintainer of this package, feel free to submit a pull request as I am not longer interested in it as I am not using it. Also, as always, pull requests are welcome.

closed 2021-05-07 by BKPepe