openwrt/packages #15506: mosquitto init.d config issue with log_type and port

colinw-smartrg, 2021-04-24:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @karlp
Environment: x86, 64, 21.02

Description:
uci config file conversion issues in the main mosquitto.mosquitto seciton.

log_type is wrongly converted to log_dest
This breaks mosquitto running.

port is deprecated in mosquitto config, but is not option in the uci config.
It is recommended to use the listener block instead.
In some cases having port in the config will cause mosquitto to not start due to binding issues.

My local fixes for these issues:
```
diff --git a/net/mosquitto/files/etc/init.d/mosquitto b/net/mosquitto/files/etc/init.d/mosquitto
index b42fa88..603b988 100755
--- a/net/mosquitto/files/etc/init.d/mosquitto
+++ b/net/mosquitto/files/etc/init.d/mosquitto
@@ -63,8 +63,7 @@ convert_mosq_general() {
                echo "bind_address localhost" >> $TCONF
        fi
 
-       config_get port "$1" port 1883
-       echo "port $port" >> $TCONF
+       append_if "$1" port
        append_if "$1" acl_file
        append_optional_bool "$1" allow_anonymous
        append_optional_bool "$1" allow_duplicate_messages
@@ -75,7 +74,7 @@ convert_mosq_general() {
        append_optional_bool "$1" log_timestamp
        config_get log_types "$1" log_types
        for log_type in $log_types; do
-               echo "log_dest $log_type" >> $TCONF
+               echo "log_type $log_type" >> $TCONF
        done
        append_if "$1" max_inflight_bytes
        append_if "$1" max_inflight_messages

```

karlp, 2021-04-24:
Looks good, will take care of this monday

karlp, 2021-04-26:
Thanks for this, it's now applied to 1907 (log type only) and 2102 and master (both port and log type)

closed 2021-04-26 by karlp