openwrt/packages #15512: samba4: /etc/init.d/samba4 bogus io_uring check

Hello71, 2021-04-25:
#14463 causes a file "2" to be created every time `/etc/init.d/samba4 start` (or restart) is run. however, this check is bogus in the first place. openwrt/lede releases are built with CONFIG_KERNEL_KALLSYMS=n, so /proc/kallsyms doesn't even exist.

Hello71, 2021-04-25:
@Andy2244, @hnyman

Andy2244, 2021-04-26:
Ah thanks missed that it does not work on `SMALL_FLASH`, which cant install samba4 anyway because of its size? Not sure if extroot works on small devices?
```
config KERNEL_KALLSYMS
	bool "Compile the kernel with symbol table information"
	default y if !SMALL_FLASH
```
So normal devices can use `/proc/kallsyms` which i was testing with.

Yes the `>>2` is a actual error and will be fixed in the next version.

Hello71, 2021-04-26:
it is `!SMALL_FLASH`, but also buildbot sets CONFIG_KERNEL_KALLSYMS=n.

Andy2244, 2021-04-26:
> buildbot sets CONFIG_KERNEL_KALLSYMS=n

@hnyman Why is buildbot doing this? This way my own local dev builds, do not match what is "released" via buildbot?

hnyman, 2021-04-26:
Note that the final release configs are different (smaller) than the development snapshots config in buildbot.

To my knowledge, the KALLSYMS are normally set (for development snapshots and private builds), but the release config has them removed in order to optimize the final release images, which are not meant for detailed debugging.

Andy2244, 2021-04-26:
> but the release config has them removed in order to optimize the final release images, which are not meant for detailed debugging.

Have to test myself, how much bytes `CONFIG_KERNEL_KALLSYMS=n` actually saves, but guess i have no way to runtime detect the kernel feature than and need to rely on the static data or just try&error at startup.

PR #15577 (2021-05-06): samba4: update to 4.13.8
Maintainer: me
Compile tested: arm/master
Run tested: arm/mvebu/master 

Description:
* update to 4.13.8
* remove faulty io_uring kernel detection
* fixes CVE's: CVE-2020-27840, CVE-2021-20277, CVE-2020-27840, CVE-2021-20277, CVE-2020-27840, CVE-2021-20277, CVE-2021-20254
* resolves #15512
commit 93b34d4d (2021-05-06): samba4: update to 4.13.8

* update to 4.13.8
* remove faulty io_uring kernel detection
* fixes CVE's: CVE-2020-27840, CVE-2021-20277, CVE-2020-27840, CVE-2021-20277, CVE-2020-27840, CVE-2021-20277, CVE-2021-20254
* resolves #15512
commit 29ad9f1e (2021-05-07): samba4: update to 4.13.8

* update to 4.13.8
* remove faulty io_uring kernel detection
* fixes CVE's: CVE-2020-27840, CVE-2021-20277, CVE-2020-27840, CVE-2021-20277, CVE-2020-27840, CVE-2021-20277, CVE-2021-20254
* resolves #15512

(cherry picked from commit 93b34d4ddad2768c27d03a78e723b5910964aaa0)
closed 2021-05-06 by neheb