openwrt/packages #15513: https-dns-proxy: Memory leak

rhester72, 2021-04-25:
Maintainer: @stangri
Environment: aarch64, r16357-f8669c174e (Raspberry Pi 4 snapshot as of 2021-03-29)

Description:

There is a persistent and gradual memory leak in https-dns-proxy, at least using default settings:

```
config main 'config'
        option update_dnsmasq_config '*'
        list force_dns_port '53'
        list force_dns_port '853'
        option force_dns '1'

config https-dns-proxy
        option listen_addr '127.0.0.1'
        option user 'nobody'
        option group 'nogroup'
        option listen_port '5053'
        option bootstrap_dns '8.8.8.8,8.8.4.4'
        option resolver_url 'https://dns.google/dns-query'

config https-dns-proxy
        option bootstrap_dns '1.1.1.1,1.0.0.1,2606:4700:4700::1111,2606:4700:4700::1001'
        option resolver_url 'https://cloudflare-dns.com/dns-query'
        option listen_port '5054'
```

This can be pretty clearly demonstrated with a memory profile:

[image]

Usage patterns are typical of a 5-person household (2 working from home, 2 high school age children in remote learning) with standard usage patterns (video streaming, web browsing, Zoom sessions, etc.).

stangri, 2021-04-25:
@rhester72 -- which version of https-dns-proxy are you using?

@aarond10 -- Aaron, any other telemetry you'd need to look into this?

rhester72, 2021-04-25:
Version: 2021-01-17-4

aarond10, 2021-04-27:
That does seem odd. It could be fragmentation but the fact it's a straight line suggests otherwise. I'd love to know what version of libcurl you're running via something like this:

```
$ ./https_dns_proxy -vvv
[I] 1619485933.784025 main.c:194 Built Apr 16 2021 20:02:29.
[I] 1619485933.784052 main.c:195 System c-ares: 1.17.1
[I] 1619485933.784086 main.c:196 System libcurl: libcurl/7.76.0 OpenSSL/1.1.1k zlib/1.2.11 zstd/1.4.9 libidn2/2.3.0 libpsl/0.21.1 (+libidn2/2.3.0) libssh2/1.9.0 nghttp2/1.43.0
...
```

I'll run mine under valgrind for a few days and see if I can replicate. I'm looking at 4.5MiB of RAM usage after about 20 days of uptime but I haven't been tracking growth over time.

rhester72, 2021-04-27:
```
[I] 1619486619.901302 main.c:192 Built Mar 29 2021 07:29:16.
[I] 1619486619.901327 main.c:193 System c-ares: 1.17.1
[I] 1619486619.901381 main.c:194 System libcurl: libcurl/7.75.0 wolfSSL/4.7.0
```

aarond10, 2021-04-27:
A look over the 7.76 release log makes me wonder if your leak is [this](https://github.com/curl/curl/issues/6616). (In the [7.76.0 changelog](https://curl.se/changes.html). 

Can you try updating libcurl and check again?

The other curious thing is that it's a straight line. I doubt the load hitting the daemon is going to be uniform across day and night so the fact this is a linear leak suggests a walltime process -- something like SSL session renegotiation.

rhester72, 2021-04-27:
Installed libcurl4 version 7.76.0-1 and restarted https-dns-proxy, will resume monitoring and report back.

rhester72, 2021-05-01:
There's initial evidence of the same growth pattern that was seen before, but I wonder if updating libcurl4 actually made any difference.

On OpenWRT, libcurl4 depends on libwolfssl4.7.0.66253b90, which appears to be the package that would contain the fix in question.  The fix was on Feb 17 - according to SourceDateEpoch, the source elements for libworlfssl4.7.0 are from Feb 23 (and thus should be included).

I'll continue to let it run and see if the growth continues unabated, but assuming that libwolfssl4.7.0 does indeed contain this:

wolfssl: don't store a NULL sessionid

I'm not sure where else to look at this point.

aarond10, 2021-05-02:
If you're technical enough, valgrind in memcheck mode with a debug build
would point at where the memory is originally allocated and that would give
a very good idea where to look. I'm still running in that mode here. I'll
check mine later (arch, later libcurl, libssl).

rhester72, 2021-05-02:
Unfortunately the only trafficked device I have is my real production router :/

aarond10, 2021-05-02:
I just killed my valgrind instance and got this:

```
==3140225==
==3140225== HEAP SUMMARY:
==3140225==     in use at exit: 8,117 bytes in 20 blocks
==3140225==   total heap usage: 7,170,546 allocs, 7,170,526 frees,
7,354,291,978 bytes allocated
==3140225==
==3140225== LEAK SUMMARY:
==3140225==    definitely lost: 0 bytes in 0 blocks
==3140225==    indirectly lost: 0 bytes in 0 blocks
==3140225==      possibly lost: 0 bytes in 0 blocks
==3140225==    still reachable: 8,117 bytes in 20 blocks
==3140225==         suppressed: 0 bytes in 0 blocks
```
This has been running for 6 days and serving requests, so I think there's a
fair chance it's some interaction with WolfSSL and libcurl.

rhester72, 2021-05-02:
I concur - hopefully things will improve with an updated WolfSSL down the road.  Thank you for looking!

curtdept, 2021-05-29:
The version we are running in OWRT is quite old, if you want to try the new one...

New 010-ninja.patch
`
--- a/src/logging.h
+++ b/src/logging.h
@@ -36,6 +36,20 @@ enum _LogSeverity {
   LOG_MAX
 };
 
+#define STRIPPATH(s)\
+    (sizeof(s) > 2 && (s)[sizeof(s)-2] == '/' ? (s) + sizeof(s) - 1 : \
+    sizeof(s) > 3 && (s)[sizeof(s)-3] == '/' ? (s) + sizeof(s) - 2 : \
+    sizeof(s) > 4 && (s)[sizeof(s)-4] == '/' ? (s) + sizeof(s) - 3 : \
+    sizeof(s) > 5 && (s)[sizeof(s)-5] == '/' ? (s) + sizeof(s) - 4 : \
+    sizeof(s) > 6 && (s)[sizeof(s)-6] == '/' ? (s) + sizeof(s) - 5 : \
+    sizeof(s) > 7 && (s)[sizeof(s)-7] == '/' ? (s) + sizeof(s) - 6 : \
+    sizeof(s) > 8 && (s)[sizeof(s)-8] == '/' ? (s) + sizeof(s) - 7 : \
+    sizeof(s) > 9 && (s)[sizeof(s)-9] == '/' ? (s) + sizeof(s) - 8 : \
+    sizeof(s) > 10 && (s)[sizeof(s)-10] == '/' ? (s) + sizeof(s) - 9 : \
+    sizeof(s) > 11 && (s)[sizeof(s)-11] == '/' ? (s) + sizeof(s) - 10 : (s))
+
+#define __FILENAME__ STRIPPATH(__FILE__)
+
 #define DLOG(...) _log(__FILENAME__, __LINE__, LOG_DEBUG, __VA_ARGS__)
 #define ILOG(...) _log(__FILENAME__, __LINE__, LOG_INFO, __VA_ARGS__)
 #define WLOG(...) _log(__FILENAME__, __LINE__, LOG_WARNING, __VA_ARGS__)
`

New top of Makefile
`
include $(TOPDIR)/rules.mk

PKG_NAME:=https-dns-proxy
PKG_VERSION:=2021-05-14
PKG_RELEASE:=1

PKG_SOURCE_PROTO:=git
PKG_SOURCE_URL:=https://github.com/aarond10/https_dns_proxy
PKG_SOURCE_DATE:=2021-01-17
PKG_SOURCE_VERSION:=fa05dcfc46ff6d58999aab1e35bcf1b36b1eb19e
PKG_MIRROR_HASH:=4fd40c4e64010c5c49629a4de830ab6e53c649458f0adeda5399b350cbe7f319
PKG_MAINTAINER:=Stan Grishin <stangri@melmac.net>
PKG_LICENSE:=MIT
PKG_LICENSE_FILES:=LICENSE
`
`

aarond10, 2021-05-30:
That code to strip trailing '/' seems less than ideal.

stangri, 2021-05-30:
> That code to strip trailing '/' seems less than ideal.

You mean this: https://github.com/stangri/packages/blob/cf75ad2e6c5bde352dc57fd70f047e4bbe9fefe4/net/https-dns-proxy/files/https-dns-proxy.init#L190 ?

aarond10, 2021-05-30:
I mean in the comment above @curtdept mentions `010-ninja.patch` which has this cascade of sizeof().

aarond10, 2021-05-30:
The code is also incorrect -- if there is no slash and a short string, the code would read undefined memory.

stangri, 2021-05-30:
> I mean in the comment above @curtdept mentions `010-ninja.patch` which has this cascade of sizeof().

Speaking of, are you planning on switching to ninja, so that no patches will be necessary?

Would you recommend we use the regular cmake for OpenWrt builds?

curtdept, 2021-05-30:
I just ported it from what was there, im assuming its doing some kind of pathfix for the owrt structure but tbh looked a bit strange to me as well.

aarond10, 2021-06-01:
It shouldn't make a difference which is used, right? I kind of assume GNU make is the most compatible (NetBSD, etc) but haven't been following ninja close enough to know if that's not really relevant anymore. Can you elaborate a little on the ninja patch? Why is filename important to fix in debug logs?

stangri, 2021-06-01:
@neheb Rosen, can you please address @aarond10 questions about ninja make?

Andrew, Rosen is pioneering switch to ninja make in OpenWrt and has created the original patch.

neheb, 2021-06-01:
As of https://github.com/aarond10/https_dns_proxy/pull/109 , upstream fixed the issue. That patch should not be there.

I doubt it's causing issues though. That macro is a constant expression.

stangri, 2021-06-02:
@neheb thanks, I've removed the patches directory from the most recent PRs: https://github.com/openwrt/packages/pull/15726 https://github.com/openwrt/packages/pull/15727 https://github.com/openwrt/packages/pull/15728

I don't think it's the patch which is causing problems either.

aarond10, 2021-06-16:
Is this still an issue with the latest code? Have we ruled out WolfSSL bug as possible cause?

rhester72, 2021-06-18:
Haven't ruled out WolfSSL as it still hasn't been upgraded in OpenWRT.

stangri, 2021-08-18:
@rhester72 have you tried version 2021-07-29-1?

rhester72, 2021-08-19:
@stangri Is that version in the openwrt master branch?

stangri, 2021-08-19:
> @stangri Is that version in the openwrt master branch?

And 21.02 too I believe.

aarond10, 2021-08-19:
FWIW, I can't replicate this on arch linux @head with OpenSSL. 18 days of
runtime under valgrind

$ ps -p 1739632 -o lstart
                 STARTED
Mon Aug  2 08:48:05 2021

==2220068== LEAK SUMMARY:
==2220068==    definitely lost: 0 bytes in 0 blocks
==2220068==    indirectly lost: 0 bytes in 0 blocks
==2220068==      possibly lost: 0 bytes in 0 blocks
==2220068==    still reachable: 8,189 bytes in 20 blocks
==2220068==         suppressed: 0 bytes in 0 blocks

I'll check back on this later but freshly started binary is using 7MiB of
RAM now (more than I expected, but likely shared http2/curl/ssl overheads).

$ sudo cat /proc/2222392/smaps_rollup | head -n 4
5604933b5000-7ffcaf933000 ---p 00000000 00:00 0
 [rollup]
Rss:                7384 kB
Pss:                3560 kB
Pss_Anon:           1940 kB

If that doesn't grow over time, I'm not sure what to check next.

rhester72, 2021-08-20:
@aarond10 There's STRONG suspicion at this point the issue lies strictly with WolfSSL.  I haven't had a chance to check the package with the 6/21 source, but will do so over the next day or so (more correctly, I'll get it installed then and start it soaking, but it takes a number of days at minimum for the leak to truly become visible).

aarond10, 2021-08-20:
Acknowledged (WolfSSL). I'll keep a layman's eye on memory usage on my arch
install too today until it plateaus. So far:

```
09:12:33 8340 kB
10:42:47 8544 kB
11:30:00 10576 kB
12:22:57 10580 kB
13:03:34 10576 kB
```

aarond10, 2021-08-20:
Can confirm no leak here. I would have expected to see even a 4kiB
increase. Has been stable now for 4 hours (using marginally less RAM than 4
hours ago).

15:14:00 10572 kB

rhester72, 2021-08-20:
VERY preliminary data, but the angle of the RSS and VSZ graphs for me look exactly like they did before after 4 hours.  Latest WolfSSL package in 21.02 is libwolfssl4.7.0.66253b90 (same hash as the original report) with epoch of 1624308195 (considerably newer than the first report).

The master branch has a different hash (ba20a816) but an even earlier epoch of 1621775498 so I didn't even bother testing it.

Will report back after a few days of real-world use, but I'm not encouraged.  :/

stangri, 2021-08-20:
@rhester72 I'm sorry if I missed it, but which platform/router and OpenWrt version are you using? I can try building a version of curl which would depend on another core SSL library for your testing.

rhester72, 2021-08-20:
No worries - I'm running on a Pi 4, currently 21.02 RC4 (so the package in question is the aarch64_cortex-a72 variant).

stangri, 2021-08-20:
> No worries - I'm running on a Pi 4, currently 21.02 RC4 (so the package in question is the aarch64_cortex-a72 variant).

So on top of libopenssl, you'll need libcurl, libnghttp and curl IPKs from https://dev.melmac.net/repo/aarch64/. I don't think opkg will install a local libcurl unless you completely remove libcurl first. So I'd push all three packages to /tmp/, then `opkg remove libcurl curl`, then `opkg install /tmp/*.ipk`.

rhester72, 2021-08-21:
Some questions:

- I don't have curl installed at all, is there an unwritten dependency on it from https-dns-proxy?
- Why is a new libnghttp needed, as it doesn't appear to be dependent on anything besides libc?

Based on the dependency tree, from what I can see on my install, simply swapping out libcurl4 should be enough, which I'll attempt now.

rhester72, 2021-08-21:
Also, this was slightly disturbing...executing a "service https-dns-proxy stop" *uninstalled the package*.

rhester72, 2021-08-21:
Up and running against openssl and monitoring:

```
https-dns 23924 nobody  mem       REG                7,0            1140 /usr/sbin/https-dns-proxy (path dev=0,19)
https-dns 23924 nobody  mem       REG                7,0             936 /usr/lib/libcrypto.so.1.1 (path dev=0,19)
https-dns 23924 nobody  mem       REG                7,0             937 /usr/lib/libssl.so.1.1 (path dev=0,19)
https-dns 23924 nobody  mem       REG                7,0             928 /usr/lib/libnghttp2.so.14.20.1 (path dev=0,19)
https-dns 23924 nobody  mem       REG              179,2             287 /lib/libgcc_s.so.1 (path dev=0,19, inode=4294967583)
https-dns 23924 nobody  mem       REG                7,0             497 /usr/lib/libev.so.4.0.0 (path dev=0,19)
https-dns 23924 nobody  mem       REG                7,0             943 /usr/lib/libcurl.so.4.7.0 (path dev=0,19)
https-dns 23924 nobody  mem       REG                7,0             922 /usr/lib/libcares.so.2.4.2 (path dev=0,19)
https-dns 23924 nobody  mem       REG              179,2             289 /lib/libsetlbf.so (path dev=0,19, inode=4294967585)
https-dns 23924 nobody  mem       REG              179,2             285 /lib/libc.so (path dev=0,19, inode=4294967581)
```

stangri, 2021-08-21:
> * I don't have curl installed at all, is there an unwritten dependency on it from https-dns-proxy?

No, only libcurl.

> * Why is a new libnghttp needed, as it doesn't appear to be dependent on anything besides libc?

It includes support for HTTP/2 and the version of libcurl I've posted shouldn't work without it. If you install curl and run `curl --version` what does it say?

> Also, this was slightly disturbing...executing a "service https-dns-proxy stop" _uninstalled the package_.

It is disturbing, if you can reproduce this, lmk.

rhester72, 2021-08-21:
> > * Why is a new libnghttp needed, as it doesn't appear to be dependent on anything besides libc?
> 
> It includes support for HTTP/2 and the version of libcurl I've posted shouldn't work without it. If you install curl and run `curl --version` what does it say?

I think the explanation is that the appropriate version of libnghttp is already in 21.02 for support of libcurl4 7.78.  At any rate, leaving libnghttp untouched from the OpenWRT repo, all appears to be working fine by just swapping out libcurl (which automatically pulled in openssl).

> > Also, this was slightly disturbing...executing a "service https-dns-proxy stop" _uninstalled the package_.
> 
> It is disturbing, if you can reproduce this, lmk.

Not touching it for now, but will see if I can reproduce later.

rhester72, 2021-08-21:
So far, so good - RSS is variable as expected (peaking at just over 13MB in total for both processes) and VSZ stabilized at ~20MB for both processes roughly 4 hours in.

stangri, 2021-08-22:
There was also an update to the principal binary which is now compiled with debug symbols for OpenWrt, you may want to update to the latest version, it should have been built for all branches by now.

rhester72, 2021-08-22:
Updated and restarted.

aarond10, 2021-08-22:
Interestingly... Two days later, 12508 kiB. Possibly just fragmentation,
will keep you posted.

rhester72, 2021-08-25:
Still seeing (un)healthy jumps in VSZ over time, continuing to monitor...pattern isn't as steep as with wolf but it's hard to understand why it keeps wanting more.

rhester72, 2021-08-29:
I saw very little unexpected growth over a week, but was forced to restart earlier today.

At this point, I'm comfortable with the ticket being closed, as I see no evidence that this is an issue with https-dns-proxy.  It's clearly a leak in WolfSSL under certain conditions, but I don't think removing curl's dependence on wolf for OpenWRT is the 'right' solution, merely a workaround.  I will leave reporting it as an upstream wolf problem and shepherding it through resolution as an exercise for someone else as I simply lack the strength.  *LOL*

Thanks *VERY* much to all who contributed to helping track this!

stangri, 2021-08-29:
@cotequeiroz I'm not sure if there's any/enough actionable information but if you have time to look into this, seemingly WolfSSL-related issue, let us know.

@rhester72 if you could continue using libcurl-openssl at least while you're staying on 21.02.0-rc4 to conclude that it's a libcurlWolfSSL leak that'd be great.

Closing this for now as it doesn't seem to be caused by `https-dns-proxy` binary itself.

closed 2021-08-29 by stangri

aarond10, 2021-08-29:
One last data point for posterity showing no sign of growth:

```
$ ps -p 2222392 -o lstart && sudo cat /proc/2222392/smaps_rollup | head -n 4 && date
                 STARTED
Fri Aug 20 08:50:15 2021
5604933b5000-7ffcaf933000 ---p 00000000 00:00 0                          [rollup]
Rss:               12144 kB
Pss:                8210 kB
Pss_Anon:           7208 kB
Mon 30 Aug 2021 09:50:37 AEST
```

rhester72, 2021-11-12:
I am pleased to report that the issue appears corrected in the wolfssl update that was part of 21.02.1:

[image]