openwrt/packages #15541: mwan3 track interface with TOR

perachemarc, 2021-04-29:
Dear all, 

I want to track an interface connected to TOR network. On TOR, there is no support for ICMP packets; thus we need a tracking method using TCP only. It is possible to use curl for example. Do it makes sense to add a new tracking method based on curl. For example curl -s --interface   [interface] [tracking IP].

Best,
Marc

feckert, 2021-04-29:
We already support other tracking methods. Have a look at the [source](https://github.com/openwrt/packages/blob/master/net/mwan3/files/usr/sbin/mwan3track#L67).
However, these only work reliably in the master or owrt-21.02 branch, as we have a c wrapper for [interface pinning](https://github.com/openwrt/packages/blob/master/net/mwan3/src/sockopt_wrap.c) handling here.

perachemarc, 2021-04-29:
I'm currently using 21.02_rc1. I'll try asap with the master.

HikariWS, 2021-05-01:
I haven't tested yet but mwan3 supports httping, which runs over HTTP/S and works as ping counting the latency to receive the HTTP/s response. Doesn't it work for u?

perachemarc, 2021-05-01:
I tried httping, but I was not able to select a tun* interface.

perachemarc, 2021-05-01:
With this small modification in mwan3track file, 
@@ -50,6 +50,12 @@
 				return 1
 			}
 			;;
+		curl)                                                                                              
+                        which curl 1>/dev/null 2>&1 || {                                                             
+                                $LOG warn "Missing curl. Please install curl package."                              
+                                return 1                                                                              
+                        }                                                                                             
+                        ;;  
 		*)
 			$LOG warn "Unsupported tracking method: $track_method"
 			return 2
@@ -178,7 +184,10 @@
 					;;
 					nping-arp)
 						result=$(nping -e $DEVICE -c $count $track_ip --arp | grep Lost | awk '{print $12}')
-					;;
+					;;                                                                                                                                   
+                                        curl)                                                                                                                                  
+                                                curl -s --interface $DEVICE $track_ip &> /dev/null                                                                
+                                                result=$?
 				esac
 				if [ $check_quality -eq 0 ]; then
 					if [ $result -eq 0 ]; then

I was able to use curl and my problem is solved with a modified mwan3_2.8.16-1

feckert, 2023-05-03:
Issue was fixed by modifying the mwan3 source by @perachemarc.

closed 2023-05-03 by feckert