openwrt/packages #15556: shairport-sync: built without libdaemon

humaita-github, 2021-05-01:
Maintainer: @neheb 
Environment: openwrt-21.02.0-rc1-ath79-generic-tplink_tl-wr710n-v1-squashfs-sysupgrade

Description:

Cannot run shairport-sync as a daemon with "shairport-sync -d". In previous versions (e.g. 19.0x) this worked. The error message is 

```
shairport-sync was built without libdaemon, so does not support daemonisation using the -d, --daemon, -j or --justDaemoniseNoPIDFile options
```

neheb, 2021-06-11:
this is very strange. No changes to libdaemon have been made.

closed 2021-06-12 by neheb

humaita-github, 2021-06-20:
Hi,

thank you for the fix. The initial error message related to libdaemon in now gone in OpenWrt 21.02.0-rc3. However, neither my iPhone 12 iOS 14.6 and my iPad mini running iOS 12.5.4 will find Sharport to play music. In previous versions (e.g. 19.0x) this worked perfectly.

Here is the output of 'shairport-sync -vvv' with the default config. I doesnt work with my custom config either.

<PRE>
# shairport-sync -vvv
         0.031699691 "shairport.c:461" looking for configuration file at full path "/etc/shairport-sync.conf"
         0.034930553 "shairport.c:1729" Started!
         0.001790806 "shairport.c:1756" software version: "3.3.8-libdaemon-mbedTLS-tinysvcmdns-ALSA-metadata-sysconfdir:/etc"
         0.001991687 "shairport.c:1762" log verbosity is 3.
         0.011758376 "audio_alsa.c:998" alsa: init() -- alsa_backend_state => abm_disconnected.
         0.002122278 "audio_alsa.c:1015" alsa: alsa_maximum_stall_time of 0.200000 sec.
         0.004743681 "audio_alsa.c:1302" alsa: disable_standby_mode is "never".
         0.001931402 "audio_alsa.c:1304" alsa: disable_standby_mode_silence_threshold is 0.040000 seconds.
         0.001350570 "audio_alsa.c:1306" alsa: disable_standby_mode_silence_scan_interval is 0.004000 seconds.
         0.002512424 "audio_alsa.c:1345" alsa: output device name is "default".
         0.000682842 "shairport.c:1780" The processor is running big-endian.
         0.001774421 "shairport.c:1810" disable resend requests is off.         0.004016769 "audio_alsa.c:1968" keep_dac_busy is now "no"

         0.002171668 "shairport.c:1814" diagnostic_drop_packet_fraction is 0.000000. A value of 0.0 means no packets will be dropped deliberately.
         0.000288806 "shairport.c:1815" statistics_requester status is 0.
         0.000145011 "shairport.c:1817" daemon status is 0.
         0.001397309 "shairport.c:1818" daemon pid file path is "/var/run/shairport-sync/shairport-sync.pid".
         0.000631048 "shairport.c:1820" rtsp listening port is 5000.
         0.000183835 "shairport.c:1821" udp base port is 6001.
         0.000228726 "shairport.c:1822" udp port range is 10.
         0.000152021 "shairport.c:1823" player name is "hostname".
         0.001467915 "shairport.c:1824" backend is "alsa".
         0.000204140 "shairport.c:1825" run_this_before_play_begins action is "(null)".
         0.000150346 "shairport.c:1826" run_this_after_play_ends action is "(null)".
         0.001105954 "shairport.c:1827" wait-cmd status is 0.
         0.000191866 "shairport.c:1828" run_this_before_play_begins may return output is 0.
         0.000151765 "shairport.c:1829" run_this_if_an_unfixable_error_is_detected action is "(null)".
         0.000150171 "shairport.c:1830" run_this_before_entering_active_state action is  "(null)".
         0.001200164 "shairport.c:1831" run_this_after_exiting_active_state action is  "(null)".
         0.000364541 "shairport.c:1832" active_state_timeout is  10.000000 seconds.
         0.000157631 "shairport.c:1833" mdns backend "(null)".
         0.000140370 "shairport.c:1834" userSuppliedLatency is 0.
         0.000142191 "shairport.c:1837" interpolation setting is "basic".
         0.000142195 "shairport.c:1838" interpolation soxr_delay_threshold is 30.
         0.000151646 "shairport.c:1839" resync time is 0.050000 seconds.
         0.001590625 "shairport.c:1840" allow a session to be interrupted: 0.
         0.000211051 "shairport.c:1841" busy timeout time is 120.
         0.000328646 "shairport.c:1842" drift tolerance is 0.001995 seconds.
         0.000168231 "shairport.c:1843" password is "(null)".
         0.000139995 "shairport.c:1844" ignore_volume_control is 0.
         0.000141486 "shairport.c:1848" volume_max_db is not set
         0.000135315 "shairport.c:1850" volume range in dB (zero means use the range specified by the mixer): 0.
         0.000155856 "shairport.c:1854" volume_range_combined_hardware_priority (1 means hardware mixer attenuation is used first) is 0.
         0.001232189 "shairport.c:1856" playback_mode is 0 (0-stereo, 1-mono, 1-reverse_stereo, 2-both_left, 3-both_right).
         0.000370891 "shairport.c:1857" disable_synchronization is 0.
         0.000150096 "shairport.c:1858" use_mmap_if_available is 1.
         0.000140395 "shairport.c:1860" output_format automatic selection is enabled.
         0.000149426 "shairport.c:1864" output_rate automatic selection is enabled.
         0.000145965 "shairport.c:1868" audio backend desired buffer length is 0.200000 seconds.
         0.000158931 "shairport.c:1870" audio_backend_buffer_interpolation_threshold_in_seconds is 0.120000 seconds.
         0.006844834 "shairport.c:1871" audio backend latency offset is 0.000000 seconds.
         0.000279601 "shairport.c:1873" audio backend silence lead-in time is "auto".
         0.001067274 "shairport.c:1877" zeroconf regtype is "_raop._tcp".
         0.000194770 "shairport.c:1878" decoders_supported field is 1.
         0.000141681 "shairport.c:1879" use_apple_decoder is 0.
         0.000141345 "shairport.c:1880" alsa_use_hardware_mute is 0.
         0.001560836 "shairport.c:1884" no special mdns service interface was requested.
         0.000431701 "shairport.c:1888" configuration file name "/etc/shairport-sync.conf" resolves to "/etc/shairport-sync.conf".
         0.002466569 "shairport.c:1894" metadata enabled is 1.
         0.000331151 "shairport.c:1895" metadata pipename is "/tmp/shairport-sync-metadata".
[... cut ...]
         0.000143251 "tinysvcmdns.c:1386"

         6.432745713 "tinysvcmdns.c:1504" data from=192.168.1.112 size=85

         0.000238616 "tinysvcmdns.c:756" rr_data_len goes beyond packet buffer: 4488 > 26

         0.000113085 "tinysvcmdns.c:894" error parsing answer #0

         2.813292025 "tinysvcmdns.c:1504" data from=192.168.1.112 size=85

         0.000217806 "tinysvcmdns.c:756" rr_data_len goes beyond packet buffer: 4485 > 26

         0.000111621 "tinysvcmdns.c:894" error parsing answer #0

</PRE>

neheb, 2021-06-20:
Is this ALAC content?

humaita-github, 2021-06-20:
Hi, I tried an MP3 through the iPhone's Music App (which was synced from my computer), I tested Spotify, and an online radio App. I don't think the problem is related to the format of the music, since shairplay doesn't even show up as a speaker that I can select.

neheb, 2021-06-20:
Based on the logs, it seems to be an issue with avahi-daemon. Don't quite know what that is.

humaita-github, 2021-06-21:
Hi, thank you for the feedback.

Avahi implements network discovery for apple protocols. 

Yes, I am using "shairport-sync-mini".

Someone seems to be having similar problems, and since yesterday there is a patch available. I wonder if this fixes this issue? https://github.com/mikebrady/shairport-sync/issues/1098

neheb, 2021-06-21:
Sounds like it.