openwrt/packages #15560: libuuid1 has missing dependencies: librt

BUG-reporter, 2021-05-01:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Atheros AR9132 rev 2
TP-Link TL-WR1043ND v1
OpenWrt 21.02.0-rc1 r16046-59980f7aaf / LuCI openwrt-21.02 branch git-21.106.55967-06dd6b5

Description:
Trying to install ```luci-app-upnp``` package I am getting the following warning:
libuuid1 has missing dependencies: librt
```
Formating code blocks by wrapping them with pairs of ```
```
[image]

derekpiasecki, 2021-08-17:
Issue still present in rc4:
- Qualcomm Atheros QCA9558 ver 1 rev 0
- EnGenius EPG5000
- OpenWrt 21.02.0-rc4 r16256-2d5ee43dc6 / LuCI openwrt-21.02 branch git-21.188.55099-e52e1de

champtar, 2021-08-20:
This is a bug in openwrt core for me, and it's weird because the package is present: https://downloads.openwrt.org/releases/21.02.0-rc4/targets/ath79/generic/packages/librt_1.1.24-3_mips_24kc.ipk

champtar, 2021-08-20:
Can you show the output of:
```
opkg update
opkg list librt
opkg list-installed librt
opkg status librt
```

chx, 2021-09-05:
The problem is a bit circular: 

1. opkg update hits https://downloads.openwrt.org
2. it is using wget to download
3. wget-ssl needs librt 
4. but librt is not installed by default 

Checkmate.

Looking at https://downloads.openwrt.org/releases/21.02.0/targets/ath79/generic/packages/ and the size of librt and even downloading it, there's literally nothing in there. The data.tar.gz you get contains a completely empty /lib directory. It is possible the bug is in wget-ssl, I do not know what's going on.

champtar, 2021-09-05:
@chx can you give exact repro steps ? You installed image X, then you run command Y, ...

chx, 2021-09-05:
I updated openwrt 19 stable to openwrt 21 stable on an TP-Link Archer C7 V2 AC1750 (openwrt-19.07.8-ath79-generic-tplink_archer-c7-v2-squashfs-sysupgrade.bin) 

ddns in luci complained I didn't have wget installed, so I looked in the packages. wget-ssl needed librt which was unavailable so I installed wget-nossl. Now every opkg command fails.

I ssh'd in , edited  /etc/opkg/distfeeds.conf to use http instead of https, it started working.

I do not know why `opkg install librt_1.1.24-3_mips_24kc.ipk` wanted to download a package instead of installing it. I saw some discussion at https://lists.openwrt.org/pipermail/openwrt-devel/2021-May/035147.html but `/proc/cpuinfo` says `cpu model: MIPS 74Kc V5.0` so I believe that was the right architecture/file. 

Some changes that would improve the situation:

1. I am not sure whether https://github.com/openwrt/packages/issues/14356 made it into the release
2. There's no point in using HTTPS in `/etc/opkg/distfeeds.conf` when the packages are signed anyways.
2. Even after installing wget-ssl , you need to remove wget-nossl otherwise still wget-nossl runs. wget-ssl should "trump" wget-nossl, not sure whether opkg has `conflicts` feature but even if not, it should be packaged so that wget-ssl is further up in `PATH` than wget-nossl.
3. If librt is zero bytes anyways why is it simply not installed by default?

champtar, 2021-09-06:
TP-Link Archer C7 V2 AC1750 image is [here](https://downloads.openwrt.org/releases/21.02.0/targets/ath79/generic/openwrt-21.02.0-ath79-generic-tplink_archer-c7-v2-squashfs-sysupgrade.bin), cpu is MIPS 74Kc, and all the [packages](https://downloads.openwrt.org/releases/21.02.0/targets/ath79/generic/packages/) and [kmods](https://downloads.openwrt.org/releases/21.02.0/targets/ath79/generic/kmods/5.4.143-1-79c5dc6db69102eb2943a96b9bec8b63/) are mips 24kc, so that might be why opkg doesn't find librt

champtar, 2021-09-07:
I just built openwrt-ath79-generic-tplink_archer-c7-v2-squashfs-sysupgrade.bin and everything is mips 24kc so cpu model might not be the issue

champtar, 2021-09-07:
@chx can you show me the output of `opkg print-architecture` ?

champtar, 2021-09-07:
and also `opkg info librt`

champtar, 2021-09-07:
Testing with malta image (`qemu-system-mips -kernel openwrt-21.02.0-malta-be-vmlinux-initramfs.elf -nographic -m 256`) I'm able to `opkg remove` and `opkg install` librt

chx, 2021-09-07:
```
root@OpenWrt:~# opkg print-architecture
arch all 1
arch noarch 1
arch mips_24kc 10
root@OpenWrt:~# opkg info librt
Package: librt
Version: 1.1.24-3
Depends: libpthread
Status: install hold installed
Section: libs
Essential: yes
Architecture: mips_24kc
Size: 904
Filename: librt_1.1.24-3_mips_24kc.ipk
Description: POSIX.1b RealTime extension library
Installed-Time: 1630809727

champtar, 2021-09-07:
I know librt is an empty package, but it's still not normal that you get error when trying to install it. Are you able to reproduce the issue one more time ?

chx, 2021-09-07:
I can't because librt is essential so I can't remove it now.

champtar, 2021-09-07:
You can force the removal

chx, 2021-09-07:
```
 opkg install librt_1.1.24-3_mips_24kc.ipk
Installing librt (1.1.24-3) to root...
Downloading https://downloads.openwrt.org/releases/21.02.0/targets/ath79/generic/packages/librt_1.1.24-3_mips_24kc.ipk
Configuring librt.
```

It still doesn't like the local file.

champtar, 2021-09-08:
I'm not sure `opkg install` is supposed to install local files
So you can't reproduce the librt install failure ?

chx, 2021-09-08:
I removed `wget-ssl bind-host bind-dig bind-libs libuv1`  and I was able to remove librt by forcing. Sure enough luci reports librt not available when trying to install `wget-ssl`:

[image]

You can install it from command line if you want. But if you continue down the luci path and install wget-nossl instead, you are doomed. Here's why: by default `/usr/bin/wget` is a symlink to `/bin/uclient-fetch` but once you install wget-nossl, it is a symlink to `/usr/libexec/wget-nossl` which, together with the https in default packages breaks opkg.

chx, 2021-09-08:
```
root@OpenWrt:~# opkg remove wget-ssl bind-host bind-dig bind-libs libuv1
Removing package bind-host from root...
Removing package bind-dig from root...
Removing package bind-libs from root...
Removing package libuv1 from root...
root@OpenWrt:~# opkg remove librt
Refusing to remove essential package librt.
        Removing an essential package may lead to an unusable system, but if
        you enjoy that kind of pain, you can force opkg to proceed against
        its will with the option: --force-removal-of-essential-packages
No packages removed.
root@OpenWrt:~# opkg remove --force-removal-of-essential-packages librt
Removing essential package librt under your coercion.
        If your system breaks, you get to keep both pieces
Removing package librt from root...
root@OpenWrt:~# opkg install wget-ssl
Installing wget-ssl (1.21.1-1) to root...
Downloading https://downloads.openwrt.org/releases/21.02.0/packages/mips_24kc/packages/wget-ssl_1.21.1-1_mips_24kc.ipk
Installing librt (1.1.24-3) to root...
Downloading https://downloads.openwrt.org/releases/21.02.0/targets/ath79/generic/packages/librt_1.1.24-3_mips_24kc.ipk
Configuring librt.
Configuring wget-ssl.
```

champtar, 2021-09-08:
I believe this has just been fixed by @jow-, I'll close the issue for now
https://github.com/openwrt/packages/pull/16580
https://github.com/openwrt/luci/issues/5342
https://bugs.openwrt.org/index.php?do=details&task_id=4006

closed 2021-09-08 by champtar

chx, 2021-09-08:
That fixed bug #1 but the following two are still a problem:

1. wget-nossl breaks opkg on the system as it is shipped.
1. opkg install librt_1.1.24-3_mips_24kc.ipk wants to download it instead of installing. I verified above with md5sum the local file is the same as the one it downloads.

champtar, 2021-09-09:
To be clear opkg is not in core repo, bug/feature requests should be reported at https://bugs.openwrt.org/
Also 1 bug per ticket.

> That fixed bug #1 but the following two are still a problem:
> 
>   1. wget-nossl breaks opkg on the system as it is shipped.

We could prefer uclient-fetch with ssl support over wget-nossl, but that's pretty easy to work around if you 'break' your system

>   2. opkg install librt_1.1.24-3_mips_24kc.ipk wants to download it instead of installing. I verified above with md5sum the local file is the same as the one it downloads.

Where have you seen `opkg install` is supposed to install local ipk ?

chx, 2021-09-09:
> where have you seen opkg install is supposed to install local ipk ?

Huh! everywhere. https://forum.openwrt.org/t/install-packages-while-offline/66148/3 just an example. https://lists.openwrt.org/pipermail/openwrt-devel/2021-May/035147.html https://unix.stackexchange.com/a/248914/9452 the list is endless.

> 
We could prefer uclient-fetch with ssl support over wget-nossl, but that's pretty easy to work around if you 'break' your system

That'd be pretty sweet... it's not always easy to see these things when in panic :) hindsight is always 20/20.

champtar, 2021-09-09:
(I never use opkg / always build my images)
Just did some tests, opkg behavior is a bit weird
If you remove the package list (`rm /tmp/opkg-lists/openwrt_*`) it installs from the local file, if you have the package list (`opkg update`) it kinds of read the ipk and the file name and then install from the repo based on the filename

OpenWrt opkg is an old fork from yocto opkg with many patches on top, no idea if it's improved/fixed in yocto version.

derekpiasecki, 2021-09-11:
Issue still present in release.  One way to repro is to select `luci-app-upnp` to install, since it has `libuuid1` (and thus indirectly the missing `librt`) as a dependency.

Qualcomm Atheros QCA9558 ver 1 rev 0
EnGenius EPG5000
OpenWrt 21.02.0 r16279-5cc0535800 / LuCI openwrt-21.02 branch git-21.231.26241-422c175

sshaikh, 2021-11-15:
Not sure if related but after doing a sysupgrade from 19.x to 21.02.1 and restoring extroot as per https://wiki.mbirth.de/know-how/software/openwrt/sysupgrade-with-extroot.html (via the wiki), I can no longer:

1. Log in via web UI
2. run `opkg update` via the CLI:
```
wget: SSL support not available, please install one of the libustream-.*[ssl|tls] packages as well as the ca-bundle and ca-certificates packages.
```

Slightly stuck but will report back once I fix it.