openwrt/packages #15561: [luci-app-banip] UCI entry not found and indeed no "service" found on OpenWRT 19.07

darrepac, 2021-05-03:
Maintainer: @dibdot 
Environment: Zyxel NBG6616 / Qualcomm Atheros QCA9558 ver 1 rev 0 / OpenWrt 19.07.7 r11306-c4a6851c72

Description:

Hi

I got a "UCI entry not found" when installing luci-app-banip.
Indeed after the install, I have no service menu

thx

dibdot, 2021-05-03:
Which banIP version did you try to install?

darrepac, 2021-05-03:
0.3.11-1
Note that it seems to work well

dibdot, 2021-05-03:
Re-install both packages (bainip & luci-app-banip) with the opkg options '--force-reinstall --force-maintainer' to start with a fresh/valid config.

darrepac, 2021-05-03:
I got the same, see below:
```
root@OpenWrt:~# opkg install --force-reinstall --force-maintainer banip
Removing package banip from root...
Not deleting modified conffile /etc/config/banip.
Not deleting modified conffile /etc/banip/banip.whitelist.
Installing banip (0.3.11-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/mips_24kc/packages/banip_0.3.11-1_all.ipk
Conffile /etc/config/banip-opkg.backup using maintainer's setting.
Conffile /etc/banip/banip.whitelist-opkg.backup using maintainer's setting.
Configuring banip.
root@OpenWrt:~# opkg install --force-reinstall --force-maintainer luci-app-banip
Removing package luci-app-banip from root...
Installing luci-app-banip (git-21.107.58152-c19c716-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/mips_24kc/luci/luci-app-banip_git-21.107.58152-c19c716-1_all.ipk
Configuring luci-app-banip.
**uci: Entry not found**
```

dibdot, 2021-05-03:
Not reproducible for me, most probably another config in your setup is broken. Try this script to find the culprit:
```
#!/bin/sh

for conf in /etc/config/*
do
	error="$(uci show "${conf}" 2>&1 >/dev/null)"
	if [ -n "${error}" ]
	then
		printf "%s\n" "ERR: ${conf}, ${error}"
	else
		printf "%s\n" "OK : ${conf}"
	fi
done
```
save it as conf_check.sh, make it executable (chmod +x) and give it a try ...

darrepac, 2021-05-03:
Output:

root@OpenWrt:/tmp# ./conf_check.sh
OK : /etc/config/banip
OK : /etc/config/collectd
OK : /etc/config/dhcp
OK : /etc/config/dropbear
OK : /etc/config/firewall
OK : /etc/config/fstab
OK : /etc/config/luci
OK : /etc/config/luci_statistics
OK : /etc/config/network
OK : /etc/config/nlbwmon
OK : /etc/config/rpcd
ERR: /etc/config/show_wifi_clients.sh, uci: Entry not found
OK : /etc/config/sqm
OK : /etc/config/system
OK : /etc/config/ubootenv
OK : /etc/config/ucitrack
OK : /etc/config/uhttpd
OK : /etc/config/vnstat
OK : /etc/config/wireless

dibdot, 2021-05-03:
Good, than it's unrelated to banIP. ;-)
Just (-re)move the above listed shell script from /etc/config (this directory is reserved for uci config files only) to another dir and try it again ...

darrepac, 2021-05-03:
I didn't put the script in /etc/config, I put in /tmp

dibdot, 2021-05-03:
Most probably you've copied this by accident (see https://openwrt.org/faq/how_to_get_a_list_of_connected_clients for reference). I'll close this ticket now.

closed 2021-05-03 by dibdot

darrepac, 2021-05-03:
oh I understood!