openwrt/packages #15578: MWAN3 successfully dialed under the master branch shows no trace！！！

zxlhhyccc, 2021-05-07:
[image]
[image]

feckert, 2021-05-07:
A little more information would not be bad what exactly your problem is

- `ubus call mwan3 status`
- `uci show mwan3`

feckert, 2021-05-07:
I have fixed with my last commit the [use case](https://github.com/openwrt/packages/issues/15391) if there is not tracking IP set.

- https://github.com/openwrt/packages/pull/15537
- https://github.com/openwrt/packages/pull/15562

My output of mwan3 status:

```
root@st-dev-07 ~ # mwan3 status
Interface status:
 interface wan is online 00h:00m:00s, uptime 00h:00m:48s and tracking is not enabled
 interface wwan is offline and tracking is paused

Current ipv4 policies:
wan_only:
 wan (100%)
wan_wwan:
 wan (100%)
wwan_only:
 unreachable

Current ipv6 policies:
wan_only:
 unreachable
wan_wwan:
 unreachable
wwan_only:
 unreachable

Directly connected ipv4 networks:
224.0.0.0/3
172.16.2.0/24
172.18.0.0/24
192.168.100.0/24
192.168.0.0/24
127.0.0.0/8

Directly connected ipv6 networks:
fe80::/64

Active ipv4 user rules:
   30  2792 - wan_wwan  all  --  *      *       0.0.0.0/0            0.0.0.0/0

Active ipv6 user rules:
```
Ubus status:

LuCI status:
[image]

closed 2023-05-03 by feckert