openwrt/packages #15579: MWAN3: UDP traffic load balance issue, traffic goes in wrong interface

tangyisheng2, 2021-05-07:
Maintainer: @feckert @aarond10
Environment:
architecture: x86
OpenWrt version: R21.4 (I use a 3rd party compile version so I am not sure about the exact version, but I guess it would be 19.07)

### Description:
I have two wan connections named wan and wan2.
Recently, I set up OpenVPN using softether vpn. When connecting to the OpenVPN using UDP, only connecting to wan2 will work, and win1 never works. However, when using TCP, both wan and wan2 works fine.
I guess it might be when connecting using UDP, the traffic goes outbound from wan2 only. Thus when connecting to wan using UDP, the OpenVPN client only receives the response from wan2, which might be the cause of this issue.
Also, I have 4 PCDN devices in my network named JDC1 - JDC4, and these PCDN will upload data in UDP. I want to set a load balance for them, that is:
JDC1 --- wan
JDC2 --- wan
JDC3 --- wan2
JDC4 --- wan2
I found JDC1 and JDC2 often have a lower upload compared to JDC3 and JDC4. I have swapped them between wan and wan2, mostly the two devices that use wan will have much lower upload traffic.
I found an old issue with the same problem #2347.
If you need any additional messages, please let me know.
Looking forward to your reply. Thanks in advance.

### Logs
Here is some log you might need:
mwan3 status
```
Interface status:
 interface wan is online and tracking is active
 interface wan2 is online and tracking is active

Current ipv4 policies:
balanced:
 wan (100%)
wan1_only:
 wan (100%)
wan2_only:
 wan2 (100%)

Current ipv6 policies:
balanced:
 default
wan1_only:
 default
wan2_only:
 default

Directly connected ipv4 networks:
172.17.0.1
119.129.229.249
127.0.0.0/8
192.168.1.255
192.168.1.0
127.255.255.255
192.168.1.0/24
172.17.255.255
127.0.0.0
192.168.1.3
113.119.10.1
127.0.0.1
172.17.0.0
113.119.10.115
224.0.0.0/3
172.17.0.0/16
119.129.228.1

Directly connected ipv6 networks:
fe80::/10
240e:3b0:200:10e4::/64
fe80::/64

Active ipv4 user rules:
 1431 88128 S JDC1  all  --  *      *       192.168.1.220        0.0.0.0/0            
 2187  439K S JDC2  all  --  *      *       192.168.1.221        0.0.0.0/0            
 6746 3357K S JDC3  all  --  *      *       192.168.1.222        0.0.0.0/0            
 5718 2815K S JDC4  all  --  *      *       192.168.1.223        0.0.0.0/0            
 3875  203K S https  tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            multiport dports 443 
31097 2037K - balanced  all  --  *      *       0.0.0.0/0            0.0.0.0/0            

Active ipv6 user rules:
    0     0 S https  tcp      *      *       ::/0                 ::/0                 multiport dports 443 
11582 1010K - balanced  all      *      *       ::/0                 ::/0
```

/etc/config/networks
```

config rule 'JDC1'
	option src_ip '192.168.1.220'
	option proto 'all'
	option use_policy 'wan1_only'
	option sticky '1'

config rule 'JDC2'
	option src_ip '192.168.1.221'
	option proto 'all'
	option use_policy 'wan1_only'
	option sticky '1'

config rule 'JDC3'
	option src_ip '192.168.1.222'
	option proto 'all'
	option use_policy 'wan2_only'
	option sticky '1'

config rule 'JDC4'
	option src_ip '192.168.1.223'
	option proto 'all'
	option use_policy 'wan2_only'
	option sticky '1'

config rule 'https'
	option sticky '1'
	option dest_port '443'
	option proto 'tcp'
	option use_policy 'balanced'

config rule 'default_rule'
	option dest_ip '0.0.0.0/0'
	option use_policy 'balanced'

config globals 'globals'
	option mmx_mask '0x3F00'
	option rtmon_interval '5'

config interface 'wan'
	option enabled '1'
	option count '2'
	option timeout '2'
	option interval '5'
	option down '4'
	option up '1'
	list track_ip 'www.baidu.com'
	list track_ip '114.114.114.114'
	list track_ip '119.29.29.29'
	option reliability '1'
	option initial_state 'online'
	option family 'ipv4'
	option track_method 'ping'
	option size '56'
	option failure_interval '5'
	option recovery_interval '5'
	option flush_conntrack 'never'

config interface 'wan2'
	option enabled '1'
	option count '2'
	option timeout '2'
	option interval '5'
	option down '4'
	option up '1'
	list track_ip 'www.baidu.com'
	list track_ip '114.114.114.114'
	list track_ip '119.29.29.29'
	option reliability '1'
	option initial_state 'online'
	option family 'ipv4'
	option track_method 'ping'
	option size '56'
	option failure_interval '5'
	option recovery_interval '5'
	option flush_conntrack 'never'

config member 'mwan1'
	option interface 'wan'
	option metric '1'

config member 'mwan2'
	option interface 'wan2'
	option metric '2'

config policy 'balanced'
	list use_member 'mwan1'
	list use_member 'mwan2'
	option last_resort 'default'

config policy 'wan1_only'
	list use_member 'mwan1'
	option last_resort 'default'

config policy 'wan2_only'
	list use_member 'mwan2'
	option last_resort 'default'

```

tangyisheng2, 2021-05-10:
> Difficult to say. Can you check your setup without mwan3 See if there are any problems there.
> 
> Using any kind of VPN is always tricky and needs to be thought through properly.
> Just for the sake of understanding, if no VPN is used, then your setup works, apart from the performance problem on wan1?

Thanks for your reply.

Since I can only remotely access the router on weekdays, I would try to check the setup with mwan3 disabled. Also, I have set the mwan3 policy to default and see if the load balance of the UDP traffic from PCDN works. Still need to observe the trafic statistics for a few days. :)

Anyway, I will post the update in the comment. Thank you.

tangyisheng2, 2021-05-27:
Hi, after monitor my traffic for about a week, I find an interesting thing.
As I mention in the issue, all UDP traffic goes to wan2. 
I tried to disable in wan2 using ```mwan3 ifdown mwan2```. After disabling it, I found the UDP traffic still goes to wan2.
It seems like that the UDP traffic is not under the control of mwan3.
```verbose
Interface status:
 interface wan1 is online and tracking is active
 interface wan2 is online and tracking is active

Current ipv4 policies:
balanced:
 wan2 (50%)
 wan1 (50%)
wan1_only:
 wan1 (100%)
wan2_only:
 wan2 (100%)

Current ipv6 policies:
balanced:
 default
wan1_only:
 unreachable
wan2_only:
 unreachable

Directly connected ipv4 networks:
113.119.10.13
192.168.1.0
127.0.0.1
113.119.10.1
127.255.255.255
224.0.0.0/3
127.0.0.0/8
172.17.255.255
192.168.1.3
119.129.228.1
172.17.0.0/16
172.17.0.0
192.168.1.255
127.0.0.0
172.17.0.1
192.168.1.0/24
119.129.229.250

Directly connected ipv6 networks:
240e:3b0:200:9116::/64
fe80::/64
fe80::/10

Active ipv4 user rules:
  899 86523 - balanced  udp  --  *      *       0.0.0.0/0            0.0.0.0/0            
  350 18893 - balanced  tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            multiport dports 443 
  820 54290 S default  all  --  *      *       0.0.0.0/0            0.0.0.0/0            

Active ipv6 user rules:
 1952  171K - balanced  udp      *      *       ::/0                 ::/0                 
    0     0 - balanced  tcp      *      *       ::/0                 ::/0                 multiport dports 443 
    2   152 S default  all      *      *       ::/0                 ::/0
```
Here is my detailed configration from mwan3.

tangyisheng2, 2021-05-28:
Hi, @feckert 

I have done some search on Google, I fount that this issue maybe related to miniupnpd. But IGD does not support multi-wan. I've found a workaround in a post in the Ubnt community. Link here: https://bbs.ui.com.cn/t/usg-wan-upnp/47970
Since the post is in Chinese, here is the translation:
1. Apply the firewall port rules opened by miniupnpd to other interfaces (wan2 is pppoe1)
```
iptables -w -t filter -I FORWARD -i pppoe1! -o pppoe1 -j MINIUPNPD
```
2. Apply the mapping rules created by miniupnpd to other interfaces

```
iptables -w -t nat -A PREROUTING -i pppoe1 -j MINIUPNPD
iptables -w -t nat -A POSTROUTING -o pppoe1 -j MINIUPNPD-POSTROUTING
iptables -w -t mangle -A PREROUTING -i pppoe1 -j MINIUPNPD
```
I tried to run these commands, but in the last one, it shows:
```
root@OpenWrt:/etc# iptables -w -t mangle -A PREROUTING -i pppoe1 -j MINIUPNPD
iptables v1.8.4 (legacy): Couldn't load target `MINIUPNPD':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
```

I think this workaroud might work. Could you help me with the troubleshooting?

Thanks in advance!

feckert, 2021-06-07:
@tangyisheng2 Sorry, I'm out.
Ask in the forum. I am not using miniupnpd
As I see it, there is no **MINIUPNPD** table for the target `mangle`.
Look at the source of miniupnpd and see what tables they create maybe that there has changed something!

closed 2021-06-07 by feckert