openwrt/packages #15581: golang: does not build with Fedora 34/golang 1.16

flyn-org, 2021-05-09:
Maintainer: @jefferyto 
Environment: x86_64 master

Description:
Building Go fails on Fedora 34, which uses golang 1.16. I suspect this is because the equivalent of `GO111MODULE` is on by default in this version.
```
( cd "/home/mike/Source/aq/aquinas/openwrt-aquinas-user/build_dir/hostpkg/go-1.16/src" ; GOROOT_FINAL="/home/mike/Source/aq/aquinas/openwrt-aquinas-user/sta)
Building Go cmd/dist using /home/mike/Source/aq/aquinas/openwrt-aquinas-user/build_dir/hostpkg/go-1.16/.go_bootstrap. (go1.4-bootstrap-20170531 linux/amd64)
go tool dist: found go.mod file in /home/mike/Source/aq/aquinas: $GOROOT must not be inside a module
make[4]: *** [Makefile:359: /home/mike/Source/aq/aquinas/openwrt-aquinas-user/build_dir/hostpkg/go-1.16/.built] Error 1
make[4]: Leaving directory '/home/mike/Source/aq/aquinas/openwrt-aquinas-user/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#20.31#4.30#53.90
    ERROR: package/feeds/packages/golang [host] failed to build.
```

flyn-org, 2021-05-15:
@cotequeiroz, I saw that you contributed a number of golang 1.16 fixes. Do you know what might be happening here?

jefferyto, 2021-05-15:
The Go package doesn't use your system Go to compile so it shouldn't matter what you have installed.

> go tool dist: found go.mod file in /home/mike/Source/aq/aquinas: $GOROOT must not be inside a module

Do you have a go.mod file in /home/mike/Source/aq/aquinas?

flyn-org, 2021-05-15:
Yes! I had not thought of Go working its way that far up the path. Obviously, this `go.mod` is not meant for the OpenWrt build.

closed 2021-05-15 by flyn-org

jefferyto, 2021-06-11:
I believe this should be fixed in Go 1.17 (https://github.com/golang/go/issues/44209).