openwrt/packages #15592: tang: add tang-show-keys

rugk, 2021-05-10:
Maintainer: @Tiboris @neheb @cotequeiroz

Description:

You could include `tang-show-keys`, which allows one to retrieve the codes on localhost. It’s mentioned e.g. [in the `clevis-encrypt-tang` man page](https://manpages.debian.org/testing/clevis/clevis-encrypt-tang.1.en.html)

As such, for server admins it’s likely a good idea to have that included here, is not it?

It’s also not really large 1075 bytes, at least teh Fedora build I use for reference here:
```
$ ls -la $(which tang-show-keys)
-rwxr-xr-x. 1 root root 1075  8. Feb 12:06 /usr/bin/tang-show-keys
```

lespinasse, 2023-07-15:
Could the script be changed to use wget -O- "http://127.0.0.1:$port/adv" instead ? that's what I'm doing locally to avoid extra dependencies.

Related question, what is the bash dependency used for ? On my test systems I see all tang scripts starting with /bin/sh, which points to busybox. The scripts work fine that way, so do we really need the bash dependency ? depending on the openwrt host being used, flash space often comes at a premium...

nmav, 2023-08-30:
Should we add a package dependency for either curl or wget for tang-show-keys? One doesn't have to use the program and a dependency means that curl or wget will come with tang nevertheless. What do you think?

lespinasse, 2023-08-30:
wget (or at least a limited version of it) can be depended to be present on all openwrt systems, because it is provided by uclient-fetch which opkg itself depends on. That is why I was recommending it over curl for the tang-show-keys script.

One more thing, the wget command should be wget -qO- "http://127.0.0.1:$port/adv" (I forgot the -q in my previous message) and it would be good to have $port default to the same value as in the rest of the package (I believe 9090 ?)

nmav, 2023-08-30:
Let's try that, I've pushed it to master.

commit 148b1b2d (2023-08-30): tang: tang-show-keys uses wget instead of curl

Resolves: #15592
commit eb822fe2 (2023-08-31): tang: tang-show-keys uses wget instead of curl

Resolves: #15592
closed 2023-08-30 by nmav