openwrt/packages #15610: luci-app-acme - execute command after renew cert

Emulty, 2021-05-11:
Please add field to acme settings page of LuCI for configure executing commands (--reloadcmd) after renew cert.

An example I have a lighthttpd in my openwrt. Now we don't have simple solution to solve auto prepare cert and restart demon. If add field for setting commands executing on stage of run-acme script it solve all problems.

Unfortunately, I can't pass pass the parameters to acme.sh at the certificate update execution stage without making significant edits to the run-acme script::
`--reloadcmd "cat ${domain_dir}/fullchain.cer ${domain_dir}/${main_domain}.key > /etc/lighttpd-self-signed.pem; /etc/init.d/lighttpd restart"`

feckert, 2021-05-12:
We have already support for uhttpd and nginx.
Maybe you could add this for lighttpd see https://github.com/openwrt/packages/blob/master/net/acme/files/run.sh#L193

MarceloRuiz, 2021-11-03:
Having the option to use --reloadcmd is really important for those who would like to run a custom sh script for copying certificates to multiple servers not running on openwrt (especially useful after obtaining star subdomain certificates). At least users will have some flexibility to deploy the certificates where they need them without having to modify the provided scripts and risk those modifications to be gone after an update...

feckert, 2021-11-03:
I would use the hotplug.d mechanism. So that we append the following arg to the acme calls
`--reloadcmd /sbin/hotplug acme`
Then everyone can place their script under `/etc/hotplug.d/acme`.
All scripts in this directory will then be called on a reload

MarceloRuiz, 2021-11-03:
> I would use the hotplug.d mechanism. So that we append the following arg to the acme calls `--reloadcmd /sbin/hotplug acme` Then everyone can place their script under `/etc/hotplug.d/acme`. All scripts in this directory will then be called on a reload

Something like that would be great!

Neustradamus, 2022-08-15:
@Emulty, @feckert, @MarceloRuiz: Have you seen this PR from @stokito?
- https://github.com/openwrt/packages/pull/19033

feckert, 2022-08-16:
See https://github.com/openwrt/packages/commit/e84f6514538f0638507ce8066f801374db6f4150

closed 2022-08-16 by feckert