openwrt/packages #15617: mwan3: 1X WAN + 2x QMI load balancing- interfaces up and down like a jojo

xewonder, 2021-05-13:
Maintainer: @feckert @aarond10
Environment:
architecture: ATH79
OpenWrt version: OpenWrt - OpenWrt SNAPSHOT r16427-f4f4259ca1 LuCI - git-21.095.59656-3561927 (But it's the same with latest snapshot)

Description:

I have a WD323 router with 2XQMI modules.
Trying to load balance the WAN + the 2 modems but every time i put any sort of traffic through, the interfaces start to go down and up like a jo-jo. At the end just traffic to the wan interface seems to go though.. but very erratically.

```
Interface status:
 interface wan is online 00h:06m:26s, uptime 00h:22m:25s and tracking is active
 interface modem1 is online 00h:06m:26s, uptime 00h:07m:21s and tracking is active
 interface modem2 is online 00h:06m:27s, uptime 00h:07m:20s and tracking is active

Current ipv4 policies:
balance_all:
 modem2 (33%)
 modem1 (33%)
 wan (33%)
balance_modems:
 modem2 (50%)
 modem1 (50%)

Current ipv6 policies:
balance_all:
 unreachable
balance_modems:
 unreachable

Directly connected ipv4 networks:
192.168.1.0/24
127.0.0.0/8
10.170.232.120/30
224.0.0.0/3
192.168.70.0/24
192.168.0.0/16
100.68.31.224/29
```

Not sure if it matters but i have:
WARNING: Interface modem2 has no default route in the main routing table
WARNING: Interface modem1 has no default route in the main routing table

even though the QMI modules are online.

Am i doing something wrong or is there a bug?

From the toubleshooter:

```
Software-Version
-------------------------------------------------
OpenWrt - OpenWrt SNAPSHOT r16427-f4f4259ca1
LuCI - git-21.095.59656-3561927

Output of "ip a show"
-------------------------------------------------
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 78:a3:51:73:dd:2d brd ff:ff:ff:ff:ff:ff
    inet6 fe80::7aa3:51ff:fe73:dd2d/64 scope link 
       valid_lft forever preferred_lft forever
3: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
[... 136 lines cut ...]
 pkts bytes target     prot opt in     out     source               destination         
  639  119K MARK       all  --  eth1   *       0.0.0.0/0            0.0.0.0/0            match-set mwan3_connected src mark match 0x0/0x3f00 /* default */ MARK or 0x3f00
   17   640 MARK       all  --  eth1   *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00 /* wan */ MARK xset 0x100/0x3f00

Chain mwan3_ifaces_in (1 references)
 pkts bytes target     prot opt in     out     source               destination         
 1145  160K mwan3_iface_in_wan  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00
  488 41045 mwan3_iface_in_modem1  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00
  425 37534 mwan3_iface_in_modem2  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00

Chain mwan3_policy_balance_all (1 references)
 pkts bytes target     prot opt in     out     source               destination         
  102  6613 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00 statistic mode random probability 0.33300000010 /* modem2 1 3 */ MARK xset 0x300/0x3f00
   80  7551 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00 statistic mode random probability 0.50000000000 /* modem1 1 2 */ MARK xset 0x200/0x3f00
   85  8335 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00 /* wan 1 1 */ MARK xset 0x100/0x3f00

Chain mwan3_policy_balance_modems (0 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00 statistic mode random probability 0.50000000000 /* modem2 1 2 */ MARK xset 0x300/0x3f00
    0     0 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00 /* modem1 1 1 */ MARK xset 0x200/0x3f00

Chain mwan3_rules (1 references)
 pkts bytes target     prot opt in     out     source               destination         
  267 22499 mwan3_policy_balance_all  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00
```

xewonder, 2021-05-13:
further to the above, when i put traffic through this is what i get in the Diagnostics:

Interface status:
 interface wan is online 00h:15m:29s, uptime 00h:31m:27s and tracking is active
 interface modem1 is error (16) and tracking is active
 interface modem2 is error (16) and tracking is active

feckert, 2021-05-13:
You do not have a default route set so you get error(16)https://github.com/openwrt/packages/blob/332d64a5bbedc1e07b9c53ba297e05ef2971224d/net/mwan3/files/lib/mwan3/mwan3.sh#L1045-L1050

xewonder, 2021-05-13:
> 
> 
> You do not have a default route set so you get error(16)
> 
> https://github.com/openwrt/packages/blob/332d64a5bbedc1e07b9c53ba297e05ef2971224d/net/mwan3/files/lib/mwan3/mwan3.sh#L1045-L1050

Hello,

How can you set a default route on a QMI interface?  They just pick it up when they dial in. But the WAN interface set to DHCP client seems to be ok?

feckert, 2021-05-13:
As I see it, the default route is set for ipv4. Could it be that you are using ipv6?

xewonder, 2021-05-13:
no... don't think so...

the tests are done via wifi (bridget to lan) and everything ip6 is disabled..

feckert, 2021-05-13:
When I look at the output, I can't see anything wrong.
Have you change the tracking options?
If you go online via radio, then you have to give the ping tracking more time, otherwise the mwan3tracker thinks that the interface is down and if the connection is well it goes online again.

xewonder, 2021-05-13:
> 
> 
> When I look at the output, I can't see anything wrong.
> Have you change the tracking options?
> If you go online via radio, then you have to give the ping tracking more time, otherwise the mwan3tracker thinks that the interface is down and if the connection is well it goes online again.
> 
> The output:
> 
> ```
> WARNING: Interface modem2 has no default route in the main routing table
> WARNING: Interface modem1 has no default route in the main routing table
> ```
> 
> is from the LuCI?

yes, from luci...

this is my mwan3 config... what shold i change?

```
config globals 'globals'
	option local_source 'lan'
	option mmx_mask '0x3F00'
	list rt_table_lookup '220'

config interface 'wan'
	option family 'ipv4'
	option initial_state 'online'
	option track_method 'ping'
	option count '1'
	option size '56'
	option max_ttl '60'
	option check_quality '0'
	option timeout '4'
	option interval '10'
	option failure_interval '5'
	option recovery_interval '5'
	option down '5'
	option up '5'
	option enabled '1'
	list track_ip '193.92.137.52'
	option reliability '1'
	list flush_conntrack 'ifup'
	list flush_conntrack 'ifdown'
	list flush_conntrack 'connected'
	list flush_conntrack 'disconnected'

config rule 'default_rule_v4'
	option dest_ip '0.0.0.0/0'
	option family 'ipv4'
	option proto 'all'
	option sticky '0'
	option use_policy 'balance_all'

config interface 'modem1'
	option family 'ipv4'
	option initial_state 'online'
	option track_method 'ping'
	option count '1'
	option size '56'
	option max_ttl '60'
	option check_quality '0'
	option timeout '4'
	option interval '10'
	option failure_interval '5'
	option recovery_interval '5'
	option down '5'
	option up '5'
	option enabled '1'
	list track_ip '193.92.137.52'
	option reliability '1'
	list flush_conntrack 'ifup'
	list flush_conntrack 'ifdown'
	list flush_conntrack 'connected'
	list flush_conntrack 'disconnected'

config interface 'modem2'
	option family 'ipv4'
	option initial_state 'online'
	option track_method 'ping'
	option count '1'
	option size '56'
	option max_ttl '60'
	option check_quality '0'
	option timeout '4'
	option interval '10'
	option failure_interval '5'
	option recovery_interval '5'
	option down '5'
	option up '5'
	option enabled '1'
	list track_ip '193.92.137.52'
	option reliability '1'
	list flush_conntrack 'ifup'
	list flush_conntrack 'ifdown'
	list flush_conntrack 'connected'
	list flush_conntrack 'disconnected'

config member 'wan_member'
	option interface 'wan'
	option metric '1'
	option weight '1'

config member 'modem1_member'
	option interface 'modem1'
	option metric '1'
	option weight '1'

config member 'modem2_member'
	option interface 'modem2'
	option weight '1'
	option metric '1'

config policy 'balance_all'
	list use_member 'wan_member'
	list use_member 'modem1_member'
	list use_member 'modem2_member'
	option last_resort 'unreachable'

config policy 'balance_modems'
	list use_member 'modem1_member'
	list use_member 'modem2_member'
	option last_resort 'unreachable'

```

feckert, 2021-05-13:
At first glance, it looks fine.
Except for the modem<x> interfaces, I would increase the timeout, as it is a wireless connection.
Are there any other log entries that the mwan3 writes in the log?

xewonder, 2021-05-13:
> 
> 
> At first glance, it looks fine.
> Except for the modem interfaces, I would increase the timeout, as it is a wireless connection.
> Are there any other log entries that the mwan3 writes in the log?
> 
> Edit:
> You should also enter more tracking ips, which makes tracking more reliable. The connection will not go down if tracking ip does not answer immediately.

there is something esles wrong with it...
```
root@Waveloc-V1:~# ping 194.219.110.187 
PING 194.219.110.187 (194.219.110.187): 56 data bytes
^C
--- 194.219.110.187 ping statistics ---
4 packets transmitted, 0 packets received, 100% packet loss
root@Waveloc-V1:~# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=17 ttl=112 time=2711.763 ms
64 bytes from 8.8.8.8: seq=18 ttl=112 time=1824.084 ms
64 bytes from 8.8.8.8: seq=19 ttl=112 time=901.701 ms
64 bytes from 8.8.8.8: seq=20 ttl=112 time=57.107 ms
64 bytes from 8.8.8.8: seq=21 ttl=112 time=239.604 ms
64 bytes from 8.8.8.8: seq=22 ttl=112 time=87.345 ms
64 bytes from 8.8.8.8: seq=23 ttl=112 time=195.967 ms
64 bytes from 8.8.8.8: seq=24 ttl=112 time=78.492 ms
64 bytes from 8.8.8.8: seq=25 ttl=112 time=121.644 ms
64 bytes from 8.8.8.8: seq=26 ttl=112 time=74.937 ms
64 bytes from 8.8.8.8: seq=27 ttl=112 time=75.543 ms
64 bytes from 8.8.8.8: seq=28 ttl=112 time=275.159 ms
^C
--- 8.8.8.8 ping statistics ---
29 packets transmitted, 12 packets received, 58% packet loss
round-trip min/avg/max = 57.107/553.612/2711.763 ms
root@Waveloc-V1:~# 
```

Ping is out of  the window totally... i thinnk thats why the interfaces are up and down...

xewonder, 2021-05-13:
so... what am i doing wrong??

this is my  mwan3 config:

```
config globals 'globals'
	option local_source 'lan'
	option mmx_mask '0x3F00'
	option logging '1'
	option loglevel 'error'

config interface 'wan'
	option family 'ipv4'
	option initial_state 'online'
	option track_method 'ping'
	option count '1'
	option size '56'
	option max_ttl '60'
	option check_quality '0'
	option timeout '4'
	option interval '10'
	option failure_interval '5'
	option recovery_interval '5'
	option down '5'
	option up '5'
	option enabled '1'
	list track_ip '193.92.137.52'
	list track_ip '8.8.8.8'
	list track_ip '8.8.4.4'
	list track_ip '194.219.110.187'
	option reliability '1'
	list flush_conntrack 'ifup'
	list flush_conntrack 'ifdown'
	list flush_conntrack 'connected'
	list flush_conntrack 'disconnected'

config rule 'default_rule_v4'
	option dest_ip '0.0.0.0/0'
	option family 'ipv4'
	option proto 'all'
	option sticky '0'
	option use_policy 'balance_all'

config interface 'modem1'
	option family 'ipv4'
	option initial_state 'online'
	option track_method 'ping'
	option count '1'
	option size '56'
	option max_ttl '60'
	option check_quality '0'
	option timeout '4'
	option interval '10'
	option failure_interval '5'
	option recovery_interval '5'
	option down '5'
	option up '5'
	option reliability '1'
	option enabled '0'
	list track_ip '193.92.137.52'
	list track_ip '8.8.8.8'
	list track_ip '8.8.4.4'
	list track_ip '194.219.110.187'
	list flush_conntrack 'ifup'
	list flush_conntrack 'ifdown'
	list flush_conntrack 'connected'
	list flush_conntrack 'disconnected'

config interface 'modem2'
	option family 'ipv4'
[... 99 lines cut ...]
Thu May 13 19:41:02 2021 user.info mwan3track[1943]: Lost 20 ping(s) on interface modem2 (wwan1). Current score: 9
Thu May 13 19:42:57 2021 user.info mwan3track[1943]: Check (ping) failed for target "193.92.137.52" on interface modem2 (wwan1). Current score: 10
Thu May 13 19:43:25 2021 user.info mwan3track[1943]: Check (ping) failed for target "193.92.137.52" on interface modem2 (wwan1). Current score: 10
Thu May 13 19:43:47 2021 user.info mwan3track[1943]: Check (ping) failed for target "193.92.137.52" on interface modem2 (wwan1). Current score: 10
Thu May 13 19:43:55 2021 user.info mwan3track[1943]: Check (ping) failed for target "8.8.8.8" on interface modem2 (wwan1). Current score: 10
Thu May 13 19:44:21 2021 user.info mwan3track[1943]: Check (ping) failed for target "193.92.137.52" on interface modem2 (wwan1). Current score: 10
Thu May 13 19:44:29 2021 user.info mwan3track[1943]: Check (ping) failed for target "8.8.8.8" on interface modem2 (wwan1). Current score: 10
Thu May 13 20:00:06 2021 user.notice mwan3-hotplug[22145]: mwan3 hotplug on modem2_4 not called because interface disabled
Thu May 13 20:00:10 2021 user.notice mwan3-hotplug[22502]: mwan3 hotplug on modem1 not called because interface disabled
Thu May 13 20:00:12 2021 user.info mwan3track[1943]: Check (ping) failed for target "193.92.137.52" on interface modem2 (wwan1). Current score: 10
Thu May 13 20:00:12 2021 user.info mwan3track[1943]: Check (ping) failed for target "8.8.8.8" on interface modem2 (wwan1). Current score: 10
Thu May 13 20:00:12 2021 user.info mwan3track[1943]: Check (ping) failed for target "8.8.4.4" on interface modem2 (wwan1). Current score: 10
Thu May 13 20:00:12 2021 user.info mwan3track[1943]: Check (ping) failed for target "194.219.110.187" on interface modem2 (wwan1). Current score: 10
Thu May 13 20:00:12 2021 user.notice mwan3track[1943]: Interface modem2 (wwan1) is disconnecting
Thu May 13 20:00:17 2021 user.notice mwan3-hotplug[22819]: Execute ifdown event on interface modem2 (unknown)
Thu May 13 20:00:18 2021 user.info mwan3track[1943]: Detect ifdown event on interface modem2 (wwan1)
Thu May 13 20:00:18 2021 user.notice mwan3track[1943]: Interface modem2 (wwan1) is offline
Thu May 13 20:00:21 2021 user.notice mwan3-hotplug[22992]: Execute disconnected event on interface modem2 (wwan1)
Thu May 13 20:00:35 2021 daemon.err uhttpd[2367]: luci: accepted login on /admin/status/mwan for root from 192.168.70.11
Thu May 13 20:00:42 2021 user.notice mwan3-hotplug[23867]: mwan3 hotplug on modem1 not called because interface disabled
Thu May 13 20:00:47 2021 user.notice mwan3-hotplug[24267]: mwan3 hotplug on modem1_4 not called because interface disabled
root@Waveloc-V1:~# 
```

any help apreciated....

xewonder, 2021-05-13:
I have done some more testing.

By disabling mwan3 the 2 modems ping google without a fail! With mwan3 they are up and down!

[image]

```
wwan0
--- www.google.com ping statistics ---
2552 packets transmitted, 2551 packets received, 0% packet loss
round-trip min/avg/max = 33.846/65.299/465.571 ms

wwan1
--- www.google.com ping statistics ---
2592 packets transmitted, 2592 packets received, 0% packet loss
round-trip min/avg/max = 28.391/42.977/1624.625 ms
```
how can we actually see what is going on and fix it?

xewonder, 2021-05-14:
> At first glance, it looks fine.
> Except for the modem interfaces, I would increase the timeout, as it is a wireless connection.
> Are there any other log entries that the mwan3 writes in the log? [...]

You were right!!!

Changed my config to the following and it is now working as expected. 

Now for the fun of coova-chilli!!

```

config globals 'globals'
	option local_source 'lan'
	option mmx_mask '0x3F00'
	option logging '1'
	option loglevel 'debug'

config interface 'wan'
	option family 'ipv4'
	option initial_state 'online'
	option track_method 'ping'
	option size '56'
	option max_ttl '60'
	option check_quality '0'
	option interval '10'
	option failure_interval '5'
	option recovery_interval '5'
	option down '5'
	option up '5'
	option enabled '1'
	list track_ip '193.92.137.52'
	list track_ip '8.8.8.8'
	list track_ip '8.8.4.4'
	list track_ip '194.219.110.187'
	option reliability '1'
	option count '5'
	option timeout '10'
	list flush_conntrack 'ifup'
	list flush_conntrack 'ifdown'
	list flush_conntrack 'connected'
	list flush_conntrack 'disconnected'

config rule 'default_rule_v4'
	option dest_ip '0.0.0.0/0'
	option family 'ipv4'
	option proto 'all'
	option sticky '0'
	option use_policy 'balance_all'

config interface 'modem1'
	option family 'ipv4'
	option initial_state 'online'
	option track_method 'ping'
	option size '56'
	option max_ttl '60'
	option check_quality '0'
	option interval '10'
	option failure_interval '5'
	option recovery_interval '5'
	option down '5'
	option up '5'
	option reliability '1'
	list track_ip '193.92.137.52'
	list track_ip '8.8.8.8'
	list track_ip '8.8.4.4'
	list track_ip '194.219.110.187'
	option enabled '1'
	option timeout '10'
	option count '5'
	list flush_conntrack 'ifup'
	list flush_conntrack 'ifdown'
	list flush_conntrack 'connected'
	list flush_conntrack 'disconnected'

config interface 'modem2'
	option family 'ipv4'
	option initial_state 'online'
	option track_method 'ping'
	option size '56'
	option max_ttl '60'
	option check_quality '0'
	option interval '10'
	option failure_interval '5'
	option recovery_interval '5'
	option down '5'
	option up '5'
	option reliability '1'
	list track_ip '193.92.137.52'
	list track_ip '8.8.8.8'
	list track_ip '8.8.4.4'
	list track_ip '194.219.110.187'
	option enabled '1'
	option timeout '10'
	option count '5'
	list flush_conntrack 'ifup'
	list flush_conntrack 'ifdown'
	list flush_conntrack 'connected'
	list flush_conntrack 'disconnected'

config member 'wan_member'
	option interface 'wan'
	option metric '1'
	option weight '2'

config member 'modem1_member'
	option interface 'modem1'
	option metric '1'
	option weight '1'

config member 'modem2_member'
	option interface 'modem2'
	option weight '1'
	option metric '1'

config policy 'balance_all'
	list use_member 'wan_member'
	list use_member 'modem1_member'
	list use_member 'modem2_member'
	option last_resort 'unreachable'

config policy 'balance_modems'
	list use_member 'modem1_member'
	list use_member 'modem2_member'
	option last_resort 'unreachable'

```

closed 2021-05-14 by xewonder