openwrt/packages #15629: cryptodev-linux can't be built with the ath79/mikrotik snapshot SDK

rogerpueyo, 2021-05-14:
I was trying to build a set of packages for OpenWrt whey I realized that the ath79/mikrotik snapshot SDK fails to build the `cryptodev-linux` package. However, the ath79/generic and ath79/nand SDKs build it correctly.

Steps to reproduce:

1. Download latest [ath79/mikrotik snapshot SDK](https://downloads.openwrt.org/snapshots/targets/ath79/mikrotik/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64.tar.xz) and unpack it.
2. Update & install feeds:
`roger@7018:~/OpenWrt/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64$ ./scripts/feeds update -a; ./scripts/feeds install -a`
3. Compile a regular package, just to try (e.g., _iperf3_):
`roger@7018:~/OpenWrt/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64$ make -j8 package/iperf3/compile`
The menuconfig environment appears. To speed up things, deselect:
```
Global build settings  --->
   [ ] Select all target specific packages by default
   [ ] Select all kernel module packages by default
   [ ] Select all userspace packages by default
   [ ] Cryptographically sign package lists 
```
Exit &Save.
4. _iperf3_ is successfully compiled:
```
[...]
Config-build.in:9863:warning: defaults for choice values not supported
Config-build.in:10175:warning: defaults for choice values not supported
Config-build.in:10179:warning: defaults for choice values not supported
Config-build.in:10183:warning: defaults for choice values not supported
#
# No change to .config
#
 make[1] package/iperf3/compile
 make[2] -C package/toolchain clean-build
 make[2] -C package/toolchain compile
 make[2] -C feeds/base/package/libs/openssl clean-build
 make[2] -C feeds/base/package/libs/openssl compile
 make[2] -C feeds/packages/net/iperf3 compile
```
5. Compile _cryptodev-linux_:
`roger@7018:~/OpenWrt/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64$ make -j8 package/cryptodev-linux/compile`
6. _cryptodev-linux_ **fails** to compile:
```
Config-build.in:9659:warning: defaults for choice values not supported
Config-build.in:9663:warning: defaults for choice values not supported
Config-build.in:9863:warning: defaults for choice values not supported
Config-build.in:10175:warning: defaults for choice values not supported
Config-build.in:10179:warning: defaults for choice values not supported
Config-build.in:10183:warning: defaults for choice values not supported
#
# No change to .config
#
 make[1] package/cryptodev-linux/compile
 make[2] -C package/toolchain clean-build
 make[2] -C package/toolchain compile
 make[2] -C feeds/base/package/firmware/prism54-firmware clean-build
 make[2] -C feeds/base/package/firmware/linux-firmware clean-build
 make[2] -C feeds/base/package/firmware/prism54-firmware compile
 make[2] -C feeds/base/package/firmware/linux-firmware compile
 make[2] -C package/kernel/linux clean-build
 make[2] -C package/kernel/linux compile
 make[2] -C feeds/base/package/kernel/cryptodev-linux clean-build
 make[2] -C feeds/base/package/kernel/cryptodev-linux compile
    ERROR: package/feeds/base/cryptodev-linux failed to build.
make -r package/cryptodev-linux/compile: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/roger/OpenWrt/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/include/toplevel.mk:223: package/cryptodev-linux/compile] Error 1
```

Cool, let's add some verbosity:
```
roger@7018:~/OpenWrt/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64$ make -j1 V=99 package/cryptodev-linux/compile
[...]
make[2]: Leaving directory '/home/roger/OpenWrt/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/package/kernel/linux'
time: package/kernel/linux/compile#2.07#0.21#2.25
[... 9 lines cut ...]
  Building modules, stage 2.
  MODPOST 1 modules
ERROR: "crypto_ahash_final" [/home/roger/OpenWrt/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11/cryptodev.ko] undefined!
ERROR: "crypto_aead_encrypt" [/home/roger/OpenWrt/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11/cryptodev.ko] undefined!
ERROR: "crypto_aead_setkey" [/home/roger/OpenWrt/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11/cryptodev.ko] undefined!
ERROR: "crypto_aead_decrypt" [/home/roger/OpenWrt/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11/cryptodev.ko] undefined!
ERROR: "crypto_aead_setauthsize" [/home/roger/OpenWrt/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11/cryptodev.ko] undefined!
ERROR: "crypto_alloc_aead" [/home/roger/OpenWrt/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11/cryptodev.ko] undefined!
ERROR: "crypto_ahash_setkey" [/home/roger/OpenWrt/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11/cryptodev.ko] undefined!
ERROR: "crypto_alloc_ahash" [/home/roger/OpenWrt/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11/cryptodev.ko] undefined!
make[5]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[4]: *** [Makefile:1647: modules] Error 2
make[4]: Leaving directory '/home/roger/OpenWrt/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/linux-5.4.117'
make[3]: *** [Makefile:27: build] Error 2
make[3]: Leaving directory '/home/roger/OpenWrt/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11'
make[2]: *** [Makefile:61: /home/roger/OpenWrt/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11/.built] Error 2
make[2]: Leaving directory '/home/roger/OpenWrt/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/feeds/base/package/kernel/cryptodev-linux'
time: package/feeds/base/cryptodev-linux/compile#0.91#0.36#1.08
    ERROR: package/feeds/base/cryptodev-linux failed to build.
make[1]: *** [package/Makefile:116: package/feeds/base/cryptodev-linux/compile] Error 1
make[1]: Leaving directory '/home/roger/OpenWrt/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64'
make: *** [/home/roger/OpenWrt/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/include/toplevel.mk:223: package/cryptodev-linux/compile] Error 2
```

As mentioned above, the same procedure with the ath79/generic or ath79/nand SDKs gives no problem. Also the ath79/mikrotik 21.02-rc1 SDK is working fine.

micmac1, 2021-05-15:
Good morning,

I think this is likely due to this:

https://lists.openwrt.org/pipermail/openwrt-devel/2021-April/034651.html
https://lists.openwrt.org/pipermail/openwrt-devel/2021-April/034696.html

Kind regards,
Seb

rogerpueyo, 2021-05-19:
Hi Seb,

Thanks for your tips. I applied your patch to the latest source and generated the SDK for ath79/mikrotik locally. Unfortunately, when building `cryptodev-linux` using the SDK I generated, the error persists. I wondef if I may be missing some additional step?

Kind regards,
Roger

micmac1, 2021-05-21:
Can you paste your steps? Also what target you selected? Maybe I find some time to try to reproduce then.

rogerpueyo, 2021-05-21:
Sure! I selected the `ath79/mikrotik` target/subtarget, which has the broken SDK on current snapshots.

Here are the steps I followed to build the SDK locally and then try to compile `cryptodev-linux` with 1) `ath79/generic` "as is", 2) `ath79/mikrotik` "as is" and 3) `ath79/mikrotik` applying your patch. Sadly, one of them worked for me.

### 1. ath79/generic "as is"
<details>
  <summary>Click to show the details</summary>
  
- Clone the latest snapshot source code:
```
$ git clone https://git.openwrt.org/openwrt/openwrt.git openwrt-ath79-generic-as-is
...
$ cd openwrt-ath79-generic-as-is
```
- Run "make menuconfig" to select building the SDK locally. By default ath79/generic is selected, so there's nothing else to change:
```
$ make -j18 menuconfig
...
[*] Build the OpenWrt SDK
< Exit >
Do you wish to save your new configuration?
< Yes >
```
- Start the build:
```
$ make -j18
...
 make[3] -C package/boot/uboot-envtools compile
 make[2] package/install
 make[2] target/install
 make[3] -C target/linux install
 make[3] -C target/sdk install
 make[2] package/index
 make[2] json_overview_image_info
 make[2] checksum
```
- Unpack the generated SDK:
```
$ cd bin/targets/ath79/generic/
$ tar -xf openwrt-sdk-ath79-generic_gcc-8.4.0_musl.Linux-x86_64.tar.xz
$ cd openwrt-sdk-ath79-generic_gcc-8.4.0_musl.Linux-x86_64
```
- Update base feed and install packages to the SDK:
```
$ ./scripts/feeds update base; ./scripts/feeds install -a
...
Ignoring feed 'telephony' - index missing
```
- Compile package `tcpdump` using the generated SDK, not compiling all the packages, modules, etc. (works OK):
```
$ make -j18 package/tcpdump/compile
...
Global build settings  --->
   [ ] Select all target specific packages by default
   [ ] Select all kernel module packages by default
   [ ] Select all userspace packages by default
   [ ] Cryptographically sign package lists
< Exit >
Do you wish to save your new configuration?
< Yes >
...
 make[1] package/tcpdump/compile
 make[2] -C package/toolchain clean-build
 make[2] -C package/toolchain compile
 make[2] -C feeds/base/package/libs/libpcap clean-build
 make[2] -C feeds/base/package/libs/libpcap compile
 make[2] -C feeds/base/package/network/utils/tcpdump compile
```
- Compile package `cryptodev-linux` (fails):
```
[... 435 lines cut ...]
  sed 's/ko$/o/' /home/qmp/openwrt-ath79-mikrotik-patched/bin/targets/ath79/mikrotik/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11/modules.order | scripts/mod/modpost   -i ./Module.symvers -I /home/qmp/openwrt-ath79-mikrotik-patched/bin/targets/ath79/mikrotik/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11/Module.symvers  -o /home/qmp/openwrt-ath79-mikrotik-patched/bin/targets/ath79/mikrotik/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11/Module.symvers     -s -T - 
ERROR: "crypto_ahash_final" [/home/qmp/openwrt-ath79-mikrotik-patched/bin/targets/ath79/mikrotik/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11/cryptodev.ko] undefined!
ERROR: "crypto_aead_encrypt" [/home/qmp/openwrt-ath79-mikrotik-patched/bin/targets/ath79/mikrotik/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11/cryptodev.ko] undefined!
ERROR: "crypto_aead_setkey" [/home/qmp/openwrt-ath79-mikrotik-patched/bin/targets/ath79/mikrotik/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11/cryptodev.ko] undefined!
ERROR: "crypto_aead_decrypt" [/home/qmp/openwrt-ath79-mikrotik-patched/bin/targets/ath79/mikrotik/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11/cryptodev.ko] undefined!
ERROR: "crypto_aead_setauthsize" [/home/qmp/openwrt-ath79-mikrotik-patched/bin/targets/ath79/mikrotik/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11/cryptodev.ko] undefined!
ERROR: "crypto_alloc_aead" [/home/qmp/openwrt-ath79-mikrotik-patched/bin/targets/ath79/mikrotik/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11/cryptodev.ko] undefined!
ERROR: "crypto_ahash_setkey" [/home/qmp/openwrt-ath79-mikrotik-patched/bin/targets/ath79/mikrotik/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11/cryptodev.ko] undefined!
ERROR: "crypto_alloc_ahash" [/home/qmp/openwrt-ath79-mikrotik-patched/bin/targets/ath79/mikrotik/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/linux-ath79_mikrotik/cryptodev-linux-cryptodev-linux-1.11/cryptodev.ko] undefined!
make[5]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[4]: *** [Makefile:1647: modules] Error 2
[... cut ...]
    ERROR: package/feeds/base/cryptodev-linux failed to build.
make[1]: *** [package/Makefile:116: package/feeds/base/cryptodev-linux/compile] Error 1
make[1]: Leaving directory '/home/qmp/openwrt-ath79-mikrotik-patched/bin/targets/ath79/mikrotik/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64'
make: *** [/home/qmp/openwrt-ath79-mikrotik-patched/bin/targets/ath79/mikrotik/openwrt-sdk-ath79-mikrotik_gcc-8.4.0_musl.Linux-x86_64/include/toplevel.mk:223: package/cryptodev-linux/compile] Error 2
```
</details>

So that's mostly it.

micmac1, 2021-05-21:
Mmh OK. When I build an SDK like this then the SDK file itself ends up 59 MByte. The SDK file from [here](https://downloads.openwrt.org/snapshots/targets/ath79/mikrotik/) is 100 MByte. That's likely because the way you and me compiled the SDK we didn't build all the modules. Hence the Module.symvers file in the resulting SDK will lack the symbols, naturally.

Maybe compile the SDK after selecting cryptodev linux. This way all the modules it depends on should be selected.

aparcar, 2021-05-25:
I'll look into this once I found a suitable laptop charger

aparcar, 2021-05-25:
@micmac1 thank you very much for the patch, applied to master. I'll ask around for a backport.
@rogerpueyo thanks for testing and keeping the spirits high of downstream SDK projects!

micmac1, 2021-05-25:
Thanks Paul!

aparcar, 2021-05-26:
@micmac1 I'm just the dude with commit access, thanks for fixing the issue

rogerpueyo, 2021-05-26:
The SDKs are working correctly now. Thank you both! :-)