openwrt/packages #15633: nfs-kernel-server: /mnt does not support NFS export

blairfancy, 2021-05-16:
Maintainer: @neheb  
Environment: 
Opnwrt 21.02.0-rc1 on WRT1200AC router
clean new install 

Description:
after opkg install nfs-kernel-server got error :

```
point 1
point 2
point 3
exportfs: /etc/exports [1]: Neither 'subtree_check' or 'no_subtree_check' specified for export "*:/mnt".
  Assuming default behaviour ('no_subtree_check').
  NOTE: this default has changed since nfs-utils version 1.0.x

exportfs: /mnt does not support NFS export
```

i m also not able to translate my old 19.07 exports file into new openwrt version:

```
/mnt/sda3    192.168.25.0/24(rw,all_squash,no_subtree_check,insecure,anonuid=0,anongid=0) 
/mnt/sda3/protected/osmc	192.168.25.20(rw,no_root_squash,sync,no_subtree_check,insecure)
```

neheb, 2021-06-13:
Strange. I updated nfs-kernel-server so that it compiles with musl 1.2.x. Can you try mounting with -v ?

GitHuhl, 2021-06-24:
+1, also having this issue, but I'm completely new to NFS, so there's always a possibility that it's me doing something wrong. 

I'm using OpenWrt 21.02.0-rc3, r16172-2aba3e9784 on a Raspberry Pi 4b. So far I've been unable to export any mounted directories.  I get the same "does not support NFS export" error, dmesg output reports that I'm trying to mount F2FS _even though_ I'm specifying mount -t ext4 for example (yes I have the FS drivers installed, yes I've rebooted)

I'm glad to see I'm not alone. I'm actually unable to mount even a decrypted LUKS container (with the same dmesg errors as when starting/reloading nfsd)

ricsif, 2021-06-27:
+1
On a D-Link DWR-118 A2.
OpenWrt 21.02.0-rc3, r16172-2aba3e9784
For me mounting /tmp neither possible.

blairfancy, 2021-07-10:
a work around is to export a folder in /tmp/export/  and make a bind mount 

`mount --bind /mnt/sda3 /tmp/export/mnt/sda3`

and adapt export file accordingly:

```
/tmp/export     192.168.25.0/24(fsid=0,ro,all_squash,insecure,sync,no_subtree_check) 192.168.50.0/24(fsid=0,ro,all_squash,insecure,sync,no_subtree_check)
/tmp/export/mnt/sda3    192.168.25.0/24(fsid=1,rw,all_squash,no_subtree_check,insecure,anonuid=0,anongid=0) 
```
the trick is to add in /etc/init.d/fastb or elsewhere:

```
/bin/mkdir -p /tmp/export/mnt/sda3
/bin/mount --bind /mnt/sda3 /tmp/export/mnt/sda3
```
to automatically bind before nsfd start

ricsif, 2021-07-11:
@blairfancy thanks a lot, it did the trick.
Just for the record, both rows have to be in the export file (be aware, the first one is duplicated).

fpauser, 2021-11-24:
Any news on this? Just flashed a linksys wrt3200acm with OpenWrt 21.02.1 r16325-88151b8303 and also get this error. The workaround posted by @blairfancy works - but feels a bit hacky.

neheb, 2021-11-24:
@danwrt has done some work in master. Maybe it should be backported.

DavidASmith072, 2021-11-27:
I am experiencing this issue also with OpenWRT version 21.02.1 as well as version 17.07.7 on my Linksys WRT3200ACM Router.

dim6003, 2022-02-16:
idem on TP-Link Archer C2600 with OpenWrt 19.07.7 r11306-c4a6851c72 / LuCI openwrt-19.07 branch git-22.045.73925-36e5c1c, kernel 4.14.221.
Not tested (yet?) @blairfancy's trick.

Tonghaofeng, 2022-04-27:
still can not export nfs in external hard drive

Newifi-D2
--
Architecture | MediaTek MT7621 ver:1 eco:3
Target Platform | ramips/mt7621
Firmware Version | OpenWrt 21.02.1 r16325-88151b8303 / LuCI openwrt-21.02 branch git-22.083.69138-0a0ce2a
Kernel Version | 5.4.154

AleXSR700, 2023-02-27:
Is there a fix or workaround for this?

salim-b, 2023-02-27:
> Is there a fix or workaround for this?

Yes. See [this post](https://github.com/openwrt/packages/issues/15633#issuecomment-877658338) above. tl;dr: The NFSv4 root directory (the one that gets `fsid=0` in `/etc/exports`) mustn't be on the [OverlayFS](https://openwrt.org/docs/techref/flash.layout#partitioning_of_the_flash).

>  I can share subdirectories of mnt like sda1 but not the folder /mnt/ itself.

You won't be able to export the whole `/mnt` dir since that one is part of the OverlayFS.

AleXSR700, 2023-02-27:
Unfortunately exporting mnt is exactly the goal.

But I also noticed that I cannot export USB drives. Only internal drives of my NAS running OpenWRT :(

AleXSR700, 2023-02-28:
How come that mnt and everything else can be shared with SFTP/SSH or Samba?

defanator, 2025-03-09:
Came across this issue trying to figure out how to export root partition for PXE booting from Raspberry Pi 5. The workaround with bind mount doesn't work for me.

OpenWrt version:
```
# cat /etc/os-release 
NAME="OpenWrt"
VERSION="23.05.3"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 23.05.3"
VERSION_ID="23.05.3"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r23809-234f1a2efa"
OPENWRT_BOARD="ath79/generic"
OPENWRT_ARCH="mips_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 23.05.3 r23809-234f1a2efa"
```

nfs packages versions:
```
# opkg list-installed | grep -- nfs
kmod-fs-nfs - 5.15.150-1
kmod-fs-nfs-common - 5.15.150-1
kmod-fs-nfs-common-rpcsec - 5.15.150-1
kmod-fs-nfs-v4 - 5.15.150-1
kmod-fs-nfsd - 5.15.150-1
nfs-kernel-server - 2.6.2-3
nfs-utils-libs - 2.6.2-3
```

/etc/exports:
```
# cat /etc/exports 
#/mnt	*(ro,all_squash,insecure,sync)
#/srv/nfs/rpi5-mon *(rw,sync,no_subtree_check,no_root_squash)
#/srv/nfs/rpi5-mon *(rw,sync,no_subtree_check,no_root_squash,fsid=root)
/tmp/export *(fsid=0,ro,all_squash,insecure,sync,no_subtree_check)
/tmp/export/nfs/rpi5-mon *(fsid=1,rw,all_squash,no_subtree_check,insecure,anonuid=0,anongid=0)
```

mount output:
```
# mount -v
/dev/root on /rom type squashfs (ro,relatime,errors=continue)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/sda1 on /overlay type ext4 (rw,relatime)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
tmpfs on /dev type tmpfs (rw,nosuid,noexec,noatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,noatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
bpffs on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,noatime,mode=700)
/dev/mtdblock9 on /rwm type jffs2 (rw,relatime)
overlayfs:/overlay on /tmp/export/nfs/rpi5-mon type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
nfsd on /proc/fs/nfsd type nfsd (rw,relatime)
```

Restarting nfsd:
```
# /etc/init.d/nfsd restart
exportfs: /tmp/export/nfs/rpi5-mon does not support NFS export
```

Logs while attempting to mount from pi:
```
Sun Mar  9 11:54:35 2025 [1741506875.648] daemon.notice rpc.mountd[18679]: authenticated mount request from 10.9.9.116:1016 for /tmp/export/nfs/rpi5-mon (/tmp/export/nfs/rpi5-mon)
Sun Mar  9 11:54:35 2025 [1741506875.648] daemon.warn rpc.mountd[18679]: Cannot export /tmp/export/nfs/rpi5-mon, possibly unsupported filesystem or fsid= required
```

Currently I'm thinking of repartitioning USB flash drive used for extroot to include several partitions which could be mounted directly as ext4 or something. Would it help?

defanator, 2025-03-10:
> Currently I'm thinking of repartitioning USB flash drive used for extroot to include several partitions which could be mounted directly as ext4 or something. Would it help?

For the record, splitting USB drive to 2 partitions - 1 for extroot and another for direct ext4 fs for nfs exporting - worked just fine:

exports:
```
# cat /etc/exports
#/mnt	*(ro,all_squash,insecure,sync)
/warehouse/nfs *(fsid=0,ro,all_squash,insecure,sync,no_subtree_check)
#/warehouse/nfs/rpi5-mon *(fsid=1,rw,all_squash,no_subtree_check,insecure,anonuid=0,anongid=0)
/warehouse/nfs/rpi5-mon *(fsid=1,rw,no_root_squash,no_subtree_check,insecure)
```

showmount -e:
```
# showmount -e
Export list for ***:
/warehouse/nfs/rpi5-mon *
/warehouse/nfs          *
```

directories structure:
```
# ls -la /warehouse/nfs/
drwxr-xr-x    4 root     root          4096 Mar  9 16:45 .
drwxr-xr-x    4 root     root          4096 Mar  9 15:34 ..
lrwxr-xr-x    1 root     root            22 Mar  9 16:36 634a3be0 -> rpi5-mon/boot/firmware
drwxr-xr-x   18 root     root          4096 Nov 19 17:32 rpi5-mon
drwxr-xr-x   18 root     root          4096 Nov 19 17:32 rpi5-mon.bak
```

relevant bits of dhcp config:
```
config dnsmasq
	[..]
	option enable_tftp '1'
	option tftp_root '/warehouse/nfs'
	option dhcp_boot 'bootcode.bin'
	option logdhcp '0'

config boot 'linux'
	option filename 'bootcode.bin'
	option serveraddress '10.0.0.1'
	option servername '10.0.0.1'
	list dhcp_option '43,Raspberry Pi Boot'
```

mounts:
```
# mount -v
/dev/root on /rom type squashfs (ro,relatime,errors=continue)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/sda1 on /overlay type ext4 (rw,relatime)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
tmpfs on /dev type tmpfs (rw,nosuid,noexec,noatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,noatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
bpffs on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,noatime,mode=700)
/dev/sda2 on /warehouse type ext4 (rw,relatime)
nfsd on /proc/fs/nfsd type nfsd (rw,relatime)
```

blairfancy, 2025-03-10:
I don't see any bind mount on your command mount -v.

in my setup i have both line:
/dev/sda3 on /mnt/sda3 type ext4 (rw,relatime)
/dev/sda3 on /tmp/export/mnt/sda3 type ext4 (rw,relatime)

first one is physical mount point and second one is a bind to /tmp/export done with command "mount --bind /mnt/sda3 /tmp/export/mnt/sda3"

defanator, 2025-03-10:
here - https://github.com/openwrt/packages/issues/15633#issuecomment-2708722456

```
[..]
overlayfs:/overlay on /tmp/export/nfs/rpi5-mon type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
[..]
```

It was a bind mount from overlayfs extroot.

Now I do have both overlayfs and separate ext4 partition -
```
# mkdir /tmp/test1
# mkdir /tmp/test2
# mount --bind /warehouse/nfs /tmp/test1
# mount --bind /srv /tmp/test2
# mount -v | tail -2
/dev/sda2 on /tmp/test1 type ext4 (rw,relatime)
overlayfs:/overlay on /tmp/test2 type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
# 
```

(Expectedly, FS type doesn't change with bind mounts.)

Anyway it's a minor thing - everything is working just fine with direct mount from 2nd partition on the same flash drive which is being used as an extroot storage (my current device has just one USB port so using separate flash drive or USB HDD is not an option).