openwrt/packages #15640: mwan3: mwan3 doesn't take into account ipset-based rules before manual restart

sergeyrachev, 2021-05-17:
Maintainer: @aaronjg @feckert
Environment: OpenWRT 21.02 RC1;  Virtual Box VM or Docker rootfs image.

Description:
I try to configure mwan3 to be used with netify (https://www.netify.ai/developer/netify-agent/openwrt-integration)
I would like to route traffic based on ipset rules accordingly https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3
 
My setup has 2 WAN interfaces, netifyd and netify-fwa to detect traffic and manage ipset sets.

Immediately after reboot mwan3 doesn't use ipset rules althought they are available and observable with ipset list. 
After 'mwan3 restart' manually the rules appear in 'active' section and in use.

Neither 'service network restart' nor 'ifdown wanh', 'ifdown wanl', 'ifup wanh', 'ifup wanl' help.

Logs immediately after reboot. Mwan3 doesn't use ipset rules. 

```

Software-Version
-------------------------------------------------
OpenWrt - OpenWrt 21.02.0-rc1 r16046-59980f7aaf
LuCI - git-21.106.55967-06dd6b5

Output of "ip a show"
-------------------------------------------------
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 08:00:27:ff:08:82 brd ff:ff:ff:ff:ff:ff
    inet 192.166.58.4/24 brd 192.166.58.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::a00:27ff:feff:882/64 scope link 
       valid_lft forever preferred_lft forever
3: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 08:00:27:6f:b2:6c brd ff:ff:ff:ff:ff:ff
    inet 10.0.3.15/24 brd 10.0.3.255 scope global eth1
       valid_lft forever preferred_lft forever
    inet6 fe80::a00:27ff:fe6f:b26c/64 scope link 
       valid_lft forever preferred_lft forever
4: eth2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 08:00:27:eb:be:16 brd ff:ff:ff:ff:ff:ff
    inet 10.0.4.15/24 brd 10.0.4.255 scope global eth2
       valid_lft forever preferred_lft forever
    inet6 fe80::a00:27ff:feeb:be16/64 scope link 
       valid_lft forever preferred_lft forever
5: eth3: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel master br-lan state UP group default qlen 1000
    link/ether 08:00:27:41:00:21 brd ff:ff:ff:ff:ff:ff
6: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 08:00:27:41:00:21 brd ff:ff:ff:ff:ff:ff
    inet 192.166.57.106/24 brd 192.166.57.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fe80::a00:27ff:fe41:21/64 scope link 
       valid_lft forever preferred_lft forever

Output of "ip route show"
-------------------------------------------------
default via 10.0.3.2 dev eth1 proto static src 10.0.3.15 
default via 10.0.3.2 dev eth1 proto static metric 10 
default via 10.0.4.2 dev eth2 proto static metric 20 
10.0.3.0/24 dev eth1 proto kernel scope link src 10.0.3.15 
10.0.4.0/24 dev eth2 proto kernel scope link src 10.0.4.15 
192.166.57.0/24 dev br-lan proto kernel scope link src 192.166.57.106 
192.166.58.0/24 dev eth0 proto kernel scope link src 192.166.58.4

Output of "ip rule show"
-------------------------------------------------
[... 2097 lines cut ...]
      "application_category": "streaming-media"
    },
    {
      "type": "ipset",
      "application_category": "games"
    },
    {
      "type": "ipset",
      "application_category": "os-software-updates"
    },
    {
      "type": "ipset",
      "application_category": "voip"
    },
    {
      "type": "ipset",
      "application_category": "vpn-and-proxy"
    },
    {
      "type": "ipset",
      "application_category": "business"
    }
  ]
}
```

feckert, 2021-05-19:
Can you please check when the ipsets are created during the boot process. My guess is that the ipsets are not yet there when the mwan3 starts.

sergeyrachev, 2021-06-02:
Thanks for your reply.  That is exactly due to order of init script execution. mwan3 has START=19 when netifyd that manages ipset lists has START=50. The issue can be closed. 

Is there a way to adjust START value of service using 'service' command to not alter the original package provided service init script?

closed 2021-06-07 by sergeyrachev

mfoxworthy, 2022-01-27:
I think it takes a while to set the ipsets. I have changed the order and I still see the issue. I set mwan3 to 99, and it still misses them. Please reopen this issue.