openwrt/packages #15644: Passwordless wifi authentication via ssh terminal

vhm14u2c, 2021-05-18:
Hi, was not sure where to put this.
I use passwordless authentication for a couple email addresses I have (gmail and live.com)

I started thinking how wonderful if this was implemented into WiFi.

Where let's say, someone figures out your WiFi password using one of those hacking tools, but then gets stumped when it just sits there w/o logging in, but on the ssh console side, it's asking 'are you trying to login with this Mac-id @ this time/date?(y/N)'
If no input given, would default to no and keep the baddies out.

I haven't been coding since early 2000's, and wouldn't mind taking a go at it, (I'm old and rusty now if you ask me) but figured I would throw this out there in case someone may want to try some type of similar implementation of this.

Let me know if you do.  I will most likely try, or use it.

feckert, 2021-05-19:
Unfortunately, I do not understand then concerns. Can you describe in more detail what you mean by that?

vhm14u2c, 2021-05-23:
Using a similar 2 factor authentication where as long as you are at console of router using ssh, and when someone tries to login to WiFi network, would show up as console as someone trying to login, then prompts if you want to let them join.  only method would be via the person at the console to let them in, hence, password-less.

feckert, 2021-06-07:
From my point of view, this is an external software development that we can not discuss here.
For such things you can ask the [forum](https://forum.openwrt.org/) maybe they have an idea how to implement this and what you need for this.

closed 2025-11-14 by GeorgeSapkin