openwrt/packages #15653: nginx-util: build broken

anomeome, 2021-05-18:
Maintainer: @Ansuel @peter-stadler

Description: Last build on 21/05/14 worked fine, pull latest code today broken, cleaned out likely culprits to no avail:

```
/home/kc/wrtpac/source/staging_dir/host/bin/ccache /home/kc/wrtpac/source/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-10.3.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-g++ -DVERSION=1.5  -O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/home/kc/wrtpac/source/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/nginx-util-1.5=nginx-util-1.5 -Wformat -Werror=format-security -DPIC -fpic -fstack-protector-strong -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG   -Os -Wall -Werror -Wextra -g3 -Wno-unused-parameter -Wmissing-declarations -Wshadow -std=gnu++17 -MD -MT CMakeFiles/nginx-ssl-util.dir/nginx-util.cpp.o -MF CMakeFiles/nginx-ssl-util.dir/nginx-util.cpp.o.d -o CMakeFiles/nginx-ssl-util.dir/nginx-util.cpp.o -c nginx-util.cpp
In file included from nginx-util.cpp:4:
nginx-ssl-util.hpp: In lambda function:
nginx-ssl-util.hpp:169:40: error: attributes at the beginning of statement are ignored [-Werror=attributes]
  169 |                 case '^': ret += '\\'; [[fallthrough]];
      |                                        ^~~~~~~~~~~~~~~
nginx-ssl-util.hpp:170:27: error: attributes at the beginning of statement are ignored [-Werror=attributes]
  170 |                 case '_': [[fallthrough]];
      |                           ^~~~~~~~~~~~~~~
nginx-ssl-util.hpp: In lambda function:
nginx-ssl-util.hpp:169:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
  169 |                 case '^': ret += '\\'; [[fallthrough]];
      |                           ~~~~^~~~~~~
nginx-ssl-util.hpp:170:17: note: here
  170 |                 case '_': [[fallthrough]];
      |                 ^~~~
nginx-ssl-util.hpp: In lambda function:
nginx-ssl-util.hpp:169:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
  169 |                 case '^': ret += '\\'; [[fallthrough]];
      |                           ~~~~^~~~~~~
nginx-ssl-util.hpp:170:17: note: here
  170 |                 case '_': [[fallthrough]];
      |                 ^~~~
nginx-ssl-util.hpp: In lambda function:
nginx-ssl-util.hpp:169:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
  169 |                 case '^': ret += '\\'; [[fallthrough]];
      |                           ~~~~^~~~~~~
nginx-ssl-util.hpp:170:17: note: here
  170 |                 case '_': [[fallthrough]];
      |                 ^~~~
nginx-ssl-util.hpp: In lambda function:
nginx-ssl-util.hpp:169:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
  169 |                 case '^': ret += '\\'; [[fallthrough]];
      |                           ~~~~^~~~~~~
nginx-ssl-util.hpp:170:17: note: here
  170 |                 case '_': [[fallthrough]];
      |                 ^~~~
nginx-ssl-util.hpp: In lambda function:
nginx-ssl-util.hpp:169:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
  169 |                 case '^': ret += '\\'; [[fallthrough]];
      |                           ~~~~^~~~~~~
nginx-ssl-util.hpp:170:17: note: here
  170 |                 case '_': [[fallthrough]];
      |                 ^~~~
nginx-ssl-util.hpp: In lambda function:
nginx-ssl-util.hpp:169:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
  169 |                 case '^': ret += '\\'; [[fallthrough]];
      |                           ~~~~^~~~~~~
nginx-ssl-util.hpp:170:17: note: here
  170 |                 case '_': [[fallthrough]];
      |                 ^~~~
nginx-ssl-util.hpp: In lambda function:
nginx-ssl-util.hpp:169:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
  169 |                 case '^': ret += '\\'; [[fallthrough]];
      |                           ~~~~^~~~~~~
nginx-ssl-util.hpp:170:17: note: here
  170 |                 case '_': [[fallthrough]];
      |                 ^~~~
nginx-ssl-util.hpp: In lambda function:
nginx-ssl-util.hpp:169:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
  169 |                 case '^': ret += '\\'; [[fallthrough]];
      |                           ~~~~^~~~~~~
nginx-ssl-util.hpp:170:17: note: here
  170 |                 case '_': [[fallthrough]];
      |                 ^~~~
[... 135 lines cut ...]
nginx-ssl-util.hpp:170:17: note: here
  170 |                 case '_': [[fallthrough]];
      |                 ^~~~
nginx-ssl-util.hpp: In lambda function:
nginx-ssl-util.hpp:169:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
  169 |                 case '^': ret += '\\'; [[fallthrough]];
      |                           ~~~~^~~~~~~
nginx-ssl-util.hpp:170:17: note: here
  170 |                 case '_': [[fallthrough]];
      |                 ^~~~
nginx-ssl-util.hpp: In lambda function:
nginx-ssl-util.hpp:169:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
  169 |                 case '^': ret += '\\'; [[fallthrough]];
      |                           ~~~~^~~~~~~
nginx-ssl-util.hpp:170:17: note: here
  170 |                 case '_': [[fallthrough]];
      |                 ^~~~
cc1plus: all warnings being treated as errors
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:121: /home/kc/wrtpac/source/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/nginx-util-1.5/.built] Error 1
make[3]: Leaving directory '/home/kc/wrtpac/source/feeds/packages/net/nginx-util'
time: package/feeds/packages/nginx-util/compile#10.65#0.62#7.04
    ERROR: package/feeds/packages/nginx-util failed to build.

```

ptpt52, 2021-05-19:
same issue here.

peter-stadler, 2021-05-19:
Thank you for the report :-) I will look into it ...

Vladdrako, 2021-05-19:
Just replace -Werror with -Wno-error in https://github.com/openwrt/packages/blob/master/net/nginx-util/src/CMakeLists.txt#L8

peter-stadler, 2021-05-19:
This would be an option, but I prefer to solve the warning.
Actually, I do not understand why we get: 
`error: attributes at the beginning of statement are ignored [-Werror=attributes]`
It is working if I remove the `[[fallthrough]];` I think I will do that.

Ansuel, 2021-05-20:
@peter-stadler

think you should use the comment or declare c++17
(from random post online)
C++17 adds a [[fallthrough]] attribute that can be used to suppress such warnings. Note the trailing semicolon:

anomeome, 2021-05-20:
[recent netifd fixup](https://git.openwrt.org/?p=project/netifd.git;a=commit;h=02dd2f2df7cbd8bb39c122e827c27b4a765ff8bc), what forced this, gcc 10.x.x

peter-stadler, 2021-05-22:
@ansuel: I included `SET(CMAKE_CXX_STANDARD 17)` in the `CMakeLists.txt` and it has been working. I did not change anything, but now `[[fallthrough]];` is not working anymore (the compilation is called with `-std=gnu++17`). I do not understand, why ...

@anomeome: netifd uses `fallthrough;`, but this command is not recognized by `clang-tidy`. So I do not really like using it here.

Right now I have problems compiling gcc 10.3.0 using `make menuconfig && make toolchain/install`. So, I am not sure if the comment `/* fallthrough */` is working here (netifd has been using `/*fall through */` and beside that it is C not C++).

If I do not become able to test it, I will go back to my first fix copying the `^` and using `break;` instead ...

ptpt52, 2021-05-22:
as far as what I know, `/* ... */` mark just a comment, it works with c/c++ of most versions of them.
`[[fallthrough]];` attribute works only on c++17 and
somehow it is likely that `SET(CMAKE_CXX_STANDARD 17)` is not respected.

rmandrad, 2021-05-23:
It works for me if I use                fallthrough;

peter-stadler, 2021-05-24:
I did not get gcc 10.3.0 working and I do not know, when I would have more time to try it.
As I am not sure if `/* fallthrough */` is working for gcc 10, I copy the `ret += c; break;` instead of falling through.
This should solve the issue.

The problem behind is still open: It should be possible to use the c++17 attribute `[[fallthrough]];` However, I am not sure how to do it (and if it is worth it), but, we should investigate what causes the problem. 

@rmandrad: The command `fallthrough;` would work, but AFAIK it is not standard c++ and clang-tidy is not recognizing it (I use that as checker). So I hesitate using it here.

closed 2021-05-29 by anomeome