openwrt/packages #15679: Failed to update openwrt keyring file, conflict between base version and package version?  Error 255?

dachshund-digital, 2021-05-23:
Failed to update openwrt keyring file, conflict between base version and package version?  Error 255?

Details for package openwrt-keyring

    Version: 2021-02-20-49283916-2
    Size: ~393 B installed

Description

The keyring of with the developer using and gpg public keys.

Require approx. 393 B size for 1 package(s) to install.

Executing package manager

Upgrading openwrt-keyring on root from 2019-07-25-8080ef34-1 to 2021-02-20-49283916-2...
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/arm_xscale/base/openwrt-keyring_2021-02-20-49283916-2_arm_xscale.ipk

Errors

Collected errors:
 * check_data_file_clashes: Package openwrt-keyring wants to install file /etc/opkg/keys/f94b9dd6febac963
	But that file is already provided by package  * base-files
 * opkg_install_cmd: Cannot install package openwrt-keyring.

The opkg install command failed with code 255.

Any ideas how to resolve this issue?

dr-ni, 2021-05-25:
same here, any advice?

rulnick, 2021-05-25:
maybe @blogic ?

dachshund-digital, 2021-05-26:
Since I had a spare router, my backup, i told the package installer to replace the conflicting file(s) between the base files and the updated key ring files, the newer files appeared to be in the updated key ring file package.  No errors and the logging showed that in fact the older conflicting files where in the base file.  So far no issues.  Not sure if there is any explicit testing I should do to confirm the result?

dr-ni, 2021-05-30:
this seems to have fixed the problem for me:
```
root@OpenWrt:~# opkg --force-overwrite upgrade openwrt-keyring
Upgrading openwrt-keyring on root from 2019-07-25-8080ef34-1 to 2021-
02-20-49283916-2...
Downloading 
http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a9_vfpv3-d16/base/openwrt-keyring_2021-02-20-49283916-2_arm_cortex-a9_vfpv3-d16.ipk
Removing obsolete file /etc/opkg/keys/72a57f2191b211e0.
Removing obsolete file /etc/opkg/keys/b5043e70f9a75cde.
Removing obsolete file /etc/opkg/keys/5151f69420c3f508.
Removing obsolete file /etc/opkg/keys/0b26f36ae0f4106d.
Removing obsolete file /etc/opkg/keys/792d9d9b39f180dc.
Removing obsolete file /etc/opkg/keys/9ef4694208102c43.
Removing obsolete file /etc/opkg/keys/b2d571e0880ff617.
Removing obsolete file /etc/opkg/keys/c10b9afab19ee428.
Removing obsolete file /etc/opkg/keys/1035ac73cc4e59e3.
Removing obsolete file /etc/opkg/keys/dace9d4df16896bf.
Removing obsolete file /etc/opkg/keys/dd6de0d06bbd3d85.
Configuring openwrt-keyring.
root@OpenWrt:~# \ufffc
-ash: \ufffc: not found
root@OpenWrt:~# \ufffc
-ash: \ufffc: not found
root@OpenWrt:~# \ufffc
-ash: \ufffc: not found
```

```
root@OpenWrt:~# opkg update 
Downloading 
http://downloads.openwrt.org/releases/19.07.7/targets/mvebu/cortexa9/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading 
http://downloads.openwrt.org/releases/19.07.7/targets/mvebu/cortexa9/packages/Packages.sig
Signature check passed.
Downloading 
http://downloads.openwrt.org/releases/19.07.7/targets/mvebu/cortexa9/kmods/4.14.221-1-a92a3f5c5bed2671533484c7ace9d5b5/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_kmods
Downloading 
http://downloads.openwrt.org/releases/19.07.7/targets/mvebu/cortexa9/kmods/4.14.221-1-a92a3f5c5bed2671533484c7ace9d5b5/Packages.sig
Signature check passed.
Downloading 
http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a9_vfpv3-d16/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading 
http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a9_vfpv3-d16/base/Packages.sig
Signature check passed.
Downloading 
http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a9_vfpv3-d16/freifunk/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_freifunk
Downloading 
http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a9_vfpv3-d16/freifunk/Packages.sig
Signature check passed.
Downloading 
http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a9_vfpv3-d16/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading 
http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a9_vfpv3-d16/luci/Packages.sig
Signature check passed.
Downloading 
http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a9_vfpv3-d16/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading 
http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a9_vfpv3-d16/packages/Packages.sig
Signature check passed.
Downloading 
http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a9_vfpv3-d16/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading 
http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a9_vfpv3-d16/routing/Packages.sig
Signature check passed.
Downloading 
http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a9_vfpv3-d16/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading 
http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a9_vfpv3-d16/telephony/Packages.sig
Signature check passed.
root@OpenWrt:~# opkg list-upgradable 
root@OpenWrt:~#
```

dachshund-digital, 2021-05-31:
Yeah, I did the same via the UI... set the overwrite file option.

mikeasolo, 2021-07-28:
The key itself is the same, but the header format has changed.

root@owrt:/etc/opkg/keys# cat 2f8b0b98e08306bf
untrusted comment: Public usign key for 21.02 release builds
RWQviwuY4IMGvwLfs6842A0m4EZU1IjczTxKMSk3BQP8DAQLHBwdQiaU

root@owrt:/etc/opkg/keys# cat f94b9dd6febac963
untrusted comment: Public usign key for 19.07 release builds
RWT5S53W/rrJY9BiIod3JF04AZ/eU1xDpVOb+rjZzAQBEcoETGx8BXEK

root@owrt:/etc/opkg/keys# cat f94b9dd6febac963.old
OpenWrt 19.07 release signature
RWT5S53W/rrJY9BiIod3JF04AZ/eU1xDpVOb+rjZzAQBEcoETGx8BXEK

dachshund-digital, 2021-08-14:
Ah, so that explains why.  Thus the correct procedure, is in fact to let the upgraded package overwrite the default/existing.  Which seems to work of course.

closed 2021-08-14 by dachshund-digital