openwrt/packages #15689: 20.2.0 rc1 WPA3

hello1hello2, 2021-05-25:
Please be patient with me as im new to this

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (ARMv7 Processor rev 1 (v7l), Linksys WRT1900ACS, OpenWrt 21.02.0-rc1 )

Description: WPA3 does not work on 2.4ghz. It will not start with WPA3 selected and cant connect at all.
WPA3 on 5ghz drops out here and there but will start ok and can connec to it

```
Formating code blocks by wrapping them with pairs of ```
```

jamesmacwhite, 2021-05-26:
Probably should be reported to: https://bugs.openwrt.org/ as it's part of the OpenWrt core.

BKPepe, 2021-05-27:
As @jamesmacwhite said. This issue does not belong here.

closed 2021-05-27 by BKPepe