openwrt/packages #15691: syncthing: no ppc port

Chris56, 2021-05-25:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @Paul Spooren <mail@aparcar.org>
Environment: ppc, 19.07.7

Description:
I would like to use syncthing for the powerpc processor.
However there doesn't seem to be a port for it.
Any reason why?

Also the package is quite big for embedded systems. Have all symbols been stripped? Or is it just that the linked-in libraries are quite big?

Chris

aparcar, 2021-05-28:
Not sure if golang works on ppc, can you check that?

Binaries are stripped via sstrip, maybe additional optimizing is done (or needed) for Go, no idea. 

Maybe @jefferyto knows answers to both questions

Chris56, 2021-05-29:
Well, I just use a machine which requires powerpc_464fp and I found out syncthing doen't support it.

Therefore, I don't have the development environment and cannot try build it.
But someone must have the environment running ...

Chris56, 2021-05-29:
I just had a look at: https://golang.org
It says, supported is ppc64, ppc64le
Which is 64 bit powerpc!
powerpc_464fp is 32 bit ...

aparcar, 2021-05-29:
So it's an "upstream" issue which we can't fix within OpenWrt. Sorry for that.

closed 2021-05-29 by aparcar