openwrt/packages #15695: mwan3: Speed issue?

xewonder, 2021-05-27:
Maintainer: @aaronjg @feckert
Environment: OpenWRT 21.02 RC1; 

Description:
I am running mwan3 on a wd323 with 1 WAN + 2X LTE modules set on load balancing 33% x 3

WAN has (around) 50 Mbps up and 5 Mbps down
the 2 LTE have (around) 50 Mbps up and 30 Mbps down

I understand that if I run a speed test on 1 device that I will not get 150Mbps... I am getting 70 - 80.

But if I connet 3 devices I would expect them to get around 40 Mbps each. However the sum of the 3 devices is again 70 - 80.

Am i Doing something wrong?

thank you

```
Interface status:
 interface wan is online 00h:13m:07s, uptime 00h:13m:19s and tracking is active
 interface modem1 is online 00h:30m:07s, uptime 00h:30m:12s and tracking is active
 interface modem2 is online 00h:09m:51s, uptime 00h:09m:56s and tracking is active

Current ipv4 policies:
balance_all:
 wan (33%)
 modem2 (33%)
 modem1 (33%)
coova_only:
 modem1 (100%)

Current ipv6 policies:
balance_all:
 unreachable
coova_only:
 unreachable

Directly connected ipv4 networks:
100.68.59.248/29
192.168.0.0/16
224.0.0.0/3
10.126.34.56/30
192.169.90.0/24
192.168.70.0/24
192.168.123.0/24
127.0.0.0/8

Directly connected ipv6 networks:
fe80::/64

Active ipv4 user rules:
 3194  400K - balance_all  all  --  *      *       0.0.0.0/0            0.0.0.0/0            

Active ipv6 user rules:

```

```

Output of "ip a show"
-------------------------------------------------
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 78:a3:51:73:de:79 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::7aa3:51ff:fe73:de79/64 scope link 
[... 149 lines cut ...]
 pkts bytes target     prot opt in     out     source               destination         
  332 71408 MARK       all  --  eth1   *       0.0.0.0/0            0.0.0.0/0            match-set mwan3_connected src mark match 0x0/0x3f00 /* default */ MARK or 0x3f00
  682 52445 MARK       all  --  eth1   *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00 /* wan */ MARK xset 0x100/0x3f00

Chain mwan3_ifaces_in (1 references)
 pkts bytes target     prot opt in     out     source               destination         
12031 1641K mwan3_iface_in_modem1  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00
 6858  915K mwan3_iface_in_wan  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00
 4401  569K mwan3_iface_in_modem2  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00

Chain mwan3_policy_balance_all (1 references)
 pkts bytes target     prot opt in     out     source               destination         
 1090  133K MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00 statistic mode random probability 0.33300000010 /* wan 1 3 */ MARK xset 0x100/0x3f00
 1074  132K MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00 statistic mode random probability 0.50000000000 /* modem2 1 2 */ MARK xset 0x300/0x3f00
 1089  141K MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00 /* modem1 1 1 */ MARK xset 0x200/0x3f00

Chain mwan3_policy_coova_only (0 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00 /* modem1 1 1 */ MARK xset 0x200/0x3f00

Chain mwan3_rules (1 references)
 pkts bytes target     prot opt in     out     source               destination         
 3253  406K mwan3_policy_balance_all  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00

```

feckert, 2021-06-07:
Are the connections established by the router, or does the traffic come form devices that are behind the router (LAN) ?
Maybe your boxes are not fast enough?
Have you ever tested without the mwan3? How does it look there?
Radio links are influenced by various factors.

xewonder, 2021-06-07:
hello...

As i said previously all 3 connections have the full 50 mbps without mwan3. The LTE most of the times go faster.

What do you mean by "your boxes are not fast enough"?  Running "top" while testing does not seem to show any CPU or memory "problem".

Test is done by LAN port. 

Thank you

feckert, 2021-06-10:
Can use send me your mwan3 Configuration?

The lines do not look right from my point of view

```
Chain mwan3_policy_balance_all (1 references)
 pkts bytes target     prot opt in     out     source               destination         
 1090  133K MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00 statistic mode random probability 0.33300000010 /* wan 1 3 */ MARK xset 0x100/0x3f00
 1074  132K MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00 statistic mode random probability 0.50000000000 /* modem2 1 2 */ MARK xset 0x300/0x3f00
 1089  141K MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00 /* modem1 1 1 */ MARK xset 0x200/0x3f0
```

the line for the second interface  **modem2** should also have the value 0.333~ set as the **wan**

aaronjg, 2021-06-12:
Those lines look correct to me.  The rules are sequential, so there is a 1/3 probability of the first policy, and then conditioned on the rule not matching (which is 66.7%), there is a 50% chance of matching the second policy. 

However, I find your results a bit confusing. With one device and one connection, it is just picking an interface, so surprising that you are seeing speeds in excess of any single connection.

The load balancing is randomized, and not deterministic, so it is possible with three connections they are not evenly distributed across the three interfaces, you could have two landing on one interface (most likely situation) or even all three landing on the same interface.

mradalbert, 2021-07-02:
Maybe author used multi-connection speedtest like speedtest.net.
I'm trying to configure load balancing as well and 21.02 proves it difficult. On 19.07 my setup allowed to aggregate speed of 3 links in multi connection scenario at almost 100% efficiency. Now on 20.02 (same hardware (Netgear R6220)) I'm trying to get it working with no success. This time i have two 100 mbps links but only one is used (even multiple hosts share one links bandwidth). For a brief time one WiFi device got load balancing working but it's gone now and I don't recall changing anything significant in config. Back in 21.02 rc1 I tried load balancing 3 links and strangely enough it worked with 2 links but third was unused.

Here my current configs:

`
root@Normalnie:/etc/config# cat network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option packet_steering '1'
        option ula_prefix 'fd30:63fe:3ace::/48'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan3'
        list ports 'lan4'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config interface 'wan2'
        option proto 'dhcp'
        option device 'lan1'
        option hostname '*'
        option metric '20'

config interface 'wan3'
        option proto 'dhcp'
        option device 'lan2'
        option hostname '*'
        option metric '30'

config device
        option name 'wan'
        option macaddr 'f0:76:1c:9f:f3:4c'

config device
        option name 'lan1'
        option macaddr '2c:41:38:04:07:30'

config device
        option name 'lan2'
        option macaddr '44:8a:5b:a0:6e:e1'

config interface 'wan1'
        option proto 'dhcp'
        option device 'wan'
        option metric '10'
`

`
root@Normalnie:/etc/config# cat mwan3

config globals 'globals'
        option mmx_mask '0x3F00'

config rule 'default_rule_v4'
        option dest_ip '0.0.0.0/0'
        option use_policy 'balanced'
        option family 'ipv4'
        option proto 'all'
        option sticky '0'

config rule 'default_rule_v6'
        option dest_ip '::/0'
        option use_policy 'balanced'
        option family 'ipv6'
        option proto 'all'
        option sticky '0'

config interface 'wan1'
        option initial_state 'online'
        option family 'ipv4'
        option track_method 'ping'
        option reliability '1'
        option count '1'
        option size '56'
        option max_ttl '60'
        option check_quality '0'
        option timeout '4'
        option interval '10'
        option failure_interval '5'
        option recovery_interval '5'
        option down '5'
        option enabled '1'
        list track_ip '8.8.8.8'
        list track_ip '8.8.4.4'
        option up '3'

config interface 'wan2'
        option initial_state 'online'
        option family 'ipv4'
        option track_method 'ping'
        option reliability '1'
        option count '1'
        option size '56'
        option max_ttl '60'
        option check_quality '0'
        option timeout '4'
        option interval '10'
        option failure_interval '5'
        option recovery_interval '5'
        option down '5'
        option up '3'
        list track_ip '8.8.8.8'
        list track_ip '8.8.4.4'
        option enabled '1'

config interface 'wan3'
        option enabled '1'
        option initial_state 'online'
        option family 'ipv4'
        option track_method 'ping'
        option reliability '1'
        option count '1'
        option size '56'
        option max_ttl '60'
        option check_quality '0'
        option timeout '4'
        option interval '10'
        option failure_interval '5'
        option down '5'
        option up '3'
        option recovery_interval '5'
        list track_ip '8.8.8.8'
        list track_ip '8.8.4.4'

config member 'wan1_member'
        option interface 'wan1'

config member 'wan2_member'
        option interface 'wan2'

config member 'wan3_member'
        option interface 'wan3'

config policy 'balanced'
        list use_member 'wan1_member'
        list use_member 'wan2_member'
        list use_member 'wan3_member'
        option last_resort 'unreachable'
`

`
Interface status:
 interface wan1 is online 00h:31m:26s, uptime 00h:31m:47s and tracking is active
 interface wan2 is online 37h:46m:49s, uptime 43h:58m:44s and tracking is active
 interface wan3 is offline and tracking is paused

Current ipv4 policies:
balanced:
 wan2 (50%)
 wan1 (50%)

Current ipv6 policies:
balanced:
 unreachable

Directly connected ipv4 networks:
127.0.0.0/8
192.168.1.0/24
224.0.0.0/3
10.1.0.0/16

Directly connected ipv6 networks:
fe80::/64
fd30:63fe:3ace::/64

Active ipv4 user rules:
27609 6286K S default_rule_v4  all  --  *      *       0.0.0.0/0            0.0.0.0/0            

Active ipv6 user rules:
 794K  109M - balanced  all      *      *       ::/0                 ::/0
`