openwrt/packages #15696: strongswan: strongswan-swanctl should depend on /usr/lib/ipsec/starter

stintel, 2021-05-27:
Maintainer: @pprindeville @thermi
Environment: N/A

Description:

The swanctl init script requires the binary /usr/lib/ipsec/starter, which is packaged in strongswan-mod-stroke. As strongswan-swanctl doesn't depend on strongswan-mod-stroke, people will have to manually install strongswan-mod-stroke to get a working setup. The problem is that strongswan-mod-stroke also contains deprecated files, so it's probably best to split off starter to its own package, and have both strongswan-swanctl and strongswan-ipsec depend on that new package.

Thermi, 2021-05-27:
That makes sense.

pprindeville, 2021-06-01:
@Thermi Can you handle it?

pprindeville, 2021-06-02:
Actually, we can start `charon` without `/usr/lib/ipsec/starter` as @Thermi pointed out on IRC.

This is resolved with PR #15753.

closed 2021-06-02 by pprindeville

zhoulizhengithub, 2021-06-14:
> 维护者：@pprindeville @Thermi
> 环境：不适用
> 
> 描述：
> 
> swanctl init 脚本需要二进制 /usr/lib/ipsec/starter，它被打包在 strongswan-mod-stroke 中。由于 strongswan-swanctl 不依赖于 strongswan-mod-stroke，因此人们必须手动安装 strongswan-mod-stroke 才能获得有效设置。问题是 strongswan-mod-stroke 还包含已弃用的文件，因此最好将 starter 拆分为自己的包，并且 strongswan-swanctl 和 strongswan-ipsec 都依赖于该新包。

zhoulizhengithub, 2021-06-14:
> 维护者：@pprindeville @Thermi
> 环境：不适用
>
> 描述：
>
> swanctl init 脚本需要二进制 /usr/lib/ipsec/starter，它被打包在 strongswan-mod-stroke 中。由于 strongswan-swanctl 不依赖于 strongswan-mod-stroke，因此人们必须手动安装 strongswan-mod-stroke [...]

 
通用领域
生物医药
问题是我安装了 strongswan-mod-stroke 仍然无法使用ipsec up coon,
我想使用原来的 ipsec uses starter, and reads /etc/ipsec.conf 可以恢复吗，或都做成一个可选仅单一
问题是我安装了 strongswan-mod-stroke 仍然无法使用ipsec up coon,
我想使用原来的  ipsec uses starter, and reads /etc/ipsec.conf 可以恢复吗，或都做成一个可选仅单一
The problem is that I can't use IPSec up coon after strongswan mod stroke is installed,

I want to use the original IPSec uses starter, and reads / etc / ipsec.conf, can it be recovered, or can it be made into an optional tone only

Thermi, 2021-06-14:
Sorry, we don't speak or read Mandarine, so we can't understand the non-english text around it.

The init scripts are fundamentally incompatible with each other because the swanctl one does things right(tm) and uses the newer configuration backend.

You can not use the "ipsec" init script and the "swanctl" init script concurrently.
They are mututally exclusive.

You have to either stay with the old "ipsec" one, or migrate your configs to swanctl syntax and use the swanctl init script.