openwrt/packages #15697: strongswan: both ipsec and swanctl init scripts generate /var/ipsec/strongswan.conf

stintel, 2021-05-27:
Maintainer: @pprindeville @thermi
Environment: N/A

Description:

Both ipsec and swanctl init scripts generate /var/ipsec/strongswan.conf with different content, which is utterly confusing and does not allow both to be used simultaneously. This makes it difficult to migrate connections one by one.

```
$ cat /var/ipsec/strongswan.conf
# generated by /etc/init.d/swanctl
charon {
  install_routes = no
  start-scripts {
    load-all = /usr/sbin/swanctl --load-all --noprompt
  }
  syslog {
    identifier = ipsec
    daemon {
      default =
    }
  }
}
$ /etc/init.d/ipsec restart
$ cat /var/ipsec/strongswan.conf
# generated by /etc/init.d/ipsec
charon {
  load_modular = yes
  install_routes = no
  interfaces_use = wan
    plugins {
      include /etc/strongswan.d/charon/*.conf
    }
  syslog {
    identifier = ipsec
    daemon {
      default = 1
    }
  }
}

pprindeville, 2021-06-02:
@stintel Thanks for that.

@Thermi So... we have 2 possible ways forward as I see it.  (1) is have `/etc/strongswan.conf` include two disparate files from `/var` but there's the possibility that they might have conflicting settings; or (2) try to have both subsystems converge on a common configuration (which may not be possible).  What's your guidance?

Thermi, 2021-06-02:
Conflicting settings are fine. It's up to the user to figure out the mess they created.