openwrt/packages #15714: ddns-scripts-cloudflare 2.8.2-7 failed to update DDNS

dbasinge, 2021-05-29:
Maintainer: @\looi
Environment: WRT3200ACM OpenWRT 21.02-rc1
Version: 2.8.2-7

Description:
ddns-scripts-cloudflare 2.8.2-7 failed to update DDNS 
```
 091148       : parsing script '/usr/lib/ddns/update_cloudflare_com_v4.sh'
 091148       : #> /usr/bin/curl -RsS -o /var/run/ddns/myddns_ipv4.dat --stderr /var/run/ddns/myddns_ipv4.err --noproxy '*' --header 'X-Auth-Email: username '  --header 'X-Auth-Key: ***PW***'  --header 'Content-Type: application/json'  --request GET 'https://api.cloudflare.com/client/v4/zones?name=domain'
 091148  WARN : CloudFlare reported an error:
 091148       : {"success":false,"errors":[{"code":6003,"message":"Invalid request headers","error_chain":[{"code":6102,"message":"Invalid format for X-Auth-Email header"},{"code":6103,"message":"Invalid format for X-Auth-Key header"}]}],"messages":[],"result":null}
 091148 ERROR : IP update not accepted by DDNS Provider
 091148       : Waiting 600 seconds (Check Interval)
```

rmilecki, 2022-09-27:
Those messages:
* `Invalid format for X-Auth-Email header`
* `Invalid format for X-Auth-Key header`

mean that values you used (`username ` and `***PW***`) don't match required format.

---

I get the same when I try:
```
curl --header 'X-Auth-Email: username '  --header 'X-Auth-Key: ***PW***'  --header 'Content-Type: application/json' 'https://api.cloudflare.com/client/v4/zones?name=domain'
```

---

However if I try more real-like values:
```
curl --header 'X-Auth-Email: user@example.com'  --header 'X-Auth-Key: c2547eb745079dac9320b638f5e225cf483cc5cfdda41'  --header 'Content-Type: application/json' 'https://api.cloudflare.com/client/v4/zones?name=domain'
```
I then get
```
{"success":false,"errors":[{"code":9103,"message":"Unknown X-Auth-Key or X-Auth-Email"}],"messages":[],"result":null}
```

rmilecki, 2022-09-27:
Please make sure you use correct e-mail & auth code.

closed 2022-09-27 by rmilecki