openwrt/packages #15719: clamav: Doesn't build from 19.07

jagauthier, 2021-05-30:
Maintainer: @lucize 
Environment: Debian 10 building for openwrt 19.07

Description:

```
checking whether the C++ compiler works... no
configure: error: in `/home/jgauthier/openwrt/build_dir/target-mips_24kc_musl/clamav-0.102.4':
configure: error: C++ compiler cannot create executables
See `config.log' for more details
make[3]: *** [Makefile:131: /home/jgauthier/openwrt/build_dir/target-mips_24kc_musl/clamav-0.102.4/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 77
make[3]: Leaving directory '/home/jgauthier/openwrt/feeds/packages/net/clamav'
time: package/feeds/packages/clamav/compile#16.54#8.78#35.08
make[2]: *** [package/Makefile:113: package/feeds/packages/clamav/compile] Error 2
make[2]: Leaving directory '/home/jgauthier/openwrt'
make[1]: *** [package/Makefile:107: /home/jgauthier/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jgauthier/openwrt'
make: *** [/home/jgauthier/openwrt/include/toplevel.mk:227: world] Error 2
```

Attached config.log
[config.log](https://github.com/openwrt/packages/files/6566526/config.log)

neheb, 2021-05-31:
```
/home/jgauthier/openwrt/staging_dir/toolchain-mips_24kc_gcc-8.3.0_musl/bin/g++-uc: 22: [: Illegal number: sc
/home/jgauthier/openwrt/staging_dir/toolchain-mips_24kc_gcc-8.3.0_musl/lib/gcc/mips-openwrt-linux-musl/8.3.0/../../../../mips-openwrt-linux-musl/bin/ld: cannot find -lfts
```

run make package/musl-fts/compile

and run the build again.

BKPepe, 2021-06-11:
I've just checked it and clamav is compiled for Turris routers, which they are using the latest branch of OpenWrt 19.07. So this seems to be related to your build configuration and because of that, I am closing this.

closed 2021-06-11 by BKPepe