openwrt/packages #15723: stubby: trigger option and multiple WAN interfaces

jamesmacwhite, 2021-05-31:
Maintainer: @jonathanunderwood 
Environment: OpenWrt 19.07.7 mvebu Linksys WRT3200ACM

Description:

For environments that have multiple WAN interfaces, it might be advantageous to have the trigger option support multiple WAN interfaces, specified as each interface name such as `wan wanb wanc`

https://forum.openwrt.org/t/stubby-option-trigger-help/56694/4

The original poster in the OpenWrt forum thread highlights that the trigger option doesn't seem to work with multiple WANs however the following modification to the init script, would allow this:

https://github.com/openwrt/packages/blob/master/net/stubby/files/stubby.init#L271

Replaces with a for loop for each wan interface referenced:

```sh
for trigger_item in $trigger
do
    procd_add_interface_trigger "interface.*.up" "$trigger_item" "$stubby_init" start
done
```

jamesmacwhite, 2021-06-01:
All credit to @deaddy64 for the suggested code change, as per the original thread, I thought it was worth highlighting here for potential inclusion into the main package.

jonathanunderwood, 2021-06-13:
Thanks @jamesmacwhite . I am no longer maintaining this package, but I am sure that if you prepare a PR, maintainers will merge it.

jamesmacwhite, 2021-06-14:
@jonathanunderwood Thanks for letting me know. I'll close this and open a PR.

closed 2021-06-14 by jamesmacwhite